luxonis/depthai-core #666: SpatialObjectDetector doesn't produce spatial coordinates.

PeterQFR, 2022-12-21:
@themarpe Referenced from [#651 ](https://github.com/luxonis/depthai-core/issues/651#issuecomment-1358759801)

As reported, we had depth images going into the spatial object detector, however in upgrade to 2.19.1 these lost all [spatial coordinate information](https://github.com/luxonis/depthai-core/issues/651#issuecomment-1335168036). I did confirm that a valid depth image was being linked into the spatial object detector. 

I was getting alignment warnings 

```
[1944301051BBED1200] [1.5] [60.628] [SpatialDetectionNetwork(15)] [warning] Neural network inference was performed on socket 'RGB', depth frame is aligned to socket 'RIGHT'. Bounding box mapping will not be correct, and will lead to erroneus spatial values. Align depth map to socket 'RGB' using 'setDepthAlign'.
book: 67.93
book: 67.80
book: 67.54
book: 58.83
book: 53.12
book: 49.90
book: 46.88
book: 46.09
book: 45.95
book: 45.51
book: 43.68
[1944301051BBED1200] [1.5] [60.701] [SpatialDetectionNetwork(15)] [warning] Neural network inference was performed on socket 'RGB', depth frame is aligned to socket 'RIGHT'. Bounding box mapping will not be correct, and will lead to erroneus spatial values. Align depth map to socket 'RGB' using 'setDepthAlign'.

```

Until I made the depth aligment to the RGB Socket whereupon, the spatial information was sometimes populated.

I hate to raise a second issue in this but aligning depth to the RGB socket is not a simple issue, as the depth image resolution changes to the same size as the RGB camera. This is addressed in the spatial object detector examples by resizing it to the mono cameras, but I don't see why changing the socket alignment should resize the depth image.

PeterQFR, 2022-12-21:
I'm also not sure it post processing filters that were working, particularly a decimation filter, still are applied, I think they are as the depth image seems blocky and not too speckley.

However, but the change in resolution which was previously a result in applying the decimation filter, has caused me to doubt this.

PeterQFR, 2022-12-22:
Also how does this affect intrinsic parameters? Should I be using the intrinisic params for the RGB camera on this new depth image?

PeterQFR, 2023-01-11:
@themarpe Have you seen this issue? I was able to implement my own spatial detector on the Host Side which works well, so there is definitely enough information in the depth image to work with.

themarpe, 2023-01-13:
@PeterQFR sorry for the delay. Do you mind appending the MRE and expected output?

@szabi-luxonis anything upon a quick glance from your end?
First thing that comes to mind is tweaked image transformation which would yield different results in the unaligned case

szabi-luxonis, 2023-01-13:
Not immediately sure @PeterQFR , but will be able to tell after seeing the pipeline, at first it sounds like a regression.

PeterQFR, 2023-01-29:
Hey the pipeline is [here](https://github.com/luxonis/depthai-core/issues/651#issuecomment-1338684335)

PeterQFR, 2023-01-31:
I also have noticed that when I use the Image Manipulation Node to rotate my depth image 90 degrees, that, while it transposes my pixel values, this is within a image the same shape as my original image. That is, the number of rows and cols is not changed. The data is cropped, and padded with zeros in the depth image. A similar thing might be happening on the RGB side.. @szabi-luxonis is this a clue?

Erol444, 2023-01-31:
@PeterQFR wrt rotating, would [ImageManip Rotate example](https://docs.luxonis.com/projects/api/en/latest/samples/ImageManip/image_manip_rotate/#imagemanip-rotate) work??

PeterQFR, 2023-02-01:
I did look at that an using the RotatedRect was a different approach, but I didn't understand what the order of transformations were, and the image looked a bit offset. The cropping and choosing a center were a bit unclear how they were to be applied and I didn't see a lot of difference using them. 

In the end I removed the image manip and did it on the host side using opencv.
```
cv::rotate(img, img, cv::ROTATE_90_CLOCKWISE)
```
Which was exactly what I wanted. Maybe implementing with exactly this functionality would sort out this issue.
But I think that this might also be the issue with spatial detector.

szabi-luxonis, 2023-02-01:
In your [pipeline](https://user-images.githubusercontent.com/5249779/205805114-2c509a03-8bcf-4ac0-b5dc-ad174ff7454b.png) remove the ImageManip node that rotates depth image, remove the ImageManip node that rotates `preview` from `ColorCamera`. 
If you still want to rotate `preview` ,you can move the ImageManip node after `SpatialDeteectionNetwork` and rotate the `passthrough` output.
It's likely that some image transformation data is not set properly when doing ImageManip. 
Anyway, what is the reason for rotating the image in your pipeline?

PeterQFR, 2023-02-01:
the camera is rotated, and there is a definite degradation in the object detector accuracy and recall if the image is rotated 90 degrees I found.

Thomas-C258, 2023-04-05:
I had the same problem, and using stereo.setOutputSize(480,480), I get a square depth image as output of the stereo node. As a square image, it's well rotated by ImageManip. 
A spatial detection yolo doesn't seem to mind the square depth image. I did not test accuracy though.