luxonis/depthai-core #670: [update/rebase the UVC branch to latest] {update/rebase the UVC branch to latest?}

awida1, 2022-12-28:
### Start with the `why`:
UVC branch is not up to date and cannot use features like GPIO 

### Move to the `what`:
update/rebase the UVC branch to latest

### Move to the `how`:
Update branch

Erol444, 2023-01-04:
cc @themarpe whether we have a plan to updaet the UVC branch (or perhaps merge it:)?

themarpe, 2023-01-07:
@awida1 
Updated with state of latest `develop`

awida1, 2023-01-10:
@themarpe Thank you for that. I ran the "install_requirements.py" and I am getting an error "'depthai.Pipeline' has no attribute 'createUVC'"

themarpe, 2023-01-10:
@awida1 

That API is deprecated. Use `pipeline.create(dai.node.UVC)`

awida1, 2023-01-11:
Thank you for the reply. I am getting: AttributeError: module 'depthai.node' has no attribute 'UVC'

awida1, 2023-01-11:
Also what is included in the devlop branch that was updated? I assume GPIO. Anything else? I am trying to get a project working and some areas of UVC need support like video rotation. Are any of these included? Is there a place that I can see the updates? 

Thanks!

themarpe, 2023-01-12:
You can check updates here: https://github.com/luxonis/depthai-core/releases

Nothing to relation to UVC was changes, just mainlined.

For that you might want to checkout ImageManip node, which could perform such operation

closed 2023-01-12 by themarpe

awida1, 2023-01-18:
I changed the "uvc = pipeline.createUVC()" to "pipeline.create(dai.node.UVC)" 

but I am getting the  "AttributeError: module 'depthai.node' has no attribute 'UVC'" error

Erol444, 2023-01-23:
@awida1 have you installed the latest depthai library from `gen2_uvc` branch?

awida1, 2023-01-23:
I did an "install requirements" from the gen2_UVC branch. I assume it grabs the correct one? Is that the one youre talking about?

awida1, 2023-01-24:
Maybe I am getting the syntax wrong. Is there a UVC example I can run.