luxonis/depthai-core #673: DepthAI Core - Known Issues

themarpe, 2023-01-07:
This is will be augmented post, featuring present issues and recently closed ones.
Upon a fix, a comment will be made to this post.
Recognized issues are brought forward when both `main` & `develop` are affected.

Known Issues:
- Mainline IMX283 support (Reported: https://github.com/luxonis/depthai-python/issues/725)
- OAK-D Lite specific crash with a more complex pipeline (Reported: https://github.com/luxonis/depthai/issues/892)

zxiaodong-mineral, 2024-01-23:
Has this issue[ #697](https://github.com/luxonis/depthai/issues/697) been resolved?

themarpe, 2024-01-23:
@zxiaodong-mineral no, the multiprocessing issue still seems to be relevant for the time being. It was not yet identified what causes it to happen. Any additional insight/logs/... are welcome to be added alongside in the issue