luxonis/depthai-core #674: CMAKE with OpenCV

bobdavies2000, 2023-01-09:
Windows 10
latest depthai-core
latest update of Windows
latest update to OpenCV 4.7 (2 weeks ago.)
Oak-D Lite and Oak-D Pro cameras 

I am able to build and run depthai-core with CMake and I am trying to automate its installation with my software.  The complication is that I have an existing OpenCV installation (latest 4.7) which includes all the OpenCV_contrib modules as well as opencv_core and opencv_imgproc.  I try to point my 1-line depthai-core CMake command to my OpenCV installation and it fails with the following message:

-- OpenCV or required libraries (opencv_core;opencv_imgproc) not found. OpenCV Support disabled
CMake Error at examples/CMakeLists.txt:5 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

---------------------------------

The command that generates this error is:

"C:\Program Files\CMake\bin\Cmake.exe" -H. -Bbuild -DDEPTHAI_BUILD_EXAMPLES=ON -DOpenCV_DIR=..\opencv\Build -DCMAKE_CONFIGURATION_TYPES=Debug;Release 

-----------------------------------

The OpenCV_DIR option points to a directory that contains the "OpenCVConfig.cmake" but there is no "opencv-config.cmake" present in that or any other OpenCV directory.  I suspect that the file sought is called:

opencv_python_config.cmake

which is present.  

I can continue working so there is no rush to fix this but I would like to automate the installation of depthai-core in order to enable Oak-D support in my application by default.  The app supports Microsoft's Kinect for Azure and Intel RealSense cameras and their installation scripts are automated.

Bob Davies

themarpe, 2023-01-09:
@bobdavies2000 

Do you mind specifying an absolute path to `-D OpenCV_DIR=[absolute path]` and test again?

bobdavies2000, 2023-01-09:
Thanks for the prompt reply.  The 1-line command does successfully build
after changing to an absolute path.  In my experience it has never been
necessary to use a fully-qualified path so this is a surprise.  I am glad
that it works but it means that in order to install in any location, I need
to modify the script.  Is it possible to change the cmake to accommodate a
relative path?  I hope this is not a lot of work for you.

Bob

themarpe, 2023-01-09:
I am not 100% of where the issue lies in. 

Might be similar to: https://github.com/luxonis/depthai-core/pull/645/files

If you happen to get to the bottom of it, I'd glade take a look at the PR

bobdavies2000, 2023-01-10:
The solution is to use a single quote (') around the relative name of the
file.  That solution seems to be unique to your CMake scripts.  Not sure
why.  Thanks for your help.

I am interested in purchasing the latest version of the Oak-D camera.  I
have the Oak-D Pro and a couple of Oak-D Lite's (no IMU.)  Is there
something better?  Don't go to any trouble if this requires a long answer.
I can research it on the web but it is probably a short answer for you...

Bob Davies

bobdavies2000, 2023-01-10:
Nope.  I take it back.  The cmakecache.txt was responsible.  CMake had kept
the fully qualified name in the cmakecache.txt file and so it looked like
it worked.  A clean install fails with the quoted relative name.  I will
work on this a little more and let you know if I find anything....

Bob Davies

bobdavies2000, 2023-01-10:
The solution is to use the %CD% variable in the .bat file to fully qualify the path to OpenCV.  Thanks for your suggestion.

closed 2023-01-10 by bobdavies2000

themarpe, 2023-01-10:
> I am interested in purchasing the latest version of the Oak-D camera.  I
have the Oak-D Pro and a couple of Oak-D Lite's (no IMU.)  Is there
something better?

If you are after USB connectivity, the OAK-D S2 and OAK-D Pro are top of the line. We do however have OAK-D LR & SR that are newcomers, one targeting long range and the other short range applications. Feel free to check those out here: https://shop.luxonis.com/products/oak-d-lr-pre-order and here: https://shop.luxonis.com/products/oak-d-sr

> The solution is to use the %CD% variable in the .bat file to fully qualify the path to OpenCV. Thanks for your suggestion.

Great, thanks for getting back with the solution!

bobdavies2000, 2023-01-10:
I tried reviewing the specs on the Oak-D S2 but the page:

https://shop.luxonis.com/products/oak-d-s2

Does not say it has an IMU.  I would be surprised if it doesn't but I need
to make sure...

Bob Davies

themarpe, 2023-01-10:
CC: @luxonis-david

OAK-D S2 should have an IMU onboard, either BNO08x or BMI270. Checking in with reason behind the current listing...

Luxonis-David, 2023-01-11:
It does have IMU, at presence we are assembling BMI270, later on the unit will also come with the 9-DOF BNO086.