luxonis/depthai-core #675: Unexpected pipeline break

Fguer12, 2023-01-10:
As reported on Slack already, I set-up this github issue for tracability reasons. Our pipeline crashes randomly almost every hour, though the error is caught by the Watchdog. See below some information on the system and the error message.

**Camera name:** Oak-D
**Depthai version:** depthai==2.17.4.0
**Error Message:**
DUMMY - 2023.01.07 10:31:25 - ERROR - [pipeline.py:239] - Pipeline crashed!
DUMMY - 2023.01.07 10:31:25 - ERROR - [pipeline.py:240] - Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'preview' (X_LINK_ERROR)'
DUMMY - 2023.01.07 10:31:25 - INFO - [pipeline.py:130]  - tracking pipeline created

Erol444, 2023-01-15:
Hi @Fguer12, Please first try using the latest depthai version.

Erol444, 2023-01-16:
Hi @Fguer12 ,
Please provide full debug log (from the start) and running latest depthai version. Please also share [full MRE](https://docs.luxonis.com/en/latest/pages/support/), so we can reproduce the issue and debug it.
Thanks, Erik

Fguer12, 2023-01-17:
[day (6).log](https://github.com/luxonis/depthai-core/files/10439501/day.6.log)
[MRE_depthai09-01-23.py.zip](https://github.com/luxonis/depthai-core/files/10439504/MRE_depthai09-01-23.py.zip)