luxonis/depthai-core #676: Acceleration.x Looks like y

bobdavies2000, 2023-01-11:
Windows 10
latest depthai-core
latest Windows 10 updates
OpenCV 4.7 (latest)
Oak-D Pro (purchased January 2022)

The IMU acceleration.x looks like acceleration.y and switching them makes everything look correct in OpenGL.

Here is the output of imu_gyroscope_accelerometer.cpp in the latest depthai-core download:

Accelerometer timestamp: 21513 ms
Accelerometer [m/s^2]: x: -8.945 y: -0.460 z: -0.661

The -8.9 value for x is about what you would expect for the acceleration in the Y direction - almost the gravitational constant.  This problem may be related to my purchase date - January 2022 - when the Oak-D Pro first arrived.  Early version bug?

It is not important to fix this if this is a known problem.  I am going to be purchasing an Oak-D S2 soon and I only have one of the Oak-D Pro camera but I would like to confirm that I am using the IMU interface correctly and that imu_gyroscope_accelerometer.cpp is correct.

Bob Davies

szabi-luxonis, 2023-01-11:
This depends on the coordinate system. There isn't a "correct" one.
All OAK-D's report the IMU data relative to the same coordinate system.
If you want to change the coordinate system to a different one just swap the axes in your application.

bobdavies2000, 2023-01-11:
Thanks for your reply.  I will close this issue.  All my other cameras have the y-axis pointing up.  Switching X and Y for Oak-D makes all the cameras consistent and solves my problem.

closed 2023-01-11 by bobdavies2000