luxonis/depthai-core #681: [BUG] VideoEncoder ignores specified FrameRate

camblomquist, 2023-01-16:
When running a pipeline containing a VideoEncoder node, the output frame rate matches the connected camera instead of the value provided through VideoEncoder::setFrameRate or VideoEncoder::setDefaultProfilePreset.

**Minimal Reproducible Example**
[video_fps](https://github.com/luxonis/depthai-core/files/10429582/video_fps.zip)
MRE is in Python this time, though I don't think this specific behavior is due to the Python library. Script has no additional dependencies outside depthai.
Pipeline is set up and waits on output from the camera preview (30FPS) then attempts to get output from the video encoder (Target 10FPS)
Target video frame rate and the current frame rate is printed to the terminal. Preview FPS and Video FPS are shown to be the same.

**Expected behavior**
VideoEncoder outputs data at the rate provided by VideoEncoder::setFrameRate or VideoEncoder::setDefaultProfilePreset.

**Pipeline Graph**
I can't seem to get the linked tool working. Instead, please accept this lovely ASCII graph
```
                                              +--------------+                                            
                    +-----------------+    ---|input XLinkOut|                                            
                    |Color    preview |---/   +--------------+                                            
                    |Camera     video |---\   +----------------------------+       +--------------+       
                    +-----------------+    ---|input VideoEncoder bitstream|-------|input XLinkOut|       
                                              +----------------------------+       +--------------+  
```

**Attach system log**
```
{
    "architecture": "64bit WindowsPE",
    "machine": "AMD64",
    "platform": "Windows-10-10.0.19045-SP0",
    "processor": "Intel64 Family 6 Model 165 Stepping 2, GenuineIntel",
    "python_build": "tags/v3.10.9:1dd9be6 Dec  6 2022 20:01:21",
    "python_compiler": "MSC v.1934 64 bit (AMD64)",
    "python_implementation": "CPython",
    "python_version": "3.10.9",
    "release": "10",
    "system": "Windows",
    "version": "10.0.19045",
    "win32_ver": "10 10.0.19045 SP0 Multiprocessor Free",
    "packages": [
        "av==10.0.0",
        "blobconverter==1.3.0",
        "boto3==1.26.50",
        "botocore==1.29.50",
        "certifi==2022.12.7",
        "charset-normalizer==3.0.1",
        "depthai==2.19.1.0",
        "depthai-pipeline-graph==0.0.3",
        "idna==3.4",
        "jmespath==1.0.1",
        "numpy==1.23.5",
        "opencv-python==4.6.0.66",
        "Pillow==9.3.0",
        "pip==22.3.1",
        "PySide2==5.15.2.1",
        "PySimpleGUI==4.60.3",
        "python-dateutil==2.8.2",
        "PyYAML==6.0",
        "Qt.py==1.3.7",
        "requests==2.28.2",
        "s3transfer==0.6.0",
        "setuptools==65.5.0",
        "shiboken2==5.15.2.1",
        "six==1.16.0",
        "urllib3==1.26.14"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Windows DESKTOP-HAODDON 10 10.0.19045 AMD64 Intel64 Family 6 Model 165 Stepping 2, GenuineIntel"
    ]
}
```

themarpe, 2023-01-16:
Hi @camblomquist 

At the moment VideoEncoder's setFps is more of a hint to VideoEncoder and doesn't restrict the FPS per se. It still encodes frames as they arrive to specified encoding on the fly.

This could be modified to achieve the above mentioned functionality, but it currently doesn't work in this manner.

cjwawrzonek, 2024-03-08:
Is there no way to set force the framerate? As in, you're forced to record video at whatever framerate the camera decides?

cjwawrzonek, 2024-03-11:
Clarification for anybody that sees this. You can force a lower frame rate by using the setFps function on the camera node that is used as input for the VideoEncoder node. The downside here is that you can't have a different frame rate for the VideoEncoder as you do for the underlying camera, but that's fine for my use case