luxonis/depthai-core #682: [Feature-Request] Enhancement to Brightness control in CameraControl message

chengguizi, 2023-01-17:
### Start with the `why`:
Changing of gamma curve is a very useful feature to have, it could brighten up the shade area of the image to give more information for the downstream machine vision pipeline.

To my best knowledge (discussed with @Erol444 ), the gamma curve is control by `Brightness` within `CameraControl` message. And it is performed in ISP.

However, the `Brightness` option only works for ColorCamera, and not MonoCamera. This is pretty much unexpected and undesirable.

### Move to the `what`:
Add support of `Brightness` control to MonoCamera.

### Move to the `how`:
It would be great if MonoCamera could also be route to some ISP hw so that it can get gamma correction done. 
A less ideal option, would be allow the monochrome sensor to be used as ColorCamera, but with RAW8 output. Not sure if this option is any easier to implement

themarpe, 2023-01-17:
CC: @alex-luxonis  on underlying controls for this

WRT common node, @chengguizi you may try `camera_node`, which allows for both Mono or Color sensors to be used under same interface & features (dai::node::Camera)

chengguizi, 2023-01-18:
@themarpe Very interesting to see there is work for a unified `camera_node`. May I know if that branch is still in early phase, or expected to be merged to mainline soon? I guess it is quite some significant change, as it changes fundamentally how camera node is constructed

I am using a custom firmware on a custom branch, so not too easy to try that out just yet :p

In the meantime, it is still be great if we can "backport" some of the ISP features to MonoCamera, particularly this gamma correction feature.