luxonis/depthai-core #683: [Feature-Request] Add ability to query IMU model (BNO/BMI) from IMU Node

chengguizi, 2023-01-17:
### Start with the `why`:
There are functional difference between the two IMUs, so many times a different handling logic is required. At the same time, due to parts shortage, the same product may carry different IMUs depending on the batch. However, there seems no good way to query the IMU installed on PCB in the software API. We should have one!

### Move to the `what`:
Add a API output the currently detected IMU by the VPU firmware. Exposed through IMU Node API

### Move to the `how`:
Probably A firmware feature with depthai-core API addition. Hopefully not much work needed!

Related discussion: https://discord.com/channels/790680891252932659/924797392115601439/1061121384849354864

themarpe, 2023-01-17:
+1 on the request, its in the pipeline to be added

chengguizi, 2023-02-28:
Any update on this yet? :)

szabi-luxonis, 2023-03-01:
> Any update on this yet? :)

This is a low priority task at the moment, once I get to it, this issue will be updated.

szabi-luxonis, 2023-03-09:
Added in https://github.com/luxonis/depthai-core/commit/60e7ee56361efa7e50cd42081f0249cac898202e

closed 2023-03-09 by szabi-luxonis