luxonis/depthai-core #684: [Feature-Request] Add support of BMI270 interrupt mode, instead of pooling mode

chengguizi, 2023-01-17:
### Start with the `why`:
Unlike BNO085, current BMI270 is operating in polling mode in the firmware as far as I understand. That has apparent performance impact: When the pipeline become more complex / higher load. The set frequency of IMU would not be reached. It can even be just halved (e.g. it outputs 100Hz instead of configured 200Hz). Also the timestamp is no longer perfectly even as well.

This is quite undesirable for many inertial applications, as it become harder to be calibrated.

### Move to the `what`:
Understand BMI270 was just a transition parts, due to BNO parts shortage. But completing the support for BMI270 is appreciated, in view of a fair amount of Luxonis products being produced with BMI270 installed. Therefore, I would love to see BMI270 eventually have interrupt mode supported, if technically possible.

### Move to the `how`:
Likely a firmware update to let BMI270 to have interrupt mode, so we can have a consistent rate IMU output from BMI270-installed products.

chengguizi, 2023-01-17:
Example output from BMI270, non-fixed intervals
```
seq: 204174
stamp: 
  secs: 50045
  nsecs: 952712712
frame_id: "imu"
---
seq: 204175
stamp: 
  secs: 50045
  nsecs: 956714225
frame_id: "imu"
---
seq: 204176
stamp: 
  secs: 50045
  nsecs: 966713452
frame_id: "imu"
---
seq: 204177
stamp: 
  secs: 50045
  nsecs: 970712740
frame_id: "imu"
---
seq: 204178
stamp: 
  secs: 50045
  nsecs: 974781139
frame_id: "imu"
---
seq: 204179
stamp: 
  secs: 50045
  nsecs: 984719209
frame_id: "imu"
```

AndreaTagliavini, 2023-02-20:
Any updates regarding this?

themarpe, 2023-02-24:
No update on this yet. CC: @szabi-luxonis on potential improvements switching to interrupt mode, or are there other limits at play regardless?

szabi-luxonis, 2023-03-01:
> No update on this yet. CC: @szabi-luxonis on potential improvements switching to interrupt mode, or are there other limits at play regardless?

Interrupt mode didn't work. I will check with the scope, once we receive it.

szabi-luxonis, 2023-03-03:
@chengguizi added here, with details on what was the issue: https://github.com/luxonis/depthai-python/pull/777

chengguizi, 2023-03-04:
Greatly appreciate! @szabi-luxonis

szabi-luxonis, 2023-03-05:
Merged to develop

closed 2023-03-05 by szabi-luxonis