luxonis/depthai-core #688: [Feature-Request] Shorten latency of CameraControl message

chengguizi, 2023-01-21:
### Start with the `why`:
I am experimenting camera control to have autoexposure on host. similar to what discussed here https://discord.com/channels/790680891252932659/924798194221723668/1025150543716421662

However, I realise there is quite some latency from sending the camera control, until receiving the adjusted images. It is about 7 frame delays, with 30fps setting. thats about 250ms latency. This is actually quite long. I would expect the delay to be about 2-3 frames (60ms ~ 100ms) to be more reasonable.

### Move to the `what`:
From what I understand, most of the delay should be introduced by xlink. Perhaps the queue / buffer is not optimised for xlink input. I know for IMU stream, Luxonis has made special care on API such as `setBatchReportThreshold` and `setMaxBatchReports` to reduce latency. In fact, the latency i observed could be as low as a few milliseconds, really good.

Could we get the DepthAI firmware to offer the same kind of optimisation for CameraControl input as well?

### Move to the `how`:
I would imagine, we should have an API to allow us to specify Xlink Input to send CameraControl immediately, without batching or delaying. Hope this is actually possible!

themarpe, 2023-01-21:
@chengguizi I assume the latency in this case boils down to 3A algos taking these changes into account (even in manual exposure case). Lower latency is likely achievable, but this isn't something we are looking to optimize in the near term, unless its critical.

chengguizi, 2023-01-25:
Got it @themarpe , I understand there is a lot on the plate for Luxonis tech teams right now, and the Camera Control latency is not a very critical one.

I partially agree with you that the 3A algorithm should be latency-aware. But 7 frames delay (~0.3 second) is quite a lot to handle.

Let's see if this feature is on anyone else's wish list!