luxonis/depthai-core #693: [BUG] Device.isClosed() not Returning True for Oak1 PoE device when pulling network cable.

a3dbox, 2023-01-26:
**Check if issue already exists**
I have not been able to find related results on google.
Erik on the Discord channel requested I make a bug report here.

**Describe the bug**
The isClosed() method does not provide the desired results when testing with a Oak PoE network camera.  When I pull the network cable while it is running this method does not return true.  Performing the same test with an Oak1 USB camera returns True as expected.

**Minimal Reproducible Example**
```
import time
import depthai as dai  # depthai - access the camera and its data packets

def connect_and_process():
    with dai.Device(dai.Pipeline()) as device:

        while True:
            if device.isClosed():
                print("Camera Disconnected!")
                quit()

            print("Camera Connected...")
            time.sleep(1)

connect_and_process()
```

**Expected behavior**
Print out "Camera Connected..." until you physically pull the network cable from the camera, at which point it prints "Camera Disconnected!".

**Screenshots**
The following screenshot demonstrates the issue.  You can see the point the network camera was disconnected by the depthai log stating a ping was missed.
[image]

The following is the same code running with a USB camera connected:
```
/home/andrew/code/PythonEnvs/DepthAI/bin/python /home/andrew/code/Holovis/Lego/IsConnectedTest.py 
Camera Connected...
Camera Connected...
Camera Connected...
Camera Connected...
Camera Connected...
Camera Disconnected!

Process finished with exit code 0
```

themarpe, 2023-01-26:
@a3dbox thanks for the writeup

This missed ping should indeed teardown the device and the connection. I'll take a look at this.

themarpe, 2023-01-26:
@a3dbox 

```
Camera Connected...
Camera Connected...
Camera Connected...
[2023-01-26 19:21:45.303] [warning] Monitor thread (device: 14442C105183A7D200 [169.254.1.222]) - ping was missed, closing the device connection
Camera Disconnected!
```

This is on v2.19.1 /w OAK-D PoE - do you mind updating to this version and retesting?

a3dbox, 2023-01-26:
Thanks for looking into this, I will double check versions when I'm back at the office tomorrow.

a3dbox, 2023-01-27:
@themarpe , thanks for you time and I am embarrassed to say although I was working with the latest version on my laptop the testing rig was running from v2.17.  Updating to the latest fixes the issue.

Thanks very much for you time spent, its much appreciated.

closed 2023-01-27 by a3dbox