luxonis/depthai-core #701: [BUG] [critical] Fatal error. when multiple stereo depth nodes are used with Median Filters

chengguizi, 2023-01-28:
**Describe the bug**

I am experiencing this bug on both SoM-based products, including OAK-D-LR. I am trying to bring up 4 x stereo depth nodes.

However, it would only work if I set to median filter to off, like what is shown here https://github.com/luxonis/depthai-python/blob/3be64e03f488359bbb1ba339ef6ef6e4d54f998d/examples/StereoDepth/depth_preview_lr.py

If I turn on the median filter, I obtain error at the start of the run, like

```
[1844301061DF5D1200] [3.3] [2.471] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 8022E600' '0'
```

**Minimal Reproducible Example**

DepthAI Version: v2.20.1 (Python API)
Test Device: OAK-D-LR (SoM based)

The reproduction python code I will attach shortly

**Expected behavior**

4 x stereo should be brought up normally, regardless of the config settings, like median filter

chengguizi, 2023-01-28:
```
#!/usr/bin/env python3

import cv2
import depthai as dai
import numpy as np

# Closer-in minimum depth, disparity range is doubled (from 95 to 190):
extended_disparity = False
# Better accuracy for longer distance, fractional disparity 32-levels:
subpixel = False
# Better handling for occlusions:
lr_check = False

enableRectified = False

# Create pipeline
pipeline = dai.Pipeline()

# Define sources and outputs
left = pipeline.create(dai.node.ColorCamera)
center = pipeline.create(dai.node.ColorCamera)
right = pipeline.create(dai.node.ColorCamera)
LC_depth = pipeline.create(dai.node.StereoDepth)
LR_depth = pipeline.create(dai.node.StereoDepth)
CR_depth = pipeline.create(dai.node.StereoDepth)
CR_depth2 = pipeline.create(dai.node.StereoDepth)

xout_LC = pipeline.create(dai.node.XLinkOut)
xout_LR = pipeline.create(dai.node.XLinkOut)
xout_CR = pipeline.create(dai.node.XLinkOut)
xout_CR2 = pipeline.create(dai.node.XLinkOut)

xout_LC.setStreamName("disparity_LC")
if enableRectified:
    xoutl_LC = pipeline.create(dai.node.XLinkOut)
    xoutr_LC = pipeline.create(dai.node.XLinkOut)
    xoutl_LC.setStreamName("rectifiedLeft_LC")
    xoutr_LC.setStreamName("rectifiedRight_LC")

xout_LR.setStreamName("disparity_LR")
if enableRectified:
    xoutl_LR = pipeline.create(dai.node.XLinkOut)
    xoutr_LR = pipeline.create(dai.node.XLinkOut)
    xoutl_LR.setStreamName("rectifiedLeft_LR")
    xoutr_LR.setStreamName("rectifiedRight_LR")

xout_CR.setStreamName("disparity_CR")
if enableRectified:
    xoutl_CR = pipeline.create(dai.node.XLinkOut)
    xoutr_CR = pipeline.create(dai.node.XLinkOut)
    xoutl_CR.setStreamName("rectifiedLeft_CR")
    xoutr_CR.setStreamName("rectifiedRight_CR")

xout_CR2.setStreamName("disparity_CR2")
if enableRectified:
    xoutl_CR2 = pipeline.create(dai.node.XLinkOut)
    xoutr_CR2 = pipeline.create(dai.node.XLinkOut)
    xoutl_CR2.setStreamName("rectifiedLeft_CR2")
    xoutr_CR2.setStreamName("rectifiedRight_CR2")

# Properties
left.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1200_P)
left.setBoardSocket(dai.CameraBoardSocket.LEFT)
left.setIspScale(2, 3)

center.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1200_P)
center.setBoardSocket(dai.CameraBoardSocket.CENTER)
center.setIspScale(2, 3)

right.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1200_P)
right.setBoardSocket(dai.CameraBoardSocket.RIGHT)
right.setIspScale(2, 3)

LC_depth.setDefaultProfilePreset(dai.node.StereoDepth.PresetMode.HIGH_DENSITY)
LC_depth.initialConfig.setMedianFilter(dai.MedianFilter.KERNEL_5x5)
LC_depth.setLeftRightCheck(lr_check)
LC_depth.setExtendedDisparity(extended_disparity)
LC_depth.setSubpixel(subpixel)

LR_depth.setDefaultProfilePreset(dai.node.StereoDepth.PresetMode.HIGH_DENSITY)
LR_depth.initialConfig.setMedianFilter(dai.MedianFilter.KERNEL_5x5)
LR_depth.setLeftRightCheck(lr_check)
LR_depth.setExtendedDisparity(extended_disparity)
LR_depth.setSubpixel(subpixel)

CR_depth.setDefaultProfilePreset(dai.node.StereoDepth.PresetMode.HIGH_DENSITY)
CR_depth.initialConfig.setMedianFilter(dai.MedianFilter.KERNEL_5x5)
CR_depth.setLeftRightCheck(lr_check)
CR_depth.setExtendedDisparity(extended_disparity)
CR_depth.setSubpixel(subpixel)

CR_depth2.setDefaultProfilePreset(dai.node.StereoDepth.PresetMode.HIGH_DENSITY)
CR_depth2.initialConfig.setMedianFilter(dai.MedianFilter.KERNEL_5x5)
CR_depth2.setLeftRightCheck(lr_check)
CR_depth2.setExtendedDisparity(extended_disparity)
CR_depth2.setSubpixel(subpixel)

# Linking
# LC
left.isp.link(LC_depth.left)
center.isp.link(LC_depth.right)
LC_depth.disparity.link(xout_LC.input)
if enableRectified:
    LC_depth.rectifiedLeft.link(xoutl_LC.input)
    LC_depth.rectifiedRight.link(xoutr_LC.input)
# LR
left.isp.link(LR_depth.left)
right.isp.link(LR_depth.right)
LR_depth.disparity.link(xout_LR.input)
if enableRectified:
    LR_depth.rectifiedLeft.link(xoutl_LR.input)
    LR_depth.rectifiedRight.link(xoutr_LR.input)
# CR
center.isp.link(CR_depth.left)
right.isp.link(CR_depth.right)
CR_depth.disparity.link(xout_CR.input)
if enableRectified:
    CR_depth.rectifiedLeft.link(xoutl_CR.input)
    CR_depth.rectifiedRight.link(xoutr_CR.input)

# CR 2
center.isp.link(CR_depth2.left)
right.isp.link(CR_depth2.right)
CR_depth2.disparity.link(xout_CR2.input)
if enableRectified:
    CR_depth2.rectifiedLeft.link(xoutl_CR2.input)
    CR_depth2.rectifiedRight.link(xoutr_CR2.input)

maxDisp = LC_depth.initialConfig.getMaxDisparity()

# Connect to device and start pipeline
with dai.Device(pipeline) as device:
    while not device.isClosed():
        queueNames = device.getQueueEvents()
        for q in queueNames:
            message = device.getOutputQueue(q).get()
            # Display arrived frames
            if type(message) == dai.ImgFrame:
                frame = message.getCvFrame()
                if 'disparity' in q:
                    disp = (frame * (255.0 / maxDisp)).astype(np.uint8)
                    disp = cv2.applyColorMap(disp, cv2.COLORMAP_JET)
                    cv2.imshow(q, disp)
                else:
                    cv2.imshow(q, frame)
        if cv2.waitKey(1) == ord('q'):
            break
```

chengguizi, 2023-01-28:
If I turn the setting to `dai.MedianFilter.MEDIAN_OFF`, things works normally.

szabi-luxonis, 2023-01-28:
Check out latest https://github.com/luxonis/depthai-python/pull/750, median filter is fixed. @chengguizi

chengguizi, 2023-01-30:
Hi @szabi-luxonis , thanks for suggestion.

I have tried out on commit 
```
commit ab23c2817e0bd8ec86e9b3a4ffee6796f8e34822 (HEAD -> multi_stereo_support, origin/multi_stereo_support)
Author: SzabolcsGergely <szabi@luxonis.com>
Date:   Sat Jan 28 16:19:06 2023 +0200

    Update FW: Fix OOM due to too many SIPP pipelines

```

I encounter error even before moving to non-OFF median filter
```
[1844301061DF5D1200] [3.8] [1.815] [StereoDepth(3)] [error] There is no available extrinsic calibration between camera ID: '1' and '0'
[1844301061DF5D1200] [3.8] [2.993] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
Traceback (most recent call last):
  File "/home/huimin/depthai/depthai-python-4cam-custom/depthai-python/examples/StereoDepth/depth_multiple.py", line 136, in <module>
    queueNames = device.getQueueEvents()
ValueError: Device already closed or disconnected

```

Where as on v2.20.1.0, things works fine

chengguizi, 2023-01-30:
I realise I didnt have the calibration info for camera 0 (rgb). So I added it manually and flashed.

It is still give me warning on the '1' and '0' calibraiton.

And anyway, with KERNEL size 5X5, I still obtain error:
```
[1844301061DF5D1200] [3.8] [1.333] [StereoDepth(3)] [error] There is no available extrinsic calibration between camera ID: '1' and '0'
[1844301061DF5D1200] [3.8] [2.508] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 80230F08' '0'
Traceback (most recent call last):
  File "/home/huimin/depthai/depthai-python-4cam-custom/depthai-python/examples/StereoDepth/depth_multiple.py", line 136, in <module>
    queueNames = device.getQueueEvents()
ValueError: Device already closed or disconnected

```

szabi-luxonis, 2023-01-30:
> It is still give me warning on the '1' and '0' calibraiton.

Because you don't have, that's why the crash. Add a dummy one. Later it will be handled more gracefully.

chengguizi, 2023-01-30:
Yes I added the dummy one. It is still the same warning.

```
{
    "batchName": "",
    "batchTime": 1671794919,
    "boardConf": "nIR-C30M00-00",
    "boardCustom": "",
    "boardName": "BC2087",
    "boardOptions": 1025,
    "boardRev": "R0M0E0",
    "cameraData": [
        [
            2,
            {
                "cameraType": 1,
                "distortionCoeff": [
                    0.3534858822822571,
                    0.19323799014091492,
                    -0.30427980422973633,
                    0.2642624080181122,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0
                ],
                "extrinsics": {
                    "rotationMatrix": [
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ]
                    ],
                    "specTranslation": {
                        "x": 0.0,
                        "y": 0.0,
                        "z": 0.0
                    },
                    "toCameraSocket": -1,
                    "translation": {
                        "x": 0.0,
                        "y": 0.0,
                        "z": 0.0
                    }
                },
                "height": 1200,
                "intrinsicMatrix": [
                    [
                        1148.7784423828125,
                        0.0,
                        922.7352294921875
                    ],
                    [
[... 212 lines cut ...]
            ]
        ],
        "rectifiedRotationRight": [
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ]
        ],
        "rightCameraSocket": 255
    },
    "version": 7
}

```

szabi-luxonis, 2023-01-30:
@chengguizi 
It's a hard error, not a warning.
There is no calibration data found between cameras "0" and "1" therefore stereo depth can't be performed.

chengguizi, 2023-01-30:
Yes noted. But after I added the dummy one it is still the same issue. Is there anything wrong to what I added? I have all 0,1,2 cameras defined in the json (i used API to dump that json file from device.)

szabi-luxonis, 2023-01-30:
Latest https://github.com/luxonis/depthai-core/pull/695 has more graceful handling for the above case, when calibration info is wron.

chengguizi, 2023-01-31:
Thanks for the sample calib. I flashed that one into my LR device. And run the depth multiple script, with `dai.MedianFilter.KERNEL_5x5`

I got the following error now:
```
[1844301061DF5D1200] [3.8] [1.920] [system] [critical] Fatal error. Please report to developers. Log: 'MedianLut' '213'
[1844301061DF5D1200] [3.8] [2.994] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 802311B8' '0'
Traceback (most recent call last):
  File "...", line 136, in <module>
    queueNames = device.getQueueEvents()
ValueError: Device already closed or disconnected
```

Complete log with trace turned on
```
[2023-01-31 11:19:14.884] [debug] Python bindings - version: 2.20.2.0.dev+88aa2b952a430508aeab04c23c0cfe303faddb80 from  build: 2023-01-30 17:36:00 +0000
[2023-01-31 11:19:14.884] [debug] Library information - version: 2.20.2, commit: 9eb556d5a0ac17eb78f54149de207a4df6fab73e from 2023-01-30 19:10:40 +0200, build: 2023-01-30 17:19:02 +0000
[2023-01-31 11:19:14.885] [debug] Initialize - finished
[2023-01-31 11:19:14.971] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 3ms, archive read: 75ms
[2023-01-31 11:19:15.101] [debug] Device - OpenVINO version: 2022.1
[2023-01-31 11:19:15.260] [debug] Resources - Archive 'depthai-device-fwp-7d93c101945b03db76114a362af353379c7d9a74.tar.xz' open: 2ms, archive read: 373ms
[2023-01-31 11:19:15.276] [debug] Searching for booted device: DeviceInfo(name=3.8, mxid=1844301061DF5D1200, X_LINK_BOOTLOADER, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-01-31 11:19:15.281] [debug] Connected bootloader version 0.0.23+9704fa354a72599286e0be08663cee084266fda6
[2023-01-31 11:19:15.281] [info] New bootloader version available. Device has: 0.0.23+9704fa354a72599286e0be08663cee084266fda6, available: 0.0.24
[2023-01-31 11:19:16.227] [debug] Booting FW with Bootloader. Version 0.0.23+9704fa354a72599286e0be08663cee084266fda6, Time taken: 946ms
[2023-01-31 11:19:16.227] [debug] DeviceBootloader about to be closed...
[2023-01-31 11:19:16.228] [debug] XLinkResetRemote of linkId: (0)
[2023-01-31 11:19:16.781] [debug] DeviceBootloader closed, 553
[2023-01-31 11:19:16.784] [debug] Searching for booted device: DeviceInfo(name=3.8, mxid=1844301061DF5D1200, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-01-31 11:19:17.527] [trace] RPC: [1,1,10182484315255513117,[0]]
[2023-01-31 11:19:17.530] [trace] RPC: [1,1,5804304869041345055,[1.0]]
[2023-01-31 11:19:17.532] [trace] RPC: [1,1,10182484530226687077,[5000,10,true]]
[2023-01-31 11:19:17.532] [trace] Log vector decoded, size: 3
[1844301061DF5D1200] [3.8] [0.984] [system] [info] Memory Usage - DDR: 0.12 / 340.39 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 7.25 / 77.32 MiB, LeonRT Heap: 2.89 / 41.22 MiB
[1844301061DF5D1200] [3.8] [0.984] [system] [info] Temperatures - Average: 29.22 °C, CSS: 31.04 °C, MSS 28.13 °C, UPA: 28.86 °C, DSS: 28.86 °C
[1844301061DF5D1200] [3.8] [0.984] [system] [info] Cpu Usage - LeonOS 75.29%, LeonRT: 1.65%
[2023-01-31 11:19:17.536] [trace] RPC: [1,1,17425566508637143278,null]
[2023-01-31 11:19:17.538] [debug] Schema dump: {"connections":[{"node1Id":6,"node1Output":"disparity","node1OutputGroup":"","node2Id":10,"node2Input":"in","node2InputGroup":""},{"node1Id":2,"node1Output":"isp","node1OutputGroup":"","node2Id":6,"node2Input":"right","node2InputGroup":""},{"node1Id":1,"node1Output":"isp","node1OutputGroup":"","node2Id":6,"node2Input":"left","node2InputGroup":""},{"node1Id":5,"node1Output":"disparity","node1OutputGroup":"","node2Id":9,"node2Input":"in","node2InputGroup":""},{"node1Id":2,"node1Output":"isp","node1OutputGroup":"","node2Id":5,"node2Input":"right","node2InputGroup":""},{"node1Id":1,"node1Output":"isp","node1OutputGroup":"","node2Id":5,"node2Input":"left","node2InputGroup":""},{"node1Id":4,"node1Output":"disparity","node1OutputGroup":"","node2Id":8,"node2Input":"in","node2InputGroup":""},{"node1Id":2,"node1Output":"isp","node1OutputGroup":"","node2Id":4,"node2Input":"right","node2InputGroup":""},{"node1Id":0,"node1Output":"isp","node1OutputGroup":"","node2Id":4,"node2Input":"left","node2InputGroup":""},{"node1Id":3,"node1Output":"disparity","node1OutputGroup":"","node2Id":7,"node2Input":"in","node2InputGroup":""},{"node1Id":1,"node1Output":"isp","node1OutputGroup":"","node2Id":3,"node2Input":"right","node2InputGroup":""},{"node1Id":0,"node1Output":"isp","node1OutputGroup":"","node2Id":3,"node2Input":"left","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[0,{"id":0,"ioInfo":[[["","video"],{"blocking":false,"group":"","id":92,"name":"video","queueSize":8,"type":0,"waitForMessage":false}],[["","still"],{"blocking":false,"group":"","id":90,"name":"still","queueSize":8,"type":0,"waitForMessage":false}],[["","isp"],{"blocking":false,"group":"","id":89,"name":"isp","queueSize":8,"type":0,"waitForMessage":false}],[["","preview"],{"blocking":false,"group":"","id":91,"name":"preview","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","id":88,"name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","id":87,"name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inpu [... cut ...]
[2023-01-31 11:19:22.034] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
[2023-01-31 11:19:22.069] [debug] Device about to be closed...
[2023-01-31 11:19:22.069] [debug] XLinkResetRemoteTimeout returned: X_LINK_ERROR
[2023-01-31 11:19:22.069] [debug] XLinkResetRemote of linkId: (0)
[2023-01-31 11:19:23.528] [debug] Device closed, 1458
Traceback (most recent call last):
  File "...", line 136, in <module>
    queueNames = device.getQueueEvents()
ValueError: Device already closed or disconnected

```

szabi-luxonis, 2023-02-01:
@chengguizi handled in 80727e4b5d1f8f4c6ce9d2c25f429aa6aba92a94 .
Note that multi stereo branch is in regression testing phase.

szabi-luxonis, 2023-02-06:
Fixed in `develop`

closed 2023-02-06 by szabi-luxonis