luxonis/depthai-core #707: How do I stop and then restart an Oak-D camera

bobdavies2000, 2023-01-30:
My application supports multiple cameras and when I switch to or from the Oak-D camera everything is working fine.  However, when I switch back to the Oak-D camera that was previously active, it doesn't work.  I found the device.close() interface and I use that when closing the device.  I looked through all the examples and none of them close the camera.  Nor do any stop and then restart the camera.  Is there a chance you could provide an example that shows how to close and restart a camera?

masterbuchi, 2023-06-16:
I would also like to know that. device.close() doesn't really seem to help, because my python-code is throwing an "X_LINK_ERROR" when trying to create a new connection.

themarpe, 2023-06-19:
In Python, best is to leave the "device" object get closed from exiting the `with` scope

In C++ similar, letting the destructor run.

---

If you have issues reconnecting, for the time being, I suggest waiting a tad between reconnecting, as the device restarts and is temporarily unavailable after being closed