luxonis/depthai-core #719: [BUG]Not able to access UART on FFC-4P

gajena, 2023-02-07:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
I am trying to access UART on FFC-4P but no luck, I tried searching  and found [this](https://github.com/luxonis/depthai-spi-api/issues/8) with this I am getting the error
```shell
[1844301021BACB0F00] [1.1] [8.599] [Script(0)] [warning] Sending message to UART: hello world from pyserial
[1844301021BACB0F00] [1.1] [8.600] [Script(0)] [critical] SerialException: [Errno 2] could not open port /dev/ttyS0: [Errno 2] N: '/dev/ttyS0'

At:
  /home/gitlab-runner/builds/wDV4xNMi/0/luxonis/myriadx/depthai/3rdparty/cpython/Packages/pyserial/serial/serialposix.py(294): open
  /home/gitlab-runner/builds/wDV4xNMi/0/luxonis/myriadx/depthai/3rdparty/cpython/Packages/pyserial/serial/serialutil.py(180): __init__
  <script>(11): <module>
```

**Minimal Reproducible Example**
```python
#!/usr/bin/env python3
import sys, time
import depthai as dai

# Shorthand for GPIO
GPIO = dai.BoardConfig.GPIO

# Choose UART number from the list below: 0, 2, 3
uartNum = 0
uarts = {
    0: { # Suitable for the custom TG board
        'txPin' : 15,
        'rxPin' : 16,
        'txPinMode' : GPIO.ALT_MODE_2,
        'rxPinMode' : GPIO.ALT_MODE_2,
    },
    2: {  # Suitable for 1099 with U_TX and U_RX testpoints
        'txPin' : 45,
        'rxPin' : 46,
        'txPinMode' : GPIO.ALT_MODE_3,
        'rxPinMode' : GPIO.ALT_MODE_3,
    },
    3: { # Suitable for 1099 or 2099 - these go to ESP32 or SD card (3.3V pins)
        'txPin' : 34,
        'rxPin' : 35,
        'txPinMode' : GPIO.ALT_MODE_5,
        'rxPinMode' : GPIO.ALT_MODE_5,
    },
}

if len(sys.argv) < 2:
    print('Usage: main.py tty/pyserial')
    exit(0)

# Start defining a pipeline
pipeline = dai.Pipeline()

# Script node
script = pipeline.create(dai.node.Script)
script.setProcessor(dai.ProcessorType.LEON_CSS)

pyserial = """
    import serial
    import time

    # time.sleep(50)

    msg = 'hello world from pyserial'
    node.warn(f'Sending message to UART: {msg}')

    # Open UART3 port and set baudrate
    ser = serial.Serial("/dev/ttyS%d", baudrate=115200)

    while True:
        # Send hello message and print back received message
        ser.write(msg.encode())
        node.warn(f'Read: {ser.read(16)}')
        time.sleep(1)
""" % (uartNum)

tty = """
    import os, tty
    import time

    # time.sleep(50)

    msg = 'hello world from tty'
    node.warn(f'Sending message to UART: {msg}')

    # Open UART3 port, set raw and baudrate
    fd = os.open("/dev/ttyS%d", os.O_RDWR | os.O_NOCTTY)
    tty.setraw(fd)
    mode = tty.tcgetattr(fd)
    mode[tty.ISPEED] = mode[tty.OSPEED] = tty.B115200
    tty.tcsetattr(fd, tty.TCSANOW, mode)

    while True:
        # Send hello message and print back received message
        os.write(fd, msg.encode())
        node.warn(f'Read: {os.read(fd, 16)}')
        time.sleep(1)

""" % (uartNum)

if sys.argv[1] == 'tty':
    script.setScript(tty)
else:
    script.setScript(pyserial)

xout = pipeline.create(dai.node.XLinkOut)
xout.setStreamName('end')
script.outputs['end'].link(xout.input)

# Create BoardConfig and enable UART3 on pins 34 and 35
config = dai.Device.Config()
# UART TX
config.board.gpio[uarts[uartNum]['txPin']] = GPIO(GPIO.OUTPUT, uarts[uartNum]['txPinMode'])
# UART RX
config.board.gpio[uarts[uartNum]['rxPin']] = GPIO(GPIO.INPUT, uarts[uartNum]['rxPinMode'])
# Enable UART{uartNum}
config.board.uart[uartNum] = dai.BoardConfig.UART()
config.version = config.version.VERSION_2021_4

# Connect to device with pipeline
with dai.Device(config) as device:
    device.startPipeline(pipeline)
    while not device.isClosed():
        time.sleep(1)

```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Pipeline Graph**

Please also provide a screenshot of your pipeline using the [DepthAI Pipeline Graph](https://github.com/geaxgx/depthai_pipeline_graph).

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)

**Additional context**
Add any other context about the problem here.

alex-luxonis, 2023-02-07:
@gajena Can you try building from the branch `fix_uart0_script` (`python3 examples/install_requirements.py`), the issue with device `/dev/ttyS0` should be fixed.

For reference, I'm also marking below the UART pins on the AUX pin header of FFC-4P:
[image]
[image]

gajena, 2023-02-08:
uart works fine with `fix_uart0_script` branch but simple image rotation are not working with the provided MRE.
```python
#!/usr/bin/env python3

import cv2
import depthai as dai

# Create pipeline
pipeline = dai.Pipeline()

# Define source and output
camRgb = pipeline.create(dai.node.ColorCamera)
camRgb.setBoardSocket(dai.CameraBoardSocket.LEFT)
camRgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
camRgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.BGR)
camRgb.initialControl.setManualFocus(1)

crop_manip = pipeline.create(dai.node.ImageManip)
crop_manip.initialConfig.setFrameType(dai.ImgFrame.Type.NV12)
crop_manip.initialConfig.setResize(1920, 1080)
crop_manip.setNumFramesPool(1)
crop_manip.setMaxOutputFrameSize(3110400)
crop_manip.initialConfig.setRotationDegrees(11)

camRgb.isp.link(crop_manip.inputImage)

xoutVideo = pipeline.create(dai.node.XLinkOut)
xoutVideo.setStreamName("video")
xoutVideo.input.setBlocking(True)
xoutVideo.input.setQueueSize(1)
crop_manip.out.link(xoutVideo.input)

# Connect to device and start pipeline
with dai.Device(pipeline) as device:

    video = device.getOutputQueue(name="video")

    switch = False
    while True:
        videoIn = video.get()

        frame = videoIn.getCvFrame()

        cv2.imshow("video", frame)
        key = cv2.waitKey(1)
        if key == ord('q'):
            break

```

alex-luxonis, 2023-02-08:
Thanks @gajena for testing and for the new issue report. 
The `fix_uart0_script` is based on top of `develop`, and looks like the ImageManip issue is newly introduced on `develop`. Some amounts of rotation work fine, others end up producing no frames. 

Same issue can be reproduced with https://github.com/luxonis/depthai-python/blob/develop/examples/ImageManip/rgb_rotate_warp.py
after increasing the size, if freezes after few frames (when it gets to a certain angle that increases progressively):
```diff
-camRgb.setPreviewSize(640, 480)
+camRgb.setPreviewSize(1920, 1080)
...
-                rotatedRect = ((320, 240), (400, 400), angleDeg)
+                rotatedRect = ((960, 480), (1920, 1080), angleDeg)
```
and: `[critical] Fatal error. Please report to developers. Log: 'ResourceLocker' '358'`

We'll investigate.

closed 2023-02-21 by szabi-luxonis

Maimilian, 2023-04-03:
@gajena Hi, I want to use the ports 6 and 8 of the aux part of the oak ffc 4p (marked in the photo) for serial Communication. Which uart number do I have to choose and what will the programm look like? Tanks for your help, I am new to this...

gajena, 2023-04-04:
@Maimilian please check the example given at top of this issue.

Maimilian, 2023-04-04:
@gajena this means that I have to choose uart number 2 with rx and tx ports 6 and 8 right?

alex-luxonis, 2023-04-04:
@Maimilian It has to be UART number `0`, that's already configured there: `uartNum = 0`.
Note those are 1.8V pins.
Also please use the latest release `2.21.0.0`, on main or develop branch, where the UART0 issue has been fixed.

Maimilian, 2023-04-04:
@alex-luxonis Thanks for your help! I will try it in a few days...

Maimilian, 2023-04-13:
@alex-luxonis one last question: which pins do I have to choose for serial communication via these aux ports? 6 & 8 or 15 & 16 as given in the example?

Maimilian, 2023-04-13:
@gajena @alex-luxonis Hi again, serial communication works. Thanks a lot! However, now I want to have my program in standalone mode. For this reason I have to flash it. How can I flash the config?

nemc0, 2023-05-12:
hey @Maimilian, thanks for raising this issue and the help @alex-luxonis 

@Maimilian trying to get the same thing working with the FFC-4P in standalone mode, did you have any luck with this?

I did see some forum post about this not working as the USB boot loader couldnt see the eMMC