luxonis/depthai-core #721: [Feature-Request] Reboot device in standalone mode from a script

kanoaellis, 2023-02-08:
It might be useful to be able to be able to reboot the device or even just restart the script while a device is running in standalone mode.

themarpe, 2023-02-10:
@kanoaellis 

Sorry for the delay - already in, do you mind trying it out with: `shutdownResetDevice(True, True, True)` - CC: @Erol444  for documentation

closed 2023-02-10 by themarpe

developer-mayuan, 2023-02-11:
> @kanoaellis
> 
> Sorry for the delay - already in, do you mind trying it out with: `shutdownResetDevice(True, True, True)` - CC: @Erol444 for documentation

Hi @themarpe : I couldn't find the API `shutdownResetDevice` dpethai-core code base, could you point us to the definition of it? Thanks!

themarpe, 2023-02-11:
Not documented at the moment. We'll document / modify the API to be simpler.

The function exsits on the device side (Script node) only

kanoaellis, 2023-02-13:
I tried `shutdownResetDevice(True, True, True)` and it gave me:
```
NameError: name 'shutdownResetDevice' is not defined
F: [global] [    881600] [Scheduler00Thr] dispatcherResponseServe:925
no request for this response: XLINK_READ_REL_RESP 1
```

I also then tried `node.shutdownResetDevice(True, True, True)` and that gave me:
```
AttributeError: 'lpb.Node' object has no attribute 'shutdownResetDevice'
```

themarpe, 2023-02-13:
@kanoaellis 

Sorry for missing out this - its on: `pipeline.shutdownResetDevice(True, True, True)`

developer-mayuan, 2023-02-16:
> @kanoaellis
> 
> Sorry for missing out this - its on: `pipeline.shutdownResetDevice(True, True, True)`

Hi @themarpe :
You mentioned this API only exists on the device side, so how should people call it? Through a script node in the pipeline? It will be great if you can provide a minimal example code to show how to use this API correctly, thanks!

kanoaellis, 2023-02-16:
I was able to get it working in the script node and in standalone mode.
Something like
```py
script = pipeline.createScript()
script.setScript("""
pipeline.shutdownResetDevice(True, True, True)
""")
```
when creating your pipeline.