luxonis/depthai-core #723: [BUG] Install from source dependencies error 22

ladrians, 2023-02-08:
Steps from manual installation detail error.

Running the command on Noetic

```
sudo wget -qO- https://raw.githubusercontent.com/luxonis/depthai-ros/main/install_dependencies.sh | sudo bash
```
the following error appears, the connection to Internet is OK

```
-- Compiling depthai-core resources in PATCH_ONLY mode
-- Downloading Depthai device side binaries from server...
-- Downloading depthai and patch
-- commit: 32b30ef42cbe32137e3274031a049c401c46a2c1
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
-- Couldn't check if depthai-shared codebase matches between device and host
-- Downloading and checking depthai-device-fwp.tar.xz
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
-- Status error: 22;"HTTP response code said error"
Couldn't download depthai-device-fwp.tar.xz
-- Resource not found, check if commit hash is correctly specified.
CMake Error at cmake/DepthaiDownloader.cmake:223 (message):
  Aborting.
Call Stack (most recent call first):
  CMakeLists.txt:342 (DepthaiDownload)
-- Configuring incomplete, errors occurred!
See also "/tmp/depthai-core/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/depthai-core/build/CMakeFiles/CMakeError.log".
```

attaching the associated log files

ladrians, 2023-02-09:
Complementing the information, I want to use the `depthai_ros_driver` option to tweak configuration parameters and use dynamic reconfigure. This is the error detail when trying to use some of the launch files on the `depthai_ros_driver` package

```
roslaunch depthai_ros_driver camera.launch

 * /rosdistro: noetic
 * /rosversion: 1.15.15
...
[ INFO] [1675935171.631517865]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1675935171.636857779]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1675935171.685777203]: Initializing nodelet with 12 worker threads.
[ INFO] [1675935171.700341967]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ INFO] [1675935172.847187288]: No ip/mxid specified, connecting to the next available device.
[ INFO] [1675935174.873595734]: Camera 19443010510B781300 connected!
[ INFO] [1675935174.874524056]: USB SPEED: SUPER
[ INFO] [1675935174.898479333]: Device type: OAK-D-PRO-W
[ INFO] [1675935174.904308529]: Pipeline type: RGBD
[ INFO] [1675935175.055508239]: NN Family: mobilenet
[ INFO] [1675935175.204826697]: Finished setting up pipeline.
[FATAL] [1675935176.293352203]: Failed to load nodelet '/oak` of type `depthai_ros_driver/Camera` to manager `oak_nodelet_manager'
[oak_nodelet_manager-3] process has died [pid 7550, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager /nodelet_manager/load_nodelet:=oak/nodelet_manager/load_nodelet /nodelet_manager/unload_nodelet:=oak/nodelet_manager/unload_nodelet /nodelet_manager/list:=oak/nodelet_manager/list __name:=oak_nodelet_manager __log:=/home/jetson/.ros/log/b83fbd4c-a85c-11ed-a425-48b02da4ca1e/oak_nodelet_manager-3.log].
log file: /home/jetson/.ros/log/b83fbd4c-a85c-11ed-a425-48b02da4ca1e/oak_nodelet_manager-3*.log
```

If I try to use the samples another error occurs

```
roslaunch depthai_examples rgb_publisher.launch

setting /run_id to 77069a0c-a85d-11ed-a177-48b02da4ca1e
process[rosout-1]: started with pid [13470]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [13473]
process[rgb_publisher-3]: started with pid [13478]
================================================================================REQUIRED process [rgb_publisher-3] has died!
process has died [pid 13478, exit code -11, cmd /home/jetson/catkin_ws/devel/lib/depthai_examples/rgb_node __name:=rgb_publisher __log:=/home/jetson/.ros/log/77069a0c-a85d-11ed-a177-48b02da4ca1e/rgb_publisher-3.log].
log file: /home/jetson/.ros/log/77069a0c-a85d-11ed-a177-48b02da4ca1e/rgb_publisher-3*.log
Initiating shutdown!
================================================================================
```

there is no useful data on the log files.

themarpe, 2023-02-10:
@ladrians 
Hi, do you mind retrying, could be that the server was down at that moment.

Other comment would be more suitable for DepthAI ROS repository

ladrians, 2023-02-10:
great! now the process finished so I could continue, thanks!

For the other issue I will continue trying and if cannot make it work will report it on the depthai_ros_driver repo.

thanks again.

closed 2023-02-10 by ladrians