luxonis/depthai-core #728: [BUG] {std::bad_alloc exception in dai::getLazyTarXzFunction when StereoDepth node is used }

PFrieling, 2023-02-20:
Hello, i am using an OAK-FFC 4P with an OAK-SoM Pro S3. I have two OV9282 cams attached. I have written a minimal program with just two MonoCameras, a StereoDepth and an XLinkOut node. The program crashes after 10-150 disparity frames with an std::bad_alloc expetion inside the dai::getLazyTarXzFunction function. I am not able to run it without any crash.

I am using the latest rvc3_support branch (commit [3793172](https://github.com/luxonis/depthai-core/commit/37931723ca523f6ebdd8c7f85269847b483e9643)). The same code runs on my OAK-FFC 3P with an OAK S2 and depthai-core v2.20.2.

This is the code:
```cpp
#include <iostream>
#include <depthai/depthai.hpp>

int main()
{
    dai::Pipeline pipeline;
    auto monoLeft = pipeline.create<dai::node::MonoCamera>();
    auto monoRight = pipeline.create<dai::node::MonoCamera>();
    monoLeft->setBoardSocket(dai::CameraBoardSocket::CAM_B);
    monoLeft->setResolution(dai::MonoCameraProperties::SensorResolution::THE_800_P);
    monoRight->setBoardSocket(dai::CameraBoardSocket::CAM_C);
    monoRight->setResolution(dai::MonoCameraProperties::SensorResolution::THE_800_P);

    auto depth = pipeline.create<dai::node::StereoDepth>();
    depth->setDefaultProfilePreset(dai::node::StereoDepth::PresetMode::HIGH_ACCURACY);
    depth->initialConfig.setMedianFilter(dai::MedianFilter::KERNEL_7x7);
    depth->setLeftRightCheck(false);
    depth->setExtendedDisparity(false);
    depth->setSubpixel(false);

    monoLeft->out.link(depth->left);
    monoRight->out.link(depth->right);

    auto xoutLeft = pipeline.create<dai::node::XLinkOut>();
    xoutLeft->setStreamName("depth");
    depth->disparity.link(xoutLeft->input);

    dai::Device device(pipeline, dai::UsbSpeed::SUPER);
    auto qLeft = device.getOutputQueue("depth", 2, false);

    std::cout << "Spinning around!" << std::endl;
    int cnt = 0;
    while(true)
    {
        auto camImage = qLeft->get<dai::ImgFrame>();
        if(camImage == nullptr) continue;

        std::cout << "Its a frame! " << ++cnt << std::endl;
    }

    return 0;
}
```

This is the output:
```
[2023-02-20 08:08:39.377] [debug] Library information - version: 2.19.1, commit: 37931723ca523f6ebdd8c7f85269847b483e9643 from 2023-02-03 00:01:12 +0100, build: 2023-02-17 13:41:46 +0000
[2023-02-20 08:08:39.382] [debug] Initialize - finished
[2023-02-20 08:08:39.605] [debug] Device - OpenVINO version: 2022.1
[2023-02-20 08:08:39.605] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0f b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be be bb 00
[2023-02-20 08:08:39.615] [debug] OS version: 1.2, Gate version: 0.0.10
[2023-02-20 08:08:39.615] [debug] DeviceGate createSession: {"fwp_version":"0.0.1+696421aaa5a47fe03df3b695854d6dd112b30927","library_version":"2.19.1","name":"depthai_session","protected":true}
[2023-02-20 08:08:39.627] [debug] DeviceGate createSession response: {"connection_information":"","create_datetime":"2022-07-29T15:27:01.277241","destroy_datetime":null,"fwp_checksum":null,"fwp_exists":false,"fwp_version":"0.0.1+696421aaa5a47fe03df3b695854d6dd112b30927","id":"13","library_version":"2.19.1","name":"depthai_session","owner":"","protected":true,"start_datetime":null,"state":"CREATED","stop_datetime":null}
[2023-02-20 08:08:39.779] [debug] Resources - Archive 'depthai-device-fwp-b511dc1a23aa9016007f7d6e9979a5a322bcefbe.tar.xz' open: 5ms, archive read: 395ms
[2023-02-20 08:08:40.556] [debug] DeviceGate upload fwp successful
[2023-02-20 08:08:41.063] [debug] DeviceGate start fwp successful
[2023-02-20 08:08:41.479] [debug] Schema dump: {"connections":[{"node1Id":2,"node1Output":"disparity","node1OutputGroup":"","node2Id":3,"node2Input":"in","node2InputGroup":""},{"node1Id":1,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"right","node2InputGroup":""},{"node1Id":0,"node1Output":"out","node1OutputGroup":"","node2Id":2,"node2Input":"left","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[3,{"alias":"","id":3,"ioInfo":[[["","in"],{"blocking":true,"group":"","name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","parentId":-1,"properties":{"maxFpsLimit":-1.0,"metadataOnly":false,"streamName":"depth"}}],[2,{"alias":"","id":2,"ioInfo":[[["","debugDispLrCheckIt2"],{"blocking":false,"group":"","name":"debugDispLrCheckIt2","queueSize":8,"type":0,"waitForMessage":false}],[["","debugExtDispLrCheckIt2"],{"blocking":false,"group":"","name":"debugExtDispLrCheckIt2","queueSize":8,"type":0,"waitForMessage":false}],[["","outConfig"],{"blocking":false,"group":"","name":"outConfig","queueSize":8,"type":0,"waitForMessage":false}],[["","confidenceMap"],{"blocking":false,"group":"","name":"confidenceMap","queueSize":8,"type":0,"waitForMessage":false}],[["","debugDispCostDump"],{"blocking":false,"group":"","name":"debugDispCostDump","queueSize":8,"type":0,"waitForMessage":false}],[["","rectifiedRight"],{"blocking":false,"group":"","name":"rectifiedRight","queueSize":8,"type":0,"waitForMessage":false}],[["","depth"],{"blocking":false,"group":"","name":"depth","queueSize":8,"type":0,"waitForMessage":false}],[["","debugExtDispLrCheckIt1"],{"blocking":false,"group":"","name":"debugExtDispLrCheckIt1","queueSize":8,"type":0,"waitForMessage":false}],[["","debugDispLrCheckIt1"],{"blocking":false,"group":"","name":"debugDispLrCheckIt1","queueSize":8,"type":0,"waitForMessage":false}],[["","rectifiedLeft"],{"blocking":false,"group":"","name":"rectifiedLeft","queueSize":8,"type":0,"waitForMessage":false}],[["","syncedRight"],{"blocking":false,"group":"","name":"syncedRight","queueSize":8,"type":0,"waitForMessage":false}],[["","syncedLeft"],{"blocking":false,"group":"","name":"syncedLeft","queueSize":8,"type":0,"waitForMessage":false}],[["","pixelDescriptors"],{"blocking":false,"group":"","name":"pixelDescriptors","queueSize":8,"type":0,"waitForMessage":false}],[["","disparity"],{"blocking":false,"group":"","name":"disparity","queueSize":8,"type":0,"waitForMessage":false}],[["","inputConfig"],{"blocking":false,"group":"","name":"inputConfig","queueSize":4,"type":3,"waitForMessage":false}],[["","left"],{"blocking":false,"group":"","name":"left","queueSize":8,"type":3,"waitForMessage":true}],[["","right"],{"blocking":false,"group":"","name":"right","queueSize":8,"type":3,"waitForMessage":true}]],"name":"StereoDepth","parentId":-1,"properties":{"depthAlignCamera":-1,"enableRectification":true,"enableRuntimeStereoModeSwitch":false,"focalLengthFromCalibration":true,"height":null,"initialConfig":{"algorithmControl":{"customDepthUnitMultiplier":1000.0,"depthAlign":1,"depthUnit":2,"disparityShift":0,"enableExtended":false,"enableLeftRightCheck":false,"enableSubpixel":false,"leftRightCheckThreshold":5,"subpixelFractionalBits":5},"censusTransform":{"enableMeanMode":true,"kernelMask":0,"kernelSize":-1,"threshold":0},"costAggregation":{"divisionFactor":1,"horizontalPenaltyCostP1":250,"horizontalPenaltyCostP2":500,"verticalPenaltyCostP1":250,"verticalPenaltyCostP2":500},"costMatching":{"confidenceThreshold":200,"disparityWidth":1,"enableCompanding":false,"invalidDisparityValue":0,"linearEquationParameters":{"alpha":0,"beta":2,"threshold":127}},"postProcessing":{"bilateralSigmaValue":0,"decimationFilter":{"decimationFactor":1,"decimationMode":0},"median":7,"spatialFilter":{"alpha":0.5,"delta":0,"enable":false,"holeFillingRadius":2,"numIterations":1},"speckleFilter":{"enable":false,"speckleRange":50},"temporalFilter":{"alpha":0.4000000059604645,"delta":0,"enable":false,"persistencyMode":3},"thresholdFilter":{"maxRange":65535,"minRange":0}}},"mesh":{"meshLeftUri":"","meshRightUri":"","meshSize":null,"stepHeight":16,"stepWidth":16},"numFramesPool":3,"numPostProcessingMemorySlices":-1,"numPostProcessingShaves":-1,"outHeight":null,"outKeepAspectRatio":true,"outWidth":null,"rectifyEdgeFillColor":0,"useHomographyRectification":null,"verticalStereo":false,"width":null}}],[1,{"alia [... cut ...]
[2023-02-20 08:08:41.479] [debug] Asset map dump: {"map":{}}
Spinning around!
[1B52B016838860C6] [192.168.197.55] [1659108425.440] [StereoDepth(2)] [warning] No calibration data provided!
Its a frame! 1
Its a frame! 2
Its a frame! 3
Its a frame! 4
Its a frame! 5
Its a frame! 6
Its a frame! 7
Its a frame! 8
Its a frame! 9
Its a frame! 10
Its a frame! 11
Its a frame! 12
Its a frame! 13
Its a frame! 14
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

themarpe, 2023-02-21:
@PFrieling thanks - we'll take a look

PFrieling, 2023-03-22:
@themarpe hi, do you have any updates? If there is something i can test to help you, it would not cause any trouble for me.

PFrieling, 2023-03-22:
Sorry, i did not mention to close the issue. Wrong button... 😃

PFrieling, 2023-03-27:
I did try the `rvc3_develop` branch in debug mode and it does not crash anymore, but after 11 up to 15 depth frames it sometimes stopps receiving more, without an error (feels similar to where it crashed before). 

I did some testing: If i use the code above, it stopps working before i got 15 depth frames in 7 out of 20 cases. If i enable extended disparity it stopps working in 9 of 20 cases. If i enable left right check, it stopps in 10 out of 20 cases. If i enable both, it stopps in 15 of 20 cases. 
I did one more test: If i remove the StereoDepth node and stream both camera images directly, it also sometimes stopps after roughly 22 frames from both cams. My code is still running, but the DataOutputQueue stops returning valid ImgFrames.

Could it be my modified ffc-4p? It originally came with an rvc2 and i used Luxonis-Davids post on Discord to change some resistors, so that it is compatible with rvc3. Otherwise, maybe my `UC-789 Rev B` cams are not 100% compatible? By the way, i have tested this on windows 10 and ubuntu 22.04, both behave the same.

PFrieling, 2023-04-04:
I have updated to LuxonisOS 1.8 and fip-oak-ffc-4p-rvc3-1.3.0, pulled the latest commits on the rvc3_develop branch, but the issue still exists.

Erol444, 2023-05-24:
Hi @PFrieling , We have since updated the FFC-4P FIP (1.5.0) and also OS (1.9). Could you try again with these latest versions?

PFrieling, 2023-05-25:
@Erol444 Somehow I am unable to ssh into my FFC-4P using my Windows pc, so i switched to another pc that runs Ubuntu. The FFC-4P is now updated to FIP 1.5.0 and OS 1.9, I am using depthai-core rvc3_develop from 24. May 2023. I do now get between 92 and 237 frames with my example code. Afterwards it freezes and says: ping was missed, closing device connection.  I guess the std::runtime_error or std::bad_alloc is thown because I am still trying to get new frames, even if the device is closed.

In some attempts, i even get no frame at all and the ping was missed before the arrival of the first frame. In such cases, simply restarting my executable works.

Is it okay that I run the StereoDepth node without any extrinsic or intrinsic calibration data? On my other FFC-3P and RVC2 I only configure the StereoDepth node and upload a custom mesh for rectification, because i only need a disparity map and no depth.

PFrieling, 2023-05-25:
I finally found the solution! While no more frames were received, lsusb showed that the FFC-4P was missing, indicating that it did a reboot/reset. After adding an additional power supply, besides the USB 3.0 connection, i was able to run my example code without any problems. So it was only caused because my USB 3 ports were unable to provide enough current.

closed 2023-05-25 by PFrieling