luxonis/depthai-core #735: To update the UVC branch to support GPIO or serial communication through USB

awida1, 2023-02-27:


themarpe, 2023-02-27:
@awida1 do you mind creating an MRE such that we can take a look into this? CC: @Erol444

awida1, 2023-02-28:
This is minimal with UVC set up to insure that is being run for testing. Thanks

`import os
import sys
import time

import blobconverter

import depthai as dai

import cv2

if sys.version_info[0] < 3:
    raise Exception["Doesn't work with Py2"]

MJPEG = False

os.environ["DEPTHAI_LEVEL"] = "debug"

progressCalled = False

def main():

    def get_pipeline():
        pipeline = dai.Pipeline()

        # Define a source - color camera
        cam = pipeline.create(dai.node.ColorCamera)
        cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
        cam.setPreviewKeepAspectRatio(False)
        cam.setPreviewSize(300, 300)
        cam.setVideoSize(1920, 1080)
        cam.setInterleaved(False)
        cam.initialControl.setManualFocus(130)

        # Create MobileNet detection network
        mobilenet = pipeline.create(dai.node.MobileNetDetectionNetwork)
        mobilenet.setBlobPath(blobconverter.from_zoo(name="face-detection-retail-0004", shaves=5))
        mobilenet.setConfidenceThreshold(0.7)
        #        cam.preview.link(mobilenet.input)

        cam.preview.link(mobilenet.input)

        uvc = pipeline.createUVC()
        cam.video.link(uvc.input)

        # Script node
        script = pipeline.create(dai.node.Script)
        mobilenet.out.link(script.inputs['dets'])

        script.setScript(f"""
        import GPIO
        MX_PIN = 40

        ret = GPIO.setup(MX_PIN, GPIO.OUT, GPIO.PULL_DOWN)
        toggleVal = True

        while True:
          data = node.io['dets'].get()  # Wait for a message from the host computer

          node.warn('GPIO toggle: ' + str(toggleVal))
          toggleVal = not toggleVal
          ret = GPIO.write(MX_PIN, toggleVal)  # Toggle the GPIO
        """)

        return pipeline

    with dai.Device(get_pipeline()) as dev:
        print(f"Connection speed: {dev.getUsbSpeed()}")

        # Connect to device with pipeline

        dev.setLogLevel(dai.LogLevel.WARN)
        dev.setLogLevel(dai.LogLevel.WARN)

        time.sleep(0.5)

        while True:
            time.sleep(0.1)

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        sys.exit(0)
`

awida1, 2023-02-28:
`import os
import sys
import time

import blobconverter

import depthai as dai

import cv2

if sys.version_info[0] < 3:
    raise Exception["Doesn't work with Py2"]

MJPEG = False

os.environ["DEPTHAI_LEVEL"] = "debug"

progressCalled = False

def main():

    def get_pipeline():
        pipeline = dai.Pipeline()

        # Define a source - color camera
        cam = pipeline.create(dai.node.ColorCamera)
        cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
        cam.setPreviewKeepAspectRatio(False)
        cam.setPreviewSize(300, 300)
        cam.setVideoSize(1920, 1080)
        cam.setInterleaved(False)
        cam.initialControl.setManualFocus(130)

        # Create MobileNet detection network
        mobilenet = pipeline.create(dai.node.MobileNetDetectionNetwork)
        mobilenet.setBlobPath(blobconverter.from_zoo(name="face-detection-retail-0004", shaves=5))
        mobilenet.setConfidenceThreshold(0.7)
        #        cam.preview.link(mobilenet.input)

        cam.preview.link(mobilenet.input)

        uvc = pipeline.createUVC()
        cam.video.link(uvc.input)

        # Script node
        script = pipeline.create(dai.node.Script)
        mobilenet.out.link(script.inputs['dets'])

        script.setScript(f"""
        import GPIO
        MX_PIN = 40

        ret = GPIO.setup(MX_PIN, GPIO.OUT, GPIO.PULL_DOWN)
        toggleVal = True

        while True:
          data = node.io['dets'].get()  # Wait for a message from the host computer

          node.warn('GPIO toggle: ' + str(toggleVal))
          toggleVal = not toggleVal
          ret = GPIO.write(MX_PIN, toggleVal)  # Toggle the GPIO
        """)

        return pipeline

    with dai.Device(get_pipeline()) as dev:
        print(f"Connection speed: {dev.getUsbSpeed()}")

        # Connect to device with pipeline

        dev.setLogLevel(dai.LogLevel.WARN)
        dev.setLogLevel(dai.LogLevel.WARN)

        time.sleep(0.5)

        while True:
            time.sleep(0.1)

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        sys.exit(0)
`

awida1, 2023-02-28:
```
import os
import sys
import time

import blobconverter

import depthai as dai

import cv2

if sys.version_info[0] < 3:
    raise Exception["Doesn't work with Py2"]

MJPEG = False

os.environ["DEPTHAI_LEVEL"] = "debug"

progressCalled = False

def main():

    def get_pipeline():
        pipeline = dai.Pipeline()

        # Define a source - color camera
        cam = pipeline.create(dai.node.ColorCamera)
        cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
        cam.setPreviewKeepAspectRatio(False)
        cam.setPreviewSize(300, 300)
        cam.setVideoSize(1920, 1080)
        cam.setInterleaved(False)
        cam.initialControl.setManualFocus(130)

        # Create MobileNet detection network
        mobilenet = pipeline.create(dai.node.MobileNetDetectionNetwork)
        mobilenet.setBlobPath(blobconverter.from_zoo(name="face-detection-retail-0004", shaves=5))
        mobilenet.setConfidenceThreshold(0.7)
        #        cam.preview.link(mobilenet.input)

        cam.preview.link(mobilenet.input)

        uvc = pipeline.createUVC()
        cam.video.link(uvc.input)

        # Script node
        script = pipeline.create(dai.node.Script)
        mobilenet.out.link(script.inputs['dets'])

        script.setScript(f"""
        import GPIO
        MX_PIN = 40

        ret = GPIO.setup(MX_PIN, GPIO.OUT, GPIO.PULL_DOWN)
        toggleVal = True

        while True:
          data = node.io['dets'].get()  # Wait for a message from the host computer

          node.warn('GPIO toggle: ' + str(toggleVal))
          toggleVal = not toggleVal
          ret = GPIO.write(MX_PIN, toggleVal)  # Toggle the GPIO
        """)

        return pipeline

    with dai.Device(get_pipeline()) as dev:
        print(f"Connection speed: {dev.getUsbSpeed()}")

        # Connect to device with pipeline

        dev.setLogLevel(dai.LogLevel.WARN)
        dev.setLogLevel(dai.LogLevel.WARN)

        time.sleep(0.5)

        while True:
            time.sleep(0.1)

if __name__ == "__main__":
    try:
        main()
    except KeyboardInterrupt:
        sys.exit(0)

```

awida1, 2023-03-14:
Hello, has the GPIO been tested with the OAK SOM yet? Thanks!