luxonis/depthai-core #737: [BUG] {Circular Blur on AR0234 with Monochrome ISP}

chengguizi, 2023-03-01:
**Describe the bug**
Regardless of the AR0234 physical sensor is color or monochrome, when I use `MonoCamera`,  there is a strange circle of blurryness. Could you try and reproduce on your end?

**Minimal Reproducible Example**
With a OAK-D-LR plugged in run
```
#!/usr/bin/env python3

from pathlib import Path
import cv2
import depthai as dai
import time

# Create pipeline
pipeline = dai.Pipeline()

# Define source and output
monoRight = pipeline.create(dai.node.MonoCamera)
xoutRight = pipeline.create(dai.node.XLinkOut)

xoutRight.setStreamName("right")

# Properties
monoRight.setBoardSocket(dai.CameraBoardSocket.RIGHT)
monoRight.setResolution(dai.MonoCameraProperties.SensorResolution.THE_1200_P)

# Linking
monoRight.out.link(xoutRight.input)

# Connect to device and start pipeline
with dai.Device(pipeline) as device:

    # Output queue will be used to get the grayscale frames from the output defined above
    qRight = device.getOutputQueue(name="right", maxSize=4, blocking=False)

    dirName = "mono_data"
    Path(dirName).mkdir(parents=True, exist_ok=True)

    while True:
        inRight = qRight.get()  # Blocking call, will wait until a new data has arrived
        # Data is originally represented as a flat 1D array, it needs to be converted into HxW form
        # Frame is transformed and ready to be shown
        cv2.imshow("right", inRight.getCvFrame())

        # After showing the frame, it's being stored inside a target directory as a PNG image
        cv2.imwrite(f"{dirName}/{int(time.time() * 1000)}.png", inRight.getFrame())

        if cv2.waitKey(1000) == ord('q'):
            break

```

**Expected behavior**
No artifacts at the circular outer area

**Screenshots**

Running the code on a monochrome AR0234
[image]

Running the code on a color AR0234
[image]

Erol444, 2023-04-06:
@alex-luxonis do you think there's a camera tuning problem happening here?

chengguizi, 2023-04-21:
@Erol444 any updates yet?

alex-luxonis, 2023-05-02:
@chengguizi Sorry for the delay here...
These appear to be a workaround:
```
cam[c].initialControl.setLumaDenoise(0)
cam[c].initialControl.setSharpness(0)
```
I added those at pipeline build step in `depthai-python/utilities/cam_test.py` and tested with an AR0234 color camera, configured to mono: `python3 utilities/cam_test.py -cams rgb,m`

It's also possible to experiment changing the values at runtime with the keys:
- `0` to select `sharpness` control
- `[` to select `luma denoise` control
- `]` to select `chroma denoise` control (not applicable to MonoCamera)

and adjusting the values with `-`/`_` and `+`/`=`

Could you check if this is an acceptable workaround for you. We're also looking to add a proper fix.

chengguizi, 2023-05-04:
@alex-luxonis  It works!

Actually seems `setLumaDenoise(0)` already fix the issue. This might be a good enough workaround :)

Look forward for a proper fix.

chengguizi, 2023-11-06:
@alex-luxonis I have tested on the latest depthai-core release 2.23. The blue still existing without the workaround. May I know if there is a proper fix being worked on still? Thanks!