luxonis/depthai-core #739: [Feature-Request] Merge of custom branch to develop

chengguizi, 2023-03-01:
### Start with the `why`:
Currently we have a custom branch to deal with our custom board. It is possible to merge it back to develop branch.

### Move to the `what`:
To keep track of the progress through this issue, suggested by @Erol444 

### Move to the `how`:
Merge the branch `left_right_separate_i2c_eeprom` to `develop` during / before the next release

chengguizi, 2023-03-02:
@alex-luxonis 
I just discovered a likely bug of this `left_right_separate_i2c_eeprom` branch, compared to the `4cam_custom` previously.

In the previous branch, I could use cama as master (FrameSyncMode::OUTPUT) and camb,c,d as slave an stream in a hardware synchronised fashion.

However, in the current branch. I can't do so. The behavious is like the following:
1. If I use cama as the master, the pipeline could run, but camb, camc, camd will be very dim in brightness in the image streamed. The image quality of b,c,d are bad.
2. If I use non-cama as the master, the rest of the camera can stream normally. But cama is not streaming beyond the first image.

Any idea why this happens? Thanks!

chengguizi, 2023-03-03:
@Erol444 
I have found a bug working with the `left_right_separate_i2c_eeprom` and the corresponding `develop` commit `40cb005c433078fe832291aed8e28727a09d2615` dated in Jan 29,2023.

For OV9282 x 2 configuration, on CAMA and CAMD, with hardware sync enabled, cama being the master: CAMD image will be exceptionally dark, under the same manual exposure setting. I am testing on a OAK-FFC-4P.

This bug seems no longer their on the latest `develop` commit, But I cant use that on our custom hardware due to this branch not merging.

Would it be possible to accelerate the merging, basically this single commit `36089a87bb5ef3d79fecf6deecf05c17a3e746af` to develop? Really appreciate!

Erol444, 2023-03-03:
@themarpe any ETA on this?

themarpe, 2023-03-05:
@chengguizi @Erol444 

Likely within this week

chengguizi, 2023-03-10:
@themarpe Are we still looking at this week? Really look forward, as I see `devel` branch is having good feature improvements / fixes, particularly on IMU! Thanks.

Erol444, 2023-03-16:
@themarpe when could we merge this?

szabi-luxonis, 2023-03-16:
Merged the custom branch into develop.

chengguizi, 2023-03-17:
Thanks! @szabi-luxonis  it seems to fix the main issues I have about IMU timestamp jitter, and the hardware sync image being too dark issue. Will test more on this. Close this for now, unless new issues pop up, thanks!

closed 2023-03-17 by chengguizi