luxonis/depthai-core #742: [BUG] Timestamp of frame event is behind host current monotonic time, impossibility

chengguizi, 2023-03-02:
**Describe the bug**
I am currently exploring the use of FrameEvent message. It nice as it has low latency, as it is being sent out at the time that exposure stops and data starts streaming (do correct me if I am wrong)

However, the timestamp the FrameEvent message gives it strange. Sometimes the timestamp indicates a future time on the host monotonic clock. This is by a couple of millisecond

**Minimal Reproducible Example**
Reproducable on a OAK-D-LR, with hardware sync from GPIO enabled
```
#!/usr/bin/env python3

# import cv2
import depthai as dai
import time
import datetime

enable_fsin = True
fps = 15
exposure = 8000

def configureColorAR0234(socket, streamName, frameEvent = False):
    
    # Define source and output
    cam = pipeline.create(dai.node.ColorCamera)
    cam.setBoardSocket(socket)
    xoutVideo = pipeline.create(dai.node.XLinkOut)

    xoutVideo.setStreamName(streamName)

    cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1200_P)
    # cam.setVideoSize(1920, 1200)
    # camA.initialControl.setAutoExposureEnable()
    cam.initialControl.setManualExposure(exposure, 200)

    cam.setFps(fps)

    xoutVideo.input.setBlocking(False)
    xoutVideo.input.setQueueSize(3)

    # Linking
    cam.isp.link(xoutVideo.input)

    if enable_fsin:
        # camA.initialControl.setFrameSyncMode(dai.CameraControl.FrameSyncMode.INPUT)
        cam.initialControl.setExternalTrigger(1,0)

    if frameEvent:
        xoutFrameEvent = pipeline.create(dai.node.XLinkOut)
        xoutFrameEvent.setStreamName("frame_event")
        xoutFrameEvent.input.setBlocking(False)
        xoutFrameEvent.input.setQueueSize(3)
        cam.frameEvent.link(xoutFrameEvent.input)

# Create pipeline
pipeline = dai.Pipeline()

configureColorAR0234(dai.CameraBoardSocket.CAM_A, "video_cama", True)
configureColorAR0234(dai.CameraBoardSocket.CAM_B, "video_camb")
configureColorAR0234(dai.CameraBoardSocket.CAM_C, "video_camc")

script = pipeline.create(dai.node.Script)
script.setProcessor(dai.ProcessorType.LEON_CSS)
script.setScript("""
    import GPIO
    import time

    MX_PIN = 41
    activeLevel = GPIO.HIGH
    timeActive = 0.001
    period = 0.1
    GPIO.setup(MX_PIN, GPIO.OUT, GPIO.PULL_DOWN)

    while True:
        time.sleep( period - timeActive )
        GPIO.write(MX_PIN, activeLevel)
        time.sleep( timeActive )
        GPIO.write(MX_PIN, not activeLevel)
""")
                

with dai.Device() as device:
    device.startPipeline(pipeline)

    # device.setTimesync(datetime.timedelta(seconds=5) , 20, False)

    videoA = device.getOutputQueue(name="video_cama", maxSize=4, blocking=False)
    videoB = device.getOutputQueue(name="video_camb", maxSize=4, blocking=False)
    videoC = device.getOutputQueue(name="video_camc", maxSize=4, blocking=False)

    frameEvent = device.getOutputQueue(name="frame_event", maxSize=4, blocking=False)

    while True:

        videoAIn = videoA.tryGet()
        videoBIn = videoB.tryGet()
        videoCIn = videoC.tryGet()

        frameEventIn = frameEvent.tryGet()

        if frameEventIn is not None:
            frameTs = frameEventIn.getTimestamp().total_seconds()

            hostNow = time.monotonic()

            if (frameTs >= hostNow):
                print(f"warning: host time is ahead of frame time {(frameTs - hostNow) * 1e3} ms, impossible (host = {hostNow}, frame ts = {frameTs})")

        # if videoAIn is not None:
        #     # cv2.imshow("camA", videoAIn.getCvFrame())
        #     print(f"camA {videoAIn.getTimestamp().total_seconds()}")

        # if videoBIn is not None:
        #     # cv2.imshow("camB", videoBIn.getCvFrame())
        #     print(f"camB {videoBIn.getTimestamp().total_seconds()}")

        # if videoCIn is not None:
        #     # cv2.imshow("camC", videoCIn.getCvFrame())
        #     print(f"camC {videoCIn.getTimestamp().total_seconds()}")

        # if cv2.waitKey(1) == ord('q'):
        #     break

```

**Expected behavior**
The timestamp should be older than the current host monotonic clock.

**Screenshots**
```
warning: host time is ahead of frame time 2.3374679999506043 ms, impossible (host = 2322.667376532, frame ts = 2322.669714)
warning: host time is ahead of frame time 2.6387759999124683 ms, impossible (host = 2322.771066224, frame ts = 2322.773705)
warning: host time is ahead of frame time 2.4276489998555917 ms, impossible (host = 2322.875276351, frame ts = 2322.877704)
warning: host time is ahead of frame time 2.6331309995839547 ms, impossible (host = 2322.979070869, frame ts = 2322.981704)
warning: host time is ahead of frame time 2.591991999906895 ms, impossible (host = 2323.083114008, frame ts = 2323.085706)
warning: host time is ahead of frame time 2.4143000000549364 ms, impossible (host = 2323.1872887, frame ts = 2323.189703)
warning: host time is ahead of frame time 2.5047849999282334 ms, impossible (host = 2323.291200215, frame ts = 2323.293705)
warning: host time is ahead of frame time 2.7853550000145333 ms, impossible (host = 2323.394918645, frame ts = 2323.397704)
warning: host time is ahead of frame time 2.896051999869087 ms, impossible (host = 2323.498969948, frame ts = 2323.501866)
warning: host time is ahead of frame time 2.7746120003939723 ms, impossible (host = 2323.602928388, frame ts = 2323.605703)
warning: host time is ahead of frame time 2.8310480001891847 ms, impossible (host = 2323.706874952, frame ts = 2323.709706)
```

themarpe, 2023-03-16:
@chengguizi 

Using getTimestamp is currently vulnerable to such issue where time sync jumps in time with large offset into a wrong direction. This is current limitation of Timesync algorithm used. TBD to be properly resolved.

Also, time.monotonic is not the right clock to compare against. Use `dai.Clock.now()`

chengguizi, 2023-03-17:
Thanks, glad to know the issue is known. Really hope it can be reolved eventually, timesync is quite important when using VPU data with external controllers :)
What is the difference of the dai.Clock.now() from the system's monotonic clock though?

themarpe, 2023-03-17:
@chengguizi if its a usb based system, then this issue is less glaring, but it's present more on PoE devices.

I am actually not 100% sure on the differences of the underlying impl. But in one if our other cases we've had there was a difference between a Pythons time.monotonic vs c++s steady_clock

chengguizi, 2023-03-17:
Got it @chengguizi  Yes I am using C++ mainly. Python is only for debugging. Good to know you guys observe this undesirable discrepency of python monotonic vs C++ steady_clock. Will keep in mind.

I do use USB based system. but i still see quite a few millisecond of that slack in time estimation. Not sure the best way around, but is your current algorithm similar to this?

https://github.com/mavlink/mavros/blob/ros2/mavros/src/plugins/sys_time.cpp#L416

themarpe, 2023-03-17:
Yes, its similar to that - I'll see about giving that a try and have a branch with that timesync impl compared to the current one - perhaps it'll act better.
Maybe over the weekend if time will permit.

chengguizi, 2023-03-17:
Thanks! This one estimate both the alpha and beta (time scale). Also it has sigmoid function to help convergence. Look forward!

themarpe, 2023-03-19:
@chengguizi 

`timesync_alt_impl` - you'll have to build the wheels/lib locally.

Please give it a try & any kind of exploration using it - validation/comparisions/... as well as with PoE devices, would be greatly appreciated, to aid in faster mainlining if it turns out to be much better:)

chengguizi, 2023-04-06:
The fix doesnt seem to work well. I actually seen time regression happens even on image timestamps (the next image in the sequence have earlier ts). This means the time sync correction might be making jumps.

Btw I am doing the following:
- I tried both turning on and off IMU streaming
- I did `m_device->setTimesync(std::chrono::milliseconds(5000), 20, false);` during setup. Does it affect anything?