luxonis/depthai-core #744: [BUG] IMU firmware updater re-writes duplicate versions of firmware, no API to check versions, log spelling issue

diablodale, 2023-03-03:
The IMU firmware updater does not check for duplicate versions of firmware. It blindly writes the same version over itself.
The updater, EEPROM apis, IMU node, etc. do not have any capability to query for the versions or know new firmware is available.

### Setup

* depthai-core v2.20.2
* all platforms
* all build tools
* OAK-D

### Repro 1: enableFirmwareUpdate() blindly writes new firmware

`imu->enableFirmwareUpdate(true)` writes new firmware even if the IMU already has the same firmware

1. Set log level to INFO.
2. Run an app that uses IMU

```
[14442C10C1A1C6D200] [3.12] [0.871] [IMU(0)] [info] IMU product ID:
[14442C10C1A1C6D200] [3.12] [0.871] [IMU(0)] [info] Part 10004148 : Version 3.9.7 Build 224
[14442C10C1A1C6D200] [3.12] [0.871] [IMU(0)] [info] Part 10003606 : Version 1.7.3 Build 337
[14442C10C1A1C6D200] [3.12] [0.871] [IMU(0)] [info] Part 10004135 : Version 5.5.1 Build 159
[14442C10C1A1C6D200] [3.12] [0.871] [IMU(0)] [info] Part 10004149 : Version 5.1.11 Build 182
```

3. Write c++ app that uses `imu->enableFirmwareUpdate(true)`
4. Run that firmware app

```
[14442C10C1A1C6D200] [3.12] [0.684] [IMU(0)] [warning] Performing IMU firmware update, do NOT disconnect the device.
[14442C10C1A1C6D200] [3.12] [0.793] [IMU(0)] [warning] IMU firmware update status: 0%
[14442C10C1A1C6D200] [3.12] [4.021] [IMU(0)] [warning] IMU firmware update status: 10%
[14442C10C1A1C6D200] [3.12] [7.244] [IMU(0)] [warning] IMU firmware update status: 20%
[14442C10C1A1C6D200] [3.12] [10.470] [IMU(0)] [warning] IMU firmware update status: 30%
[14442C10C1A1C6D200] [3.12] [13.708] [IMU(0)] [warning] IMU firmware update status: 40%
[14442C10C1A1C6D200] [3.12] [16.932] [IMU(0)] [warning] IMU firmware update status: 50%
[14442C10C1A1C6D200] [3.12] [20.158] [IMU(0)] [warning] IMU firmware update status: 60%
[14442C10C1A1C6D200] [3.12] [23.381] [IMU(0)] [warning] IMU firmware update status: 70%
[14442C10C1A1C6D200] [3.12] [26.617] [IMU(0)] [warning] IMU firmware update status: 80%
[14442C10C1A1C6D200] [3.12] [29.844] [IMU(0)] [warning] IMU firmware update status: 90%
[14442C10C1A1C6D200] [3.12] [33.067] [IMU(0)] [warning] IMU firmware update status: 100%
[14442C10C1A1C6D200] [3.12] [33.067] [IMU(0)] [info] IMU firmware update succesful!
[14442C10C1A1C6D200] [3.12] [33.290] [IMU(0)] [info] IMU product ID:
[14442C10C1A1C6D200] [3.12] [33.290] [IMU(0)] [info] Part 10004563 : Version 3.9.9 Build 2
[14442C10C1A1C6D200] [3.12] [33.290] [IMU(0)] [info] Part 10003606 : Version 1.8.0 Build 338
[14442C10C1A1C6D200] [3.12] [33.290] [IMU(0)] [info] Part 10004135 : Version 5.5.3 Build 162
[14442C10C1A1C6D200] [3.12] [33.290] [IMU(0)] [info] Part 10004149 : Version 5.1.12 Build 183
```

You can see that over ~30 seconds depthai-core wrote new IMU firmware.

5. Run the same updater app again

```
[14442C10C1A1C6D200] [3.12] [0.670] [IMU(0)] [warning] Performing IMU firmware update, do NOT disconnect the device.
[14442C10C1A1C6D200] [3.12] [0.779] [IMU(0)] [warning] IMU firmware update status: 0%
[14442C10C1A1C6D200] [3.12] [4.003] [IMU(0)] [warning] IMU firmware update status: 10%
[14442C10C1A1C6D200] [3.12] [7.226] [IMU(0)] [warning] IMU firmware update status: 20%
[14442C10C1A1C6D200] [3.12] [10.450] [IMU(0)] [warning] IMU firmware update status: 30%
[14442C10C1A1C6D200] [3.12] [13.683] [IMU(0)] [warning] IMU firmware update status: 40%
[14442C10C1A1C6D200] [3.12] [16.906] [IMU(0)] [warning] IMU firmware update status: 50%
[14442C10C1A1C6D200] [3.12] [20.129] [IMU(0)] [warning] IMU firmware update status: 60%
[14442C10C1A1C6D200] [3.12] [23.352] [IMU(0)] [warning] IMU firmware update status: 70%
[14442C10C1A1C6D200] [3.12] [26.585] [IMU(0)] [warning] IMU firmware update status: 80%
[14442C10C1A1C6D200] [3.12] [29.809] [IMU(0)] [warning] IMU firmware update status: 90%
[14442C10C1A1C6D200] [3.12] [33.031] [IMU(0)] [warning] IMU firmware update status: 100%
[14442C10C1A1C6D200] [3.12] [33.031] [IMU(0)] [info] IMU firmware update succesful!
[14442C10C1A1C6D200] [3.12] [33.256] [IMU(0)] [info] IMU product ID:
[14442C10C1A1C6D200] [3.12] [33.256] [IMU(0)] [info] Part 10004563 : Version 3.9.9 Build 2
[14442C10C1A1C6D200] [3.12] [33.256] [IMU(0)] [info] Part 10003606 : Version 1.8.0 Build 338
[14442C10C1A1C6D200] [3.12] [33.256] [IMU(0)] [info] Part 10004135 : Version 5.5.3 Build 162
[14442C10C1A1C6D200] [3.12] [33.256] [IMU(0)] [info] Part 10004149 : Version 5.1.12 Build 183
```

You can see that over ~30 seconds depthai-core did work on the IMU firmware. Was it checksumming, or did it write the same firmware back over itself? I suspect the latter. Something deep in depthai knows IMU versions since they can be seen in INFO logs. The updater codebase is not checking those versions before writing to firmware. This 2nd update should be sub-second and do no work.

### Repro 2: No API to get IMU firmware version or know it is outdated

I see no API to call into the IMU or EEPROM to get the current IMU versions and then to be able to compare it to the "current" IMU firmware stored within the depthai-core blob of the executable.
Alternatively, I do not see a flag somewhere like `bool EEPROM::ImuUpdateAvailable`.
Having either one of these, I could conditionally call `enableFirmwareUpdate(true)`.

### Repro 3: Spelling mistake in log entry. Request the mistake is forever.

Since there is no API to manage IMU versions, progress, etc., it is the LOG which provides this. Notice above there is a misspelling...

```
[14442C10C1A1C6D200] [3.12] [33.031] [IMU(0)] [info] IMU firmware update succesful!
```

`Successful` has 3 S's. Not 2 as in the log. I have need to track progress of firmware updates. Therefore I need stability in these strings. I request that the spelling never be corrected and the firmware codebase be annotated to reflect this so it isn't accidently changed in the future.

### Related

https://github.com/luxonis/depthai-core/issues/319
https://github.com/luxonis/depthai-core/issues/683
https://github.com/luxonis/depthai-core/issues/285

szabi-luxonis, 2023-03-03:
This isn't a BUG, it was intended behavior. 
enableFirmwareUpdate performs firmware update, no matter what. If it would have been otherwise, then documentation would have mentioned it.

+1 on querying IMU version, it's in the pipeline, together with querying available firmware version for update.

diablodale, 2023-03-03:
Got it. Does that mean it is technically possible for the IMU units to DOWNGRADE their firmware? That isn't a guaranteed thing for all IMUs. 🤔 If all the IMUs you incorporate into OAK units do always support DOWNGRADE, then I can imagine a possibility for depthai-core to bundle an OLDER imu firmware. And therefore `enableFirmwareUpdate()` would downgrade the firmware.

Most firmware units have a limited number of writes to their persistent storage. They are consumables. I would not want to consume one of those writes to burn the same version firmware. I also suspect the firmware is checksumed and verified. So there is no realistic scenario for corruption. I don't see any scenario to support same version blind writing.

If imu hardware supports downgrading, I see value for the depthai firmware to allow writing firmware for any situation where they are NOT the same. But if they are the same, then NOOP.

szabi-luxonis, 2023-03-06:
IMU version, firmware version, firmware update status RPCs were added, see the related linked PR.
API is subject to change prior to merge. Feel free to suggest. Main blockers were resolved, it's easy to add utility functions now.
Will add NOOP if current IMU version is same as the latest available.
Downgrading IMU firmware is possible and intended.

szabi-luxonis, 2023-03-07:
NOOP was added IF firmware versions match, otherwise no. Downgrade is still possible and intended. Debating internally whether we put this behind an optional flag or not. Mainly for testing purposes.

diablodale, 2023-03-07:
Hi. I do have feedback. I'm currently in hospital and my phone is a bit difficult to fully review. I'll try to do some now directly on the PRs. I request you wait a few days so I can get back to a full screen and fully review

szabi-luxonis, 2023-03-09:
Get well soon. 
The release will be made later this month, so there is still time to change API etc. before release, if you have anything else feel free to bring it up.
For now, this issue is considered resolved in https://github.com/luxonis/depthai-core/commit/60e7ee56361efa7e50cd42081f0249cac898202e

closed 2023-03-09 by szabi-luxonis