luxonis/depthai-core #749: [emgucv cmake error] {internal error with toolchain.cmake}

UmonDaisuke, 2023-03-04:
Hello,
during cmake configuration  (cmake 3.26) of emgucv 4.6.0 on windows, I always have the  following error

Using specified toolchain file: C:/EMGU2/emgucv/build_x86_64/Emgu.CV.Extern/depthai-core/generated/toolchain.cmake combined into: C:/EMGU2/emgucv/build_x86_64/Emgu.CV.Extern/depthai-core/generated/toolchain.cmake
CMake Error at build_x86_64/Emgu.CV.Extern/depthai-core/generated/toolchain.cmake:48 (message):
  Internal error
Call Stack (most recent call first):
  build_x86_64/CMakeFiles/3.26.0-rc5/CMakeSystem.cmake:6 (include)
  Emgu.CV.Extern/depthai-core/CMakeLists.txt:36 (project)

Thanks you for your help
Michael

themarpe, 2023-03-05:
Hi @UmonDaisuke 

Do you mind clearing Hunter cache (as per readme) and removing the build directory and try again?

Also, do you mind copying the `generated/toolchain.cmake` and other relevant files?

UmonDaisuke, 2023-03-09:
Hello,
sorry for delay. I cleared the hunter cache but it hasn't changed anything. What do you mean with "as per readme"? please find enclosed files that generated errors  
[CMakeSystem and toolhain.zip](https://github.com/luxonis/depthai-core/files/10928828/CMakeSystem.and.toolhain.zip)
Thansk a lot for support
Michael

netcorefan1, 2023-03-28:
I'm afraid you're encountering the same issue I describe in details [here](https://github.com/luxonis/depthai-core/issues/788).
There are some possible workarounds, depending from your need to use DEPTHAI and/or GPU.
Unless you didn't specified a different target, you're probably building Master branch which default to version 4.7. 
With OpenCv 5.0 as pre-release, I would avoid to use 4.6