luxonis/depthai-core #750: [Feature-Request] Documentation about  Android Emulator usb passtrough.

onuralpszr, 2023-03-05:
### Start with the `why`:
Normally in order to debug/test our android application we need to connect camera to phone and connect phone over wifi or use some sort of dongle to split the data for making work. This process little bit complex and can be simplify. 

### Move to the `what`:
- Depthai-core readme/doc needs to show "how to pass-through oak-d cameras via usb  devices into android emulator." Android emulators are basically using qemu with custom abilities for android development.

Useful links and Sources 

https://android.googlesource.com/platform/external/qemu/+/2db80f7c1921a6f5d48b998378e3792e16c968a4/README.md#:~:text=The%20Android%20Emulator%20is%20downstream,extends%20qemu%20in%20various%20ways.

https://source.android.com/docs/devices/automotive/start/passthrough

 
### Move to the `how`:
I tried many different ways to see usb camera I was able to added into android emualtor by running proper pass through comands but I was unable see "permission" but I was able to see MyriadX permission for device on my phone. 

I was trying via "android.hardware.usb" API (https://developer.android.com/guide/topics/connectivity/usb/host) reach devices and get a permission but I was also unsuccessful.

themarpe, 2023-03-05:
Hi @onuralpszr 

This is out of scope as is, but this issue can remain as a discussion point on any findings on this topic.

As a quick tip, there is DepthAI Android support but limited to specific point in time due to being an not too widely used, which could be tried in emulator. One example: https://github.com/ibaiGorordo/depthai-android-jni-example

onuralpszr, 2023-03-05:
> Hi @onuralpszr
>
> This is out of scope as is, but this issue can remain as a discussion point on any findings on this topic.
>
> As a quick tip, there is DepthAI Android support but limited to specific [...]

I already know this repo plus I even changed this into submodule way to compile latest version of depthai-core inside android ndk project with cmake and gradle and experimenting with kotlin language. And this repo can be use emulator like the one I have but I am stıll having the same problem it needs some how way to send usb pass into emulator but If I can find it. I would like to open PR into readme so people can do easier devs as well. Mobile also fun :)

onuralpszr, 2023-03-05:
> Hi @onuralpszr
>
> This is out of scope as is, but this issue can remain as a discussion point on any findings on this topic.
>
> As a quick tip, there is DepthAI Android support but limited to specific [...]

One side question also can we just use depthai-core repo as it is so use in android without adding extra libusb repo(luxonis fork/ cmake android branch) or do I still I need that ?

themarpe, 2023-03-05:
@onuralpszr 
"Custom" `libusb` is mandatory (only for CMake build system) & for the "Android" branch. Check the progress here however to getting both of those into mainline from libusb side: https://github.com/libusb/libusb/pull/1134#issuecomment-1454311419