luxonis/depthai-core #754: [BUG] cmake hangs on hunter_unpack_directory

aggiehorns, 2023-03-07:
I have been beating my head against the keyboard trying to get cmake to work.  I have traced it to hunter_unpack_directory.cmake line 143.  Expanded I can see the following.

`C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/cmake/modules/hunter_unpack_directory.cmake(143):  execute_process(COMMAND C:/Program Files/Microsoft Visual Studio/2022/Preview/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/bin/cmake.exe;-DHUNTER_INSTALL_PREFIX=C:/.hunter/_Base/cb0ea1f/04da027/7258ad3/Install;-DLIST_OF_FILES=C:/.hunter/_Base/Cellar/d43c7d359805f81df2468a0289529cc1f21fa918/d43c7d3/files.list;-DSHELL_LINK_SCRIPT=C:/.hunter/_Base/Cellar/d43c7d359805f81df2468a0289529cc1f21fa918/d43c7d3/link-all.sh;-DCELLAR_RAW_DIRECTORY=C:/.hunter/_Base/Cellar/d43c7d359805f81df2468a0289529cc1f21fa918/d43c7d3/raw;-DPYTHON_LINK_SCRIPT=C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/scripts/link-all.py;-P;C:/.hunter/_Base/Download/Hunter/0.23.322/cb0ea1f/Unpacked/scripts/link-all.cmake WORKING_DIRECTORY C:/.hunter/_Base/Cellar/d43c7d359805f81df2468a0289529cc1f21fa918/d43c7d3 RESULT_VARIABLE result OUTPUT_VARIABLE output ERROR_VARIABLE error )`

Manually running this command it appears things are stalling in `link-all.py`

This is fun.  Maybe doesn't like my default python version?

traversaro, 2024-05-13:
For anyone experiencing a similar problem of CMake hanging without any output, a good way to debug what is going on is to pass the `-DHUNTER_STATUS_DEBUG:BOOL=ON` CMake option.

In my case, I was getting the following output:
~~~
-- [hunter *** DEBUG *** 2024-05-13T09:31:43] Download error ("Failure when receiving data from the peer")
-- [hunter *** DEBUG *** 2024-05-13T09:31:43] Downloading DONE metafile (try #1 of 10):
-- [hunter *** DEBUG *** 2024-05-13T09:31:43]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:43]   -> /home/traversaro/.hunter/_Base/Cache/meta/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:43] Download error ("Failure when receiving data from the peer")
-- [hunter *** DEBUG *** 2024-05-13T09:31:43] Downloading DONE metafile (try #2 of 10):
-- [hunter *** DEBUG *** 2024-05-13T09:31:43]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:43]   -> /home/traversaro/.hunter/_Base/Cache/meta/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:43] Retry after 5 seconds (attempt #2) ...
-- [hunter *** DEBUG *** 2024-05-13T09:31:48] Download error ("Failure when receiving data from the peer")
-- [hunter *** DEBUG *** 2024-05-13T09:31:48] Downloading DONE metafile (try #3 of 10):
-- [hunter *** DEBUG *** 2024-05-13T09:31:48]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:48]   -> /home/traversaro/.hunter/_Base/Cache/meta/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:48] Retry after 5 seconds (attempt #3) ...
-- [hunter *** DEBUG *** 2024-05-13T09:31:53] Download error ("Failure when receiving data from the peer")
-- [hunter *** DEBUG *** 2024-05-13T09:31:53] Downloading DONE metafile (try #4 of 10):
-- [hunter *** DEBUG *** 2024-05-13T09:31:53]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:53]   -> /home/traversaro/.hunter/_Base/Cache/meta/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:31:53] Retry after 15 seconds (attempt #4) ...
-- [hunter *** DEBUG *** 2024-05-13T09:32:08] Download error ("Failure when receiving data from the peer")
-- [hunter *** DEBUG *** 2024-05-13T09:32:08] Downloading DONE metafile (try #5 of 10):
-- [hunter *** DEBUG *** 2024-05-13T09:32:08]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:32:08]   -> /home/traversaro/.hunter/_Base/Cache/meta/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:32:08] Retry after 60 seconds (attempt #5) ...
-- [hunter *** DEBUG *** 2024-05-13T09:33:08] Download error ("Failure when receiving data from the peer")
-- [hunter *** DEBUG *** 2024-05-13T09:33:08] Downloading DONE metafile (try #6 of 10):
-- [hunter *** DEBUG *** 2024-05-13T09:33:08]   https://raw.githubusercontent.com/cpp-pm/hunter-cache/master/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:33:08]   -> /home/traversaro/.hunter/_Base/Cache/meta/cdcf41c/nlohmann_json/3.9.1/f8a20a7/151cbba/a23c16c/da39a3e/basic-deps.DONE
-- [hunter *** DEBUG *** 2024-05-13T09:33:08] Retry after 90 seconds (attempt #6) ...
~~~

Basically, hunter was not finding some kind of cache, probably because I was using a fair recent compiler install via conda-forge. However, as I wanted to build everything from source, I also passed the `-DHUNTER_USE_CACHE_SERVERS:BOOL=NO` flag, and everything compiled fine.

mepix, 2024-09-28:
I ran into this issue on MacOS Sonoma 14.6.1 building `depthai-core-v2.28.0` release with cmake version 3.27.8.  The flag `-DHUNTER_USE_CACHE_SERVERS:BOOL=NO` did the trick, thanks!

I did not look into this further, but it does seem to be the `nlohmann_json` that causes hunter to hang.