luxonis/depthai-core #758: [BUG] OpenCV 4.x+ requires enabled C++11 support

jwinarske, 2023-03-11:
**Describe the bug**

Setup CMake with:
```
-DDEPTHAI_BUILD_EXAMPLES=ON
```

Building any of the example targets yields:
```
In file included from /Users/joel/development/depthai-core/examples/utility/utility.cpp:1:
In file included from /Users/joel/development/depthai-core/examples/utility/utility.hpp:3:
In file included from /usr/local/include/opencv4/opencv2/opencv.hpp:52:
In file included from /usr/local/include/opencv4/opencv2/core.hpp:52:
/usr/local/include/opencv4/opencv2/core/cvdef.h:666:4: error: "OpenCV 4.x+ requires enabled C++11 support"
#  error "OpenCV 4.x+ requires enabled C++11 support"
   ^
```

Regressing to OpenCV 3.x is not an acceptable solution for a number of reasons.

**Expected behavior**
OpenCV 4.x should be supported

themarpe, 2023-03-12:
@jwinarske 
Which CMake version do you have?

newer than v3.8.0 should propagate the requirement of C++14 down to linkee of the library (eg the examples).

`cpp14_examples_tests` branch should address this. Let me know

closed 2023-03-12 by themarpe

jwinarske, 2023-03-12:
Closer

Apriltag
```
/Users/joel/development/depthai-core-upstream/include/depthai/device/DeviceBase.hpp:34:10: fatal error: 'depthai-shared/device/CrashDump.hpp' file not found
#include "depthai-shared/device/CrashDump.hpp"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

depth_preview
```
/Users/joel/development/depthai-core-upstream/include/depthai/device/DeviceBase.hpp:34:10: fatal error: 'depthai-shared/device/CrashDump.hpp' file not found
#include "depthai-shared/device/CrashDump.hpp"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

jwinarske, 2023-03-12:
@themarpe nevermind.  
```
git submodule update --init --recursive
```
did the trick.

themarpe, 2023-03-12:
Great, thanks for confirmation!

diablodale, 2023-03-17:
@jwinarske the PR fix causes other problem. To better understand, I ask you provide much more setup detail.
What is your
* specific os. if you are windows type `ver`. If you are linux try looking withing /etc/os-release
* specific depthai-core version you are building. The commit ID is best.
* specific compiler toolchain you are using. This can widely vary with env vars, os alternatives, cmake vars, etc. So please be sure to get the exact compiler and linker being used
* specific cmake version. please type `cmake --version`

Alternatively, if you have a docker container with your setup, then point us to it.

jwinarske, 2023-03-17:
@diablodale this was on a Ventura M1 Max Mac.  Latest brew packages.  I haven't checked on my daily driver - Fedora 37.

In general there needs to be a CI matrix to validate this kind of change, otherwise it's a game of whack a mole...

jwinarske, 2023-03-17:
@diablodale
stable | ✅ | 4.7.0
Revision: 2
4.7.0_2
https://formulae.brew.sh/formula/opencv#default

diablodale, 2023-03-17:
Great. And all the rest?
* specific os. if you are windows type ver. If you are linux try looking withing /etc/os-release
* specific depthai-core version you are building. The commit ID is best.
* specific compiler toolchain you are using. This can widely vary with env vars, os alternatives, cmake vars, etc. So please be sure to get the exact compiler and linker being used
* specific cmake version. please type cmake --version

jwinarske, 2023-03-17:
The repro involved:
* OS: Mac Ventura (arm64)
* DepthAI-core: v2.20.2
* Compiler: AppleClang 14.0.0
* CMake 3.24.2

I am definitely not blocked, and I would consider this a low priority item.  I have since made it build elsewhere.

diablodale, 2023-03-18:
Got it. One last set of information. What are the reproduction steps to follow so we can reproduce the error?
Meaning, please list the exact command-line(s) given so that we understand how you setup your environment, what arguments/parameters you passed to cmake, etc. As above, I don't know if this is debug/release, if shared libs/not, and so many more possibilities.

We should be able to copy/paste the steps you provide into our command prompt and the see the compiler error. So please give us everything.

jwinarske, 2023-03-18:
That was last week, I was in the airport in a 5 hour layover, and I really don't remember.  I would make a note of it, and move on.  Note that I am working on a fairly large refactor of the CMake, and it will implicitly support this combo.

Hunter replacement proof here: https://github.com/jwinarske/cmake_json/

diablodale, 2023-03-19:
OpenCV 4 is well supported in depthai across Linux and Windows. Regularily tested. Personally, I write Windows apps using my own fork of Depthai-core and OpenCV 4.5.x for ~two years. I must have compiled 50+ times this past week as I coded.

I suspect your error was something due to toolchain setup, configuration, or maybe something specific to MacM1. Luxonis does support Mac, but perhaps there is something recently broken by Apple or MacM1 ecosystem that hasn't yet surfaced with many customers so to be identified.

While Hunter (and similarily VCPKG) are sometimes a pain, there is a need they both serve. When your idea is ready, I suggest you approach Luxonis. I've found they are open to considering innovation/ideas. 👍