luxonis/depthai-core #759: `setWarpTransformMatrix3x3` does not appear to have an effect

abroekhof, 2023-03-11:
I am trying to use `setWarpTransformMatrix3x3` to do a perspective transform. I've generated the transform matrix using OpenCV, and tested that it is working as expected. However, `setWarpTransformMatrix3x3` does not appear to make any changes to the input frame when I apply it. I have had success with `setWarpMesh`, though.

In the MWE below I set the matrix using the `initialConfig`, but I have also tried passing in config messages during runtime, still without effect. Also, I saw the note about width needing to be a multiple of 16 to work properly, and ensured that was the case.

MWE:
```
import cv2
import depthai as dai

camRes = dai.ColorCameraProperties.SensorResolution.THE_1080_P
camSocket = dai.CameraBoardSocket.RGB
ispScale = (1,2)

def create_pipeline(calibData):
    pipeline = dai.Pipeline()

    cam = pipeline.create(dai.node.ColorCamera)
    cam.setIspScale(ispScale)
    cam.setBoardSocket(camSocket)
    cam.setResolution(camRes)

    manip = pipeline.create(dai.node.ImageManip)
    manip.initialConfig.setWarpTransformMatrix3x3([0.45368620037807184,
                                                   0.13112723256676884,
                                                   -84.83931947069944,
                                                   -1.179324042755353e-16,
                                                   1.013957103163357,
                                                   0.0,
                                                   0.0,
                                                   0.0005668155774944997,
                                                   1.0])
    manip.setMaxOutputFrameSize(cam.getIspWidth() * cam.getIspHeight() * 3 // 2)
    cam.isp.link(manip.inputImage)

    cam_xout = pipeline.create(dai.node.XLinkOut)
    cam_xout.setStreamName("Undistorted")
    manip.out.link(cam_xout.input)

    dist_xout = pipeline.create(dai.node.XLinkOut)
    dist_xout.setStreamName("Distorted")
    cam.isp.link(dist_xout.input)

    return pipeline

with dai.Device() as device:

    calibData = device.readCalibration()
    pipeline = create_pipeline(calibData)
    device.startPipeline(pipeline)

    queues = [device.getOutputQueue(name, 4, False) for name in ['Undistorted', 'Distorted']]

    while True:
        for q in queues:
            frame = q.get().getCvFrame()
            cv2.imshow(q.getName(), frame)

        if cv2.waitKey(1) == ord('q'):
            break
```

Logs
```
[2023-03-11 08:01:16.193] [debug] Python bindings - version: 2.20.2.0.dev+d8dfcdbe7bebf7ad7016030eaa73f2193bd1faae from 2023-03-08 13:49:34 +0100 build: 2023-03-08 12:59:46 +0000
[2023-03-11 08:01:16.193] [debug] Library information - version: 2.20.2, commit: 4ff860838726a5e8ac0cbe59128c58a8f6143c6c from 2023-01-31 22:20:03 +0200, build: 2023-03-08 12:59:39 +0000
[2023-03-11 08:01:16.210] [debug] Initialize - finished
[2023-03-11 08:01:16.279] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 2ms, archive read: 82ms
[2023-03-11 08:01:16.419] [debug] Device - OpenVINO version: universal
[2023-03-11 08:01:16.419] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
[... 19 lines cut ...]
[18443010C1BF631200] [20.2] [3.109] [system] [info] Cpu Usage - LeonOS 19.75%, LeonRT: 3.24%
[18443010C1BF631200] [20.2] [4.110] [system] [info] Memory Usage - DDR: 12.76 / 340.42 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 21.98 / 77.32 MiB, LeonRT Heap: 3.72 / 41.23 MiB
[18443010C1BF631200] [20.2] [4.110] [system] [info] Temperatures - Average: 29.10 °C, CSS: 30.32 °C, MSS 28.13 °C, UPA: 29.35 °C, DSS: 28.62 °C
[18443010C1BF631200] [20.2] [4.110] [system] [info] Cpu Usage - LeonOS 21.70%, LeonRT: 3.36%
[18443010C1BF631200] [20.2] [5.111] [system] [info] Memory Usage - DDR: 12.76 / 340.42 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 21.98 / 77.32 MiB, LeonRT Heap: 3.72 / 41.23 MiB
[18443010C1BF631200] [20.2] [5.111] [system] [info] Temperatures - Average: 29.16 °C, CSS: 30.56 °C, MSS 28.62 °C, UPA: 28.86 °C, DSS: 28.62 °C
[18443010C1BF631200] [20.2] [5.111] [system] [info] Cpu Usage - LeonOS 21.65%, LeonRT: 3.35%
[18443010C1BF631200] [20.2] [6.112] [system] [info] Memory Usage - DDR: 12.76 / 340.42 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 21.98 / 77.32 MiB, LeonRT Heap: 3.72 / 41.23 MiB
[18443010C1BF631200] [20.2] [6.112] [system] [info] Temperatures - Average: 29.53 °C, CSS: 31.53 °C, MSS 29.10 °C, UPA: 28.86 °C, DSS: 28.62 °C
[18443010C1BF631200] [20.2] [6.112] [system] [info] Cpu Usage - LeonOS 21.86%, LeonRT: 3.37%
[18443010C1BF631200] [20.2] [7.113] [system] [info] Memory Usage - DDR: 12.76 / 340.42 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 21.98 / 77.32 MiB, LeonRT Heap: 3.72 / 41.23 MiB
[18443010C1BF631200] [20.2] [7.113] [system] [info] Temperatures - Average: 29.65 °C, CSS: 30.32 °C, MSS 29.35 °C, UPA: 29.83 °C, DSS: 29.10 °C
[18443010C1BF631200] [20.2] [7.113] [system] [info] Cpu Usage - LeonOS 21.65%, LeonRT: 3.27%
[18443010C1BF631200] [20.2] [8.114] [system] [info] Memory Usage - DDR: 12.76 / 340.42 MiB, CMX: 2.48 / 2.50 MiB, LeonOS Heap: 21.98 / 77.32 MiB, LeonRT Heap: 3.72 / 41.23 MiB
[18443010C1BF631200] [20.2] [8.114] [system] [info] Temperatures - Average: 30.14 °C, CSS: 31.53 °C, MSS 29.59 °C, UPA: 29.35 °C, DSS: 30.08 °C
[18443010C1BF631200] [20.2] [8.114] [system] [info] Cpu Usage - LeonOS 22.25%, LeonRT: 3.44%
[2023-03-11 08:01:25.961] [debug] Device about to be closed...
[2023-03-11 08:01:25.966] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2023-03-11 08:01:25.966] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[2023-03-11 08:01:25.966] [debug] DataOutputQueue (Undistorted) closed
[2023-03-11 08:01:25.966] [debug] DataOutputQueue (Distorted) closed
[2023-03-11 08:01:26.528] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
[2023-03-11 08:01:28.040] [debug] XLinkResetRemote of linkId: (0)
[2023-03-11 08:01:28.040] [debug] Device closed, 2079
```

themarpe, 2023-03-16:
@abroekhof 

Thanks for the report - when reworkeming ImageManip, this wasn't taken into consideration. 
For now rely in mesh, but I'll take a look at either exposing this or moving away from this kind of API

abroekhof, 2023-03-17:
Got it, thanks for the update @themarpe

sferoze, 2023-10-17:
I am trying to use this function and am getting error:

Traceback (most recent call last):
  File "/Users/feroze/DepthAI/depthai-python/examples/Camera/4cam_lightglue_stitcher6.py", line 259, in <module>
    frame_left2 = in_right1.getCvFrame()
RuntimeError: ImgFrame doesn't have enough data to encode specified frame, required 1335386880, actual 3110400. Maybe metadataOnly transfer was made?

asahtik, 2023-11-06:
Hi @sferoze, could you provide a [minimal reproducible example](https://en.wikipedia.org/wiki/Minimal_reproducible_example) so that I can look into this?