luxonis/depthai-core #763: [BUG] Cam Test Fails on OAK-FFP-3P with 2x AR0234 and IMX378

kneave, 2023-03-14:
**Check if issue already exists**

I have found no reference by searching online or the forums

**Describe the bug**
I have an OAK-FFP-3P with the following configuration:
Left: AR0234
Centre: IMX378
Right: AR0234

When I try to run cam_test.py using the develop branch of depthai-python (which includes depthai-core) it fails. The AR0234 are also not detected. Note, if I switch to the ar0234_x3 branch, all the cameras are detected.

Image showing only the IMX378 detected when using the `develop` branch:
[image]

Image showing the same script reporting all three cameras, running from the `ar0234_x3` branch:
[image]

**Minimal Reproducible Example**
```
python cam_test.py -cams rgb,c left,c right,c            

Enabled cameras:
    rgb : color
   left : color
  right : color
DepthAI version: 2.20.2.0
DepthAI path: C:\Users\keegan\Documents\GitHub\depthai-python\venv\lib\site-packages\depthai.cp39-win_amd64.pyd
Traceback (most recent call last):
  File "C:\Users\keegan\Documents\GitHub\depthai-python\utilities\cam_test.py", line 181, in <module>
    cam[c].setIsp3aFps(args.isp3afps)
AttributeError: 'depthai.node.ColorCamera' object has no attribute 'setIsp3aFps'
```

**Expected behavior**
I would expect cam_test.py to succeed and be able to stream all cameras.

**Additional context**
I have attached the simple preview script I've been using to test the cameras on each branch.
[preview.zip](https://github.com/luxonis/depthai-core/files/10972042/preview.zip)

szabi-luxonis, 2023-03-14:
When you switch to a branch, you also need to run `examples/install_requirements.py` to install the library specific to that branch.
You are using latest release, which is older than what's available on develop.

kneave, 2023-03-14:
I ran `examples\install_requirements.py` before running the above test, I just deleted and grabbed a fresh copy from git and the same holds true:

[image]

szabi-luxonis, 2023-03-14:
If you download as `.zip` it won't work, since it relies on git versioning to retrieve firmware version.
You need to clone it, together with all its submodules.

kneave, 2023-03-14:
For reference:

[image]

szabi-luxonis, 2023-03-14:
What's the output of `python3 install_requirements.py` ?

kneave, 2023-03-14:
Great minds, I was just grabbing that for you!

This is from `examples\install_requirements.py` [image]

szabi-luxonis, 2023-03-14:
Some of these commands fail:
https://github.com/luxonis/depthai-python/blob/develop/examples/install_requirements.py#L114
It's something related to your environment on windows.
Would you mind adding some debug prints and check? 
E.g.
```
    except (OSError, subprocess.CalledProcessError) as e:
        print(e)
        git_context = False
```

kneave, 2023-03-14:
You're right, I'm on a new laptop and it turns out git isn't accessible in PyCharm for some reason as it was on my old one. I ran `install_requirements.py` from the PyCharm git-bash and it updated correctly though.

The isp error has gone, the AR0234 are still not detected however.

[image]