luxonis/depthai-core #765: Command line warning D9025 : overriding '/std:c++17' with '/std:c++14'

diablodale, 2023-03-14:
Noticed that `develop` is not compiling the tests correctly with different C++ standards.
Throwing warnings on each one that is not C++14.
This is new and was not present in much earlier releases. They correctly compiled for 17, 20, etc.

### Setup

* Windows 11, Microsoft Windows [Version 10.0.22621.1344]
* VS2019 v16.11.24
* depthai-core `develop` at ddca0b29f672a88a86c2aec9876fef5e33535114

### Repro

1. Config and build for Windows x64, Release, shared libs, no opencv support, yes build tests

### Result

```
...
[build] [91/114  68% :: 63.621] Building CXX object tests\CMakeFiles\filesystem_test_17.dir\src\filesystem_test.cpp.obj
[build] cl : Command line warning D9025 : overriding '/std:c++17' with '/std:c++14'
[build] [92/114  69% :: 63.958] Building CXX object tests\CMakeFiles\filesystem_test_20.dir\src\filesystem_test.cpp.obj
[build] cl : Command line warning D9025 : overriding '/std:c++20' with '/std:c++14'
...
[build] [105/114  85% :: 98.066] Building CXX object tests\CMakeFiles\device_usbspeed_test_17.dir\src\device_usbspeed_test.cpp.obj
[build] cl : Command line warning D9025 : overriding '/std:c++17' with '/std:c++14'
[build] [106/114  85% :: 101.897] Building CXX object tests\CMakeFiles\device_usbspeed_test_20.dir\src\device_usbspeed_test.cpp.obj
[build] cl : Command line warning D9025 : overriding '/std:c++20' with '/std:c++14'
...
[build] [108/114  88% :: 107.073] Building CXX object tests\CMakeFiles\device_usbspeed_test_17_conforming.dir\src\device_usbspeed_test.cpp.obj
[build] cl : Command line warning D9025 : overriding '/std:c++17' with '/std:c++14'
[build] [109/114  89% :: 107.125] Linking CXX executable tests\device_usbspeed_test_20.exe
[build]    Creating library tests\device_usbspeed_test_20.lib and object tests\device_usbspeed_test_20.exp
...
[build] [113/114  95% :: 115.701] Building CXX object tests\CMakeFiles\device_usbspeed_test_20_conforming.dir\src\device_usbspeed_test.cpp.obj
[build] cl : Command line warning D9025 : overriding '/std:c++20' with '/std:c++14'
...
```

### Expected

No warnings. It should compile 14, 17, or 20 as intended.

### Notes

`compile_commands.json` has the following for one of the above failure EXEs `device_usbspeed_test_17.exe`

```json
{
  "directory": "C:/njs/depthai-core/build",
  "command": "C:\\PROGRA~2\\MICROS~2\\2019\\COMMUN~1\\VC\\Tools\\MSVC\\1429~1.301\\bin\\Hostx64\\x64\\cl.exe  /nologo /TP -DDEPTHAI_TARGET_CORE -DJSON_USE_IMPLICIT_CONVERSIONS=1 -Ddevice_usbspeed_test_17_EXPORTS -Ivcpkg_installed\\x64-windows-static-md-v142-sdk10\\include -I..\\include -I..\\shared\\depthai-shared\\include -I..\\shared\\depthai-bootloader-shared\\include -I..\\shared\\depthai-shared\\3rdparty -IC:\\.hunter\\_Base\\cb0ea1f\\b775583\\7d645e3\\Install\\include -DWINVER=0x0A00 -D_WIN32_WINNT=0x0A00 -DNTDDI_VERSION=0x0A000005 /arch:AVX /Gw /QIntel-jcc-erratum /guard:cf /GS  -D_CRT_SECURE_NO_WARNINGS /wd5030 /w15240 -D_CRT_SECURE_NO_WARNINGS /wd5030 /w15240   /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG -DNOMINMAX -DWIN32_LEAN_AND_MEAN -std:c++17 -std:c++14 /Fotests\\CMakeFiles\\device_usbspeed_test_17.dir\\src\\device_usbspeed_test.cpp.obj /FdTARGET_COMPILE_PDB /FS -c C:\\njs\\depthai-core\\tests\\src\\device_usbspeed_test.cpp",
  "file": "C:/njs/depthai-core/tests/src/device_usbspeed_test.cpp"
}
```

Notice the flags `-std:c++17 -std:c++14` and their order.

themarpe, 2023-03-16:
This is likely due to this: https://github.com/luxonis/depthai-core/pull/760

Will have to recheck & apply in different way

diablodale, 2023-03-17:
That PR doesn't look correct. The default c++14 is already applied with
https://github.com/luxonis/depthai-core/blob/c2b96c81f11bbd3625cbd18c634c60a969b675c3/tests/CMakeLists.txt#L31-L33

Perhaps whatever in OpenCV is throwing that error is hardcoded to look at `PROPERTY CXX_STANDARD`. I think I remember you or I intentionally didn't use that and instead used
https://github.com/luxonis/depthai-core/blob/c2b96c81f11bbd3625cbd18c634c60a969b675c3/tests/CMakeLists.txt#L61

themarpe, 2023-03-17:
https://github.com/luxonis/depthai-core/issues/758#issuecomment-1465259774

Rechecked the above as well, even if it was sub 3.8.0 the generic lambdas interface property should still set min cpp14.

--

In tests case its already applied, but missing for examples. It should pick it from core, but perhaps same option as in tests could be added

diablodale, 2023-03-17:
I've asked the originator of #758 for setup details. Nothing useful yet.

#758 does not reproduce with

* docker Ubuntu 20.04, gcc 10, opencv 4.2.0
* docker Ubuntu 22.04, clang15, opencv 4.5.4
* Windows, vs2019, opencv 4.5.4

If I were a maintainer, I would revert that PR until a repro setup/case can be created.

diablodale, 2023-03-21:
@themarpe, I suggest the PR be reverted. The originatorof #758 can't remember what they did or recreate it. So there is no customer needing a fix.

themarpe, 2023-03-23:
A revert was pushed - the tests part was reverted as they already specify a version, the examples were left as nothing sets the version there explicitly and shouldn't clash (latest `develop`)

diablodale, 2023-03-23:
LGTM. I compiled `develop` at 23e86d6955a364c3c0544c6b926a44a2c178d61d as Debug

* no related compiler errors
* `compile_commands.json` has no line with two or more `c++` substrings

diablodale, 2023-04-14:
depthai-core v2.21.0 includes the fix PR

closed 2023-04-14 by diablodale