luxonis/depthai-core #768: `develop` firmware commit breaks test examples

diablodale, 2023-03-16:
The most recent commit to `develop` https://github.com/luxonis/depthai-core/commit/e5dae8fd2f9e543bd99f9d1a73959398ef5bd0b4 breaks several test examples.

Rolling back to `set(DEPTHAI_DEVICE_SIDE_COMMIT "bd54decd22f0913323b95b547dbd435ec12d4fc7")` fixes the broken test examples.

### Setup

* Microsoft Windows 11 [Version 10.0.22621.1413]
* MSVC v16.11.24
* depthai-core `develop` at e5dae8fd2f9e543bd99f9d1a73959398ef5bd0b4
* depthai-shared `develop` at 5988f9629d9c04ba48fa938d62443d49416c6d35

### Repro

1. config and build for: windows x64, debug, shared libs, opencv support, build tests and examples, test tests and examples
2. Run ctest

### Result

multiple test/examples fail. One of them is `examples/calibration_load.exe`

```
[14442C10C1A1C6D200] [3.12] [0.746] [system] [info] Memory Usage - DDR: 0.12 / 340.34 MiB, CMX: 2.04 / 2.50 MiB, LeonOS Heap: 7.51 / 77.23 MiB, LeonRT Heap: 2.89 / 41.14 MiB
[14442C10C1A1C6D200] [3.12] [0.746] [system] [info] Temperatures - Average: 31.83 ┬░C, CSS: 33.45 ┬░C, MSS 31.29 ┬░C, UPA: 31.04 ┬░C, DSS: 31.53 ┬░C
[14442C10C1A1C6D200] [3.12] [0.746] [system] [info] Cpu Usage - LeonOS 62.73%, LeonRT: 1.89%
[14442C10C1A1C6D200] [3.12] [0.806] [MonoCamera(0)] [info] Using board socket: 1, id: 1
[14442C10C1A1C6D200] [3.12] [0.806] [MonoCamera(1)] [info] Using board socket: 2, id: 2
[14442C10C1A1C6D200] [3.12] [0.807] [system] [info] SIPP (Signal Image Processing Pipeline) internal buffer size '18432'B, DMA buffer size: '16384'B
[14442C10C1A1C6D200] [3.12] [0.843] [system] [info] MonoCamera allocated resources: no shaves; cmx slices: [13-15] 
StereoDepth allocated resources: shaves: [14-14] cmx slices: [12-12] 

[14442C10C1A1C6D200] [3.12] [0.962] [StereoDepth(2)] [error] Stereo focal length is 0. This is likely due to wrong calibration intrinsics. Can be set explicitly using 'setFocalLength' API.
[14442C10C1A1C6D200] [3.12] [1.747] [system] [info] Memory Usage - DDR: 26.99 / 340.34 MiB, CMX: 2.09 / 2.50 MiB, LeonOS Heap: 33.75 / 77.23 MiB, LeonRT Heap: 1.48 / 41.14 MiB
[14442C10C1A1C6D200] [3.12] [1.747] [system] [info] Temperatures - Average: 32.55 ┬░C, CSS: 33.69 ┬░C, MSS 32.49 ┬░C, UPA: 31.77 ┬░C, DSS: 32.25 ┬░C
[14442C10C1A1C6D200] [3.12] [1.747] [system] [info] Cpu Usage - LeonOS 20.77%, LeonRT: 8.06%
F: [global] [         0] [ThreadN] dispatcherResponseServe:925  no request for this response: XLINK_WRITE_RESP 1
```

Notice the focal error. Later the `queue->get()` throws and therefore the app crashes.

### Expected

All test and examples work correctly.

### Workaround

Revert to `set(DEPTHAI_DEVICE_SIDE_COMMIT "bd54decd22f0913323b95b547dbd435ec12d4fc7")`

szabi-luxonis, 2023-03-16:
Thanks for catching this. Our automated tests have been broken for a while, so it wasn't caught.
The fix will land in ~15 mins.

szabi-luxonis, 2023-03-16:
Done in: https://github.com/luxonis/depthai-core/commit/0c3d6be3b3574867cc99d1ad85c3f75eade40f58

closed 2023-03-16 by szabi-luxonis