luxonis/depthai-core #780: [BUG] depthai-core doesn't compile in debian unstable with gcc-12

lepalom, 2023-03-22:
Trying to build depthai-core, it failed:

```
/usr/include/spdlog/spdlog.h:145:32:   required from ‘void spdlog::trace(fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::chrono::duration<long int, std::ratio<1, 1000000> >, long unsigned int, dai::DatatypeEnum&, details::dump_info<__gnu_cxx::__normal_iterator<const unsigned char*, std::vector<unsigned char, std::allocator<unsigned char> > > >}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::chrono::duration<long int, std::ratio<1, 1000000> >, long unsigned int, dai::DatatypeEnum&, details::dump_info<__gnu_cxx::__normal_iterator<const unsigned char*, std::vector<unsigned char, std::allocator<unsigned char> > > > >]’
/<<PKGBUILDDIR>>/src/device/DataQueue.cpp:199:34:   required from here
/usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1757 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1757:7: note: ‘formattable’ evaluates to false
make[3]: *** [CMakeFiles/depthai-core.dir/build.make:177: CMakeFiles/depthai-core.dir/src/device/DataQueue.cpp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
```

the complete log is here:

http://debomatic-amd64.debian.net/distribution#unstable/depthai-core/2.20.2-1/buildlog

themarpe, 2023-03-22:
@lepalom I assume this is using a custom compilation environment with a system / custom spdlog library?

There is likely a versioning mismatch. You'll need to update spdlog to be within the ballpark version of what DepthAI uses

themarpe, 2023-03-23:
@lepalom likely yup - we have "custom logging bindings" relate to fmt library and not to standard C++ `format`

Perhaps a simpler solution is to modify that formatting call such that it also works with that spdlog version and create a PR for it, which we can then incorporate into DepthAI to make it more "spdlog version agnostic"

lepalom, 2023-03-23:
@themarpe From my point of view, to have any custom logging and not follow standard C++ is not a good idea ... but I'm more worried about another issues of depthai-core wrt the build and use process.

themarpe, 2023-03-23:
> From my point of view, to have any custom logging and not follow standard C++ is not a good idea

I mean, C++ didn't have a standard formatting to follow since C++20, which was why the fmtlib existed (and was then included in the standard). Its mostly just a difference between fmtlib being used or C++20 format being used. And the former is more "correct" actually as the C++20 format took away some capabilities of fmtlib. So one cannot just swap out fmtlib for C++20 format in all cases.

lepalom, 2023-03-23:
The question for me is if do you really "need" that capabilities of fmtlib that are not in the standard?

themarpe, 2023-03-24:
It was always part of spdlog, so in an essence, yes, we relied on it. spdlog+fmt lib was the standard, if recent version changed that, then downstream dependencies must either: switch the behavior off if possible, or update to accommodate.

For us, fmt lib is "the standard" at the moment (We dont use C++20 format) 

I'll close this, as compiling against different version of libs fall to the user. Note any PRs that expand the compatibility and address this are however appreciated

closed 2023-03-24 by themarpe