luxonis/depthai-core #781: How to chain vcpkg toolchain?

netcorefan1, 2023-03-23:
Hello,
I'm compiling a library which uses depthai-core as toolchain:
`-- Using toolchain file: ../depthai-core/generated/toolchain.cmake`
I need to set other properties and I would like to know how I can chain `"C:\dev\vcpkg\scripts\buildsystems\vcpkg.cmake"` in order to get them automatically populated.
I tried to add `include(/path/to/vcpkg/scripts/buildsystems/vcpkg.cmake)` at the end of `../depthai-core/generated/toolchain.cmake` as stated from [docs](https://learn.microsoft.com/en-us/vcpkg/users/buildsystems/cmake-integration#using-multiple-toolchain-files), but the file is automatically generated and the `include` gets removed.

There must be a simple way to do this, but I have not been able to find any info.
Thanks

themarpe, 2023-03-27:
@netcorefan1 

Try setting `CMAKE_TOOLCHAIN_PATH` it should correctly take the given toolchain into account.

netcorefan1, 2023-03-28:
Thanks for your help. `CMAKE_TOOLCHAIN_FILE` can't be changed because is set to `../depthai-core/generated/toolchain.cmake`.
Changing to vcpkg cause to raise several `depthai-core/cmake/HunterGate.cmake` errors.
To complicate things further, DEPTHAI toolchain seems to have [problems](https://github.com/luxonis/depthai-core/issues/788) with CMake configurations.

netcorefan1, 2023-03-28:
I don't remember well, but during my attempts I got it to partially working by adding the include in `CMakeSystem.cmake` (or some other cmake file), just before depthai-core include, but there were some issues and I gave up.

Based on your response in my other discussion, I have understood that DEPTHAI is only intended for Luxonix devices meaning that it has no real usage for standard web cameras and/or web video sources. So, for my specific case the obvious choice is to disable integration with emgucv framework. `CMAKE_TOOLCHAIN_FILE` is now free and I am able to use vcpkg toolchain without problem.

I was very fascinated by the abilities of depthai and its devices.  Hope to have the chance to get one of such cameras in the future. I only hope that in the meantime the integration with frameworks will become [smoother and less complex](https://github.com/luxonis/depthai-core/issues/788). By giving a quick look in [this](https://github.com/luxonis/depthai-core/issues/72) discussion, @diablodale stated that  the combination of Hunter and a dev chosen CMAKE_TOOLCHAIN_FILE are incompatible. As as final user who don't know all the technical details, I can only say that vcpkg makes everything easier. So you may want to evaluate switching to it.

diablodale, 2023-03-28:
I get it. My solution was to fork depthai-core to make it, hunter, vcpkg, opencv, and my app to successfully build together. It was non-trivial and specific to my needs (so sharing my approach isn't helpful). I have my changes on a branch and rebase it on to any new depthai-core releases.

https://xkcd.com/927/ cartoon captures the problem well. There is no native C/C++ package manager. People make their own and continuously build new ones. For example, Microsoft leads both the vcpkg and onnxruntime projects. Yet, they chose to not use vcpkg and wrote a custom package manager for onnxruntime. 🤦‍♀️

netcorefan1, 2023-03-30:
@diablodale Thanks for your detailed explanation. Yes, I saw in this and other cases the same always occurring: when multiple toolchains are needed things becomes very problematic! Hope that one day the community can make something universal. may be one only toolchain with all the other ones integrated.

closed 2023-03-30 by netcorefan1