luxonis/depthai-core #782: [BUG] Calibration Handler fails with Oak-1 OpenCV kit (no Camera data available)

MAVProxyUser, 2023-03-24:
https://github.com/luxonis/depthai-core/blob/main/include/depthai/device/CalibrationHandler.hpp
https://github.com/luxonis/depthai-core/blob/main/src/device/CalibrationHandler.cpp
The above Calibration handler can't handle the blank name returned by [Original OpenCV AI Kit: Oak-1](https://store.opencv.ai/products/oak-1)

This results in:
```
what():  There is no Camera data available corresponding to the the requested cameraId
```

As manifest here:
https://github.com/luxonis/depthai-ros/pull/262
and here:
https://github.com/luxonis/depthai-ros/issues/257#issuecomment-1480627256

```
[component_container-1] [INFO] [1679549436.807773594] [oak]: Device type: 
```
Problem does not exist with OAK-1-MAX

saching13, 2023-03-26:
Can I know when this OAK-1 eas purchased ? the initial versions didn't come with Eeprom to store the calibration data for them.

MAVProxyUser, 2023-03-26:
@saching13 it was an open box item on Ebay. https://www.ebay.com/itm/295571240836
[image]

Side of the box barcode says: "A00109", box is labeled as "AI KIT". 

Regardless of when it was purchased, how can we work around this nuance?

saching13, 2023-03-26:
ok. 
can you try [this](https://docs.luxonis.com/projects/api/en/latest/samples/calibration/calibration_reader/#calibration-reader) and see what error it gives ? 
If it throws an error. the work around would be to do an external calibration and use those intrinsics by saving them on host.

MAVProxyUser, 2023-03-26:
@saching13 this fails *as expected*? Given there is no RIGTH camera. 
```
ubuntu@ubuntu:~$ python3 test.py 
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    intrinsics = calibData.getCameraIntrinsics(dai.CameraBoardSocket.RIGHT)
IndexError: _Map_base::at
```

MAVProxyUser, 2023-03-26:
@saching13 can you explain the workaround in context of the referenced problems, aka simply starting the ROS nodes to publish *any* topic related to the cameras.

MAVProxyUser, 2023-03-26:
lol I now see there was a more complete example at: https://github.com/luxonis/depthai-python/blob/main/examples/calibration/calibration_reader.py 

It returned this error: 
```
ubuntu@ubuntu:~$ python3 calibration_reader.py 
Traceback (most recent call last):
  File "calibration_reader.py", line 17, in <module>
    M_rgb, width, height = calibData.getDefaultIntrinsics(dai.CameraBoardSocket.RGB)
RuntimeError: There is no Camera data available corresponding to the the requested cameraId
```

MAVProxyUser, 2023-03-26:
For posterity, here is my OAK-1-MAX

```
ubuntu@ubuntu:~$ python3 calibration_reader.py Stored.json
RGB Camera Default intrinsics...
[[2986.73291015625, 0.0, 2004.2607421875], [0.0, 2986.73291015625, 1510.3992919921875], [0.0, 0.0, 1.0]]
4000
3000
RGB Camera Default intrinsics...
[[2986.73291015625, 0.0, 2004.2607421875], [0.0, 2986.73291015625, 1510.3992919921875], [0.0, 0.0, 1.0]]
4000
3000
RGB Camera resized intrinsics... 3840 x 2160 
[[2.86726343e+03 0.00000000e+00 1.92409021e+03]
 [0.00000000e+00 2.86726343e+03 1.08998328e+03]
 [0.00000000e+00 0.00000000e+00 1.00000000e+00]]
RGB Camera resized intrinsics... 4056 x 3040 
[[3.02854736e+03 0.00000000e+00 2.03232056e+03]
 [0.00000000e+00 3.02854736e+03 1.53054480e+03]
 [0.00000000e+00 0.00000000e+00 1.00000000e+00]]
Traceback (most recent call last):
  File "calibration_reader.py", line 53, in <module>
    M_left, width, height = calibData.getDefaultIntrinsics(dai.CameraBoardSocket.LEFT)
RuntimeError: There is no Camera data available corresponding to the the requested cameraId
```

and its JSON

```
ubuntu@ubuntu:~$ cat Stored.json 
{
    "batchName": "Maxwell",
    "batchTime": 1667974639,
    "boardConf": "nIR-C02M00-00",
    "boardCustom": "",
    "boardName": "NG9093",
    "boardOptions": 0,
    "boardRev": "R3M0E4",
    "cameraData": [
        [
            0,
            {
                "cameraType": 0,
                "distortionCoeff": [
                    -7.164836883544922,
                    20.304182052612305,
                    0.00017078577366191894,
                    -0.00011530541814863682,
                    -16.3474063873291,
                    -7.179931163787842,
                    20.356590270996094,
                    -16.410999298095703,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0,
                    0.0
                ],
                "extrinsics": {
                    "rotationMatrix": [
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ],
                        [
                            0.0,
                            0.0,
                            0.0
                        ]
                    ],
                    "specTranslation": {
                        "x": -0.0,
                        "y": -0.0,
                        "z": -0.0
                    },
                    "toCameraSocket": -1,
                    "translation": {
                        "x": 0.0,
                        "y": 0.0,
                        "z": 0.0
                    }
                },
                "height": 3000,
                "intrinsicMatrix": [
                    [
                        2986.73291015625,
                        0.0,
                        2004.2607421875
                    ],
                    [
                        0.0,
                        2986.73291015625,
                        1510.3992919921875
                    ],
                    [
                        0.0,
                        0.0,
                        1.0
                    ]
                ],
                "lensPosition": 130,
                "specHfovDeg": 68.7938003540039,
                "width": 4000
            }
        ]
    ],
    "hardwareConf": "F0-FV00-BC000",
    "imuExtrinsics": {
        "rotationMatrix": [
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ]
        ],
        "specTranslation": {
            "x": 0.0,
            "y": 0.0,
            "z": 0.0
        },
        "toCameraSocket": -1,
        "translation": {
            "x": 0.0,
            "y": 0.0,
            "z": 0.0
        }
    },
    "miscellaneousData": [],
    "productName": "OAK-1-MAX",
    "stereoRectificationData": {
        "leftCameraSocket": 255,
        "rectifiedRotationLeft": [
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ]
        ],
        "rectifiedRotationRight": [
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                0.0,
                0.0
            ]
        ],
        "rightCameraSocket": 255
    },
    "version": 7
}
```

saching13, 2023-03-28:
> @saching13 can you explain the workaround in context of the referenced problems, aka simply starting the ROS nodes to publish _any_ topic related to the cameras.

you can calibrate and save the json file on the host and pass it as a param to it. Adam can help on how to set that up.

MAVProxyUser, 2023-03-28:
@saching13 yeah, my goal would be to simply make it work transparently for anyone that happens to own said camera variant, and chooses to use the existing ROS nodes. In the mean time I simply bought a new camera.

saching13, 2023-03-28:
> my goal would be to simply make it work transparently for anyone that happens to own said camera variant, and chooses to use the existing ROS nodes.
I understand the concern. But since it is missing what we can do is override and publish a standard dummy camera info for that as default with a warning. Unless the user can calibrate and add an argument of that.

MAVProxyUser, 2023-03-28:
sounds completely reasonable! This is my first week with you all's code base so, forgive some of my uninformed questions. I appreciate the help.

MAVProxyUser, 2023-04-13:
@Serafadam I'm sadly stuck on Noetic on a few of my instances, will this be backported to the other branches as well?

Serafadam, 2023-09-14:
Hi @MAVProxyUser , sorry, this got lost among notifications, yes it was backported soon afterwards to Noetic.