luxonis/depthai-core #783: [BUG] OAK-FFC-6P IMU not detected

rozni, 2023-03-24:
When trying to read IMU data from the OAK-FFC-6P board, the IMU node fails and the pipeline halts. The following error is logged:

> [###] [192.168.197.55] [###] [IMU(0)] [error] Node threw exception, stopping the node. Exception message: IMU BMI270 not detected​

Run any of the IMU examples to reproduce.

szabi-luxonis, 2023-03-24:
@rozni use latest `rvc3_develop`

szabi-luxonis, 2023-04-06:
You need to update your FIP, if the board is very old:
1. Download [fip-oak-ffc-6p-rvc3-1.2.0.bin](https://luxonisos.fra1.digitaloceanspaces.com/release/recovery/fip-oak-ffc-6p-rvc3-1.2.0.bin)
2. Copy to device, `scp fip-oak-ffc-6p-rvc3-1.2.0.bin root@192.168.197.55:~`
3. On device: movisoc-fwu -a fip-oak-ffc-6p-rvc3-1.2.0.bin`

rozni, 2023-04-06:
It was flashed originally with that FIP but still no change even running the suggested steps explicitly.

The device was ordered in January and camera nodes, for example, work.

szabi-luxonis, 2023-04-06:
CC: @Erol444 on FFC-6P errata docs.

Erol444, 2023-05-05:
Hi @rozni , Could you try again with latest `rvc3_develop`?

rozni, 2023-05-05:
Hi, yes, just tried again and it still fails.

Some sample output (depthai-core):
```
$ git rev-parse HEAD
a6e4368c23f32fbc943ab6bc971cd1fde834a92a

$ ./build/examples/device_information 
Searching for all available devices...

Found device: 192.168.197.55 mxid: 40726016838860C6 state: 5

Booting the first available camera (192.168.197.55)...
Available camera sensors: 
terminate called after throwing an instance of 'dai::EepromError'
  what():  No or invalid EEPROM configuration flashed
Aborted (core dumped)

$ ./build/examples/imu_gyroscope_accelerometer 
[40726016838860C6] [192.168.197.55] [1659362205.657] [IMU(0)] [error] Node threw exception, stopping the node. Exception message: IMU not detected. Your board doesn't have IMU or requires firmware update. 0, 375

$ ./build/examples/imu_rotation_vector 
[40726016838860C6] [192.168.197.55] [1659362230.885] [IMU(0)] [error] Node threw exception, stopping the node. Exception message: IMU not detected. Your board doesn't have IMU or requires firmware update. 0, 375
```

And on depthai-python:
```
$ git rev-parse HEAD
434e5482de175558d4f0444291850c7d711e609c

$ python3 examples/Script/script_get_device_info.py 
{
    "deviceId": "serialtodo",
    "fwVersion": "0.0.1-861f7310a4658ee1b5891b02a2cbfdde8c81aeed"
}

$ python3 examples/IMU/imu_version.py 
[40726016838860C6] [192.168.197.55] [1659362363.007] [system] [error] No IMU version detected, but specified in DTS. Update FIP.
IMU type: NONE, firmware version: 0.0.0

$ python3 examples/IMU/imu_gyroscope_accelerometer.py 
[40726016838860C6] [192.168.197.55] [1659362385.976] [IMU(0)] [error] Node threw exception, stopping the node. Exception message: IMU not detected. Your board doesn't have IMU or requires firmware update. 0, 375

$ python3 examples/IMU/imu_rotation_vector.py 
[40726016838860C6] [192.168.197.55] [1659362421.207] [system] [error] No IMU version detected, but specified in DTS. Update FIP.
IMU type: NONE, firmware version: 0.0.0
Rotation vector output is supported only by BNO086!
```

There is LuxonisOS 1.8 running and the FIP version is 1.2.0.

Erol444, 2023-05-05:
@rozni it might be a HW issue. Let's continue our conversation on email.

closed 2023-05-05 by Erol444