luxonis/depthai-core #784: [BUG] {Title of the bug}

openmoverse, 2023-03-24:
Hello everyone.

We're trying to get a camera work with extrnal signal (in the docs its called external continuous mode). We've been connecting the FSYNC input pin to a PWM and we expect to get captures at the frequency set by the PWM and the software. 
What we get is flickering and high noise images of the IR streams (we're working with those for now).

[image]
here's what our streams show.

[image]
here's our input to the FSYNC pin

```
import depthai as dai
import cv2
import time

pipeline = dai.Pipeline()
fps = 30
# camRgb = pipeline.create(dai.node.ColorCamera)
# camRgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.RGB)
# camRgb.setIspScale(2,3)
# camRgb.setFps(60)
# camRgb.initialControl.setFrameSyncMode(dai.CameraControl.FrameSyncMode.INPUT)
# # camRgb.initialControl.setExternalTrigger(4,3)

# xoutRgb = pipeline.create(dai.node.XLinkOut)
# xoutRgb.setStreamName("color")
# camRgb.isp.link(xoutRgb.input)

monoLeft = pipeline.create(dai.node.MonoCamera)
monoLeft.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
# monoLeft.initialControl.setManualExposure(1e6/100, 100)
monoLeft.setBoardSocket(dai.CameraBoardSocket.LEFT)
monoLeft.setFps(fps)
monoLeft.initialControl.setFrameSyncMode(dai.CameraControl.FrameSyncMode.INPUT)

xoutLeft = pipeline.create(dai.node.XLinkOut)
xoutLeft.setStreamName("left")
monoLeft.out.link(xoutLeft.input)

monoRight = pipeline.createMonoCamera()
# monoRight.initialControl.setManualExposure(1e6/100, 100) #Get rid of flickering
monoRight.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
monoRight.setBoardSocket(dai.CameraBoardSocket.RIGHT)
monoRight.setFps(fps)
monoRight.initialControl.setFrameSyncMode(dai.CameraControl.FrameSyncMode.INPUT)

xoutRight = pipeline.create(dai.node.XLinkOut)
xoutRight.setStreamName("right")
monoRight.out.link(xoutRight.input)

# Connect to device with pipeline
# config = dai.Device.Config()
# config.board.gpio[6] = dai.BoardConfig.GPIO(dai.BoardConfig.GPIO.OUTPUT,
#                                             dai.BoardConfig.GPIO.Level.HIGH)

with dai.Device(pipeline,dai.DeviceInfo("194430105108801300")) as device:
    # arr = ['left', 'right', 'color']
    arr = ['left', 'right']
    queues = {}
    frames = {}

    for name in arr:
        queues[name] = device.getOutputQueue(name)

    print("Starting...")

    while True:
        for name in arr:
            if queues[name].has():
                frames[name]=queues[name].get().getCvFrame()

        for name, frame in frames.items():
            cv2.imshow(name, frame)

        key = cv2.waitKey(1)
        if key == ord('q'):
            break
```
and here's our sample code

Note, we always match the FPS in software with the Hz on the PWM (this is required right?)
We've also messed around with the duty cycle of the PWN, currently operating at 10%. What would be the correct value for this?

Thank you

nemccarthy, 2023-06-10:
did you have any luck with this @openmoverse

moversevlad, 2023-06-14:
Yeah, we did in fact had some progress with this but we havent accessed the results properly yet.

arubin, 2023-06-16:
We are experiencing this exact issue with no response ... I opened a discussion on the support forum referencing this issue to increase visibility to the Luxonis firmware team: https://discuss.luxonis.com/d/1657-problem-syncing-multiple-oak-d-poe-with-fsync/2

themarpe, 2023-06-19:
Hi @openmoverse - sorry for the delay

> Note, we always match the FPS in software with the Hz on the PWM (this is required right?)

Yes, depending on the sensor used (eg OV9282) this is required. Also note, that OV9282 is highly strict about its timings on FSync input, so make sure that matches down to `us` range.

---

The "external trigger mode" should be more forgiving - that is however WIP for OV9282/9792

themarpe, 2023-06-19:
@arubin 

IMX sensors aren't fsynced at the moment, but soft synchronized. To your issue, I'd first double check the frequency (setFps) & PWM input frequnecy and also make sure that before streaming is started, that remains in a steady state (afaik low).

We are working on some other parts at the moment, but we might revisit this within some upcoming scope of work