luxonis/depthai-core #787: [BUG] {Can not ssh into OAK-FFC-6P module}

Macbull, 2023-03-28:
**Describe the bug**
Can not access the device OAK-FFC-6P through SSH or python API. Need more documentation.

**Minimal Reproducible Example**
1. Connect OAK-FFC-6P module to PC using usb-c cable.
2. Try using cam_test.py script from rvc3_support and rvc3_develop branch of depthai-python.
   - Error : `No DepthAI camera module found`
3. Open powershell and use `ssh root@169.254.0.55`. 
   - Error: `ssh: connect to host 169.254.0.55 port 22: Connection timed out`
4. Use `ipconfig` command
    - Output:  IP address of usb ethernet interface is `169.254.86.247`
5. Set static ip address of usb ethernet interface as  `169.254.0.10`. Subnet mask as `255.255.0.0`
6. I was not sure what to set as default gateway (in windows) or default router (in ubuntu). I tried `169.254.0.10` and `169.254.0.55`. Nothing worked.
7. Use `ipconfig` command again
    - Output:  IP address of usb ethernet interface is `169.254.0.10`
8. Try using  `ssh root@169.254.0.55` again
    - Error: `port 22 : No route to host`

**Expected behavior**
After following the documentation, I was expecting to be able to either ssh into the device to use it, or directly use it on my host PC using depthai-python. There is no gateway mentioned in the ssh section of the document. Could you please describe the whole process on either windows or linux of setting up static ip address and ssh into the device ?

Macbull, 2023-03-28:
I tried the whole process again. I was able to SSH into device from a linux machine after updating LuxonisOS and other software. The static IP assigned to linux host machine was in same subnet by default. So ssh worked without changing IP address.

On windows, I had to manually set static IP address, The address 192.168.197.10 didn't work for some reason. I used 192.168.197.59, which worked instantly.

closed 2023-03-28 by Macbull

stevenuecke7, 2023-05-03:
@Macbull , this may be an odd question, but what password did you use to SSH into the module?

Hiram31, 2023-07-24:
> 

Hi @Macbull, did you figure out the password? I was asked to provide the password to SSH into the module too.