luxonis/depthai-core #792: [BUG] {possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'nn' (X_LINK_ERROR)'}

loginabhay, 2023-03-30:
Getting this issue "possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'nn' (X_LINK_ERROR)'" while running the camera module with multiple devices randomly.

Erol444, 2023-04-03:
Hi @loginabhay , This is usually caused by incorrect pipeline or connection issues. If the former, please [submit full MRE](https://docs.luxonis.com/en/latest/pages/support/).
Thanks, Erik

loginabhay, 2023-04-13:
Hi @Erol444 , Thanks for the response i am using two oak d cameras and below is the pipeline which i am using

#include <chrono>
#include <iostream>

// Includes common necessary includes for development using depthai library
#include "depthai/depthai.hpp"

using namespace std;

static const std::vector<std::string> labelMap = {"background", "aeroplane", "bicycle",     "bird",  "boat",        "bottle", "bus",
                                                  "car",        "cat",       "chair",       "cow",   "diningtable", "dog",    "horse",
                                                  "motorbike",  "person",    "pottedplant", "sheep", "sofa",        "train",  "tvmonitor"};

std::shared_ptr<dai::Pipeline> getPipeline(std::string nnPath){
    auto pipeline = std::make_shared<dai::Pipeline>();

    auto camRgb = pipeline->create<dai::node::ColorCamera>();
    camRgb->setPreviewSize(300, 300);
    camRgb->setResolution(dai::ColorCameraProperties::SensorResolution::THE_4_K);
    camRgb->setInterleaved(false);
    camRgb->setIspScale(1,3);
    camRgb->setPreviewKeepAspectRatio(false);

    auto xoutFrames = pipeline->create<dai::node::XLinkOut>();
    xoutFrames->setStreamName("frames");
    camRgb->video.link(xoutFrames->input);

    auto spatialDetectionNetwork = pipeline->create<dai::node::MobileNetSpatialDetectionNetwork>();
    spatialDetectionNetwork->setBlobPath(nnPath);
    spatialDetectionNetwork->setConfidenceThreshold(0.5f);
    spatialDetectionNetwork->input.setBlocking(false);
    spatialDetectionNetwork->setBoundingBoxScaleFactor(0.5);
    spatialDetectionNetwork->setDepthLowerThreshold(100);
    spatialDetectionNetwork->setDepthUpperThreshold(5000);
    camRgb->preview.link(spatialDetectionNetwork->input);

    auto monoLeft = pipeline->create<dai::node::MonoCamera>();
    auto monoRight = pipeline->create<dai::node::MonoCamera>();
    auto stereo = pipeline->create<dai::node::StereoDepth>();
    stereo->setDefaultProfilePreset(dai::node::StereoDepth::PresetMode::HIGH_DENSITY);
    stereo->setDepthAlign(dai::CameraBoardSocket::RGB);
    stereo->setOutputSize(monoLeft->getResolutionWidth(), monoLeft->getResolutionHeight());

    monoLeft->setResolution(dai::MonoCameraProperties::SensorResolution::THE_400_P);
    monoLeft->setBoardSocket(dai::CameraBoardSocket::LEFT);
    monoRight->setResolution(dai::MonoCameraProperties::SensorResolution::THE_400_P);
    monoRight->setBoardSocket(dai::CameraBoardSocket::RIGHT);
    monoLeft->out.link(stereo->left);
    monoRight->out.link(stereo->right);

    auto xoutRgb = pipeline->create<dai::node::XLinkOut>();
    xoutRgb->setStreamName("rgb");
    spatialDetectionNetwork->passthrough.link(xoutRgb->input);

    auto xoutNN = pipeline->create<dai::node::XLinkOut>();
    xoutNN->setStreamName("nn");
    spatialDetectionNetwork->out.link(xoutNN->input);

    auto xoutDepth = pipeline->create<dai::node::XLinkOut>();
    xoutDepth->setStreamName("depth");
    stereo->depth.link(spatialDetectionNetwork->inputDepth);
    spatialDetectionNetwork->passthroughDepth.link(xoutDepth->input);

    return pipeline;
}

int main(int argc, char** argv){
    std::string nnPath(BLOB_PATH);

[... 73 lines cut ...]
                    cv::putText(oframe, depthY.str(), cv::Point(x1 + 10, y1 + 65), cv::FONT_HERSHEY_TRIPLEX, 0.5, 255);
                    std::stringstream depthZ;
                    depthZ << "Z: " << (int)detection.spatialCoordinates.z << " mm";
                    cv::putText(oframe, depthZ.str(), cv::Point(x1 + 10, y1 + 80), cv::FONT_HERSHEY_TRIPLEX, 0.5, 255);

                    cv::rectangle(oframe, cv::Rect(cv::Point(x1, y1), cv::Point(x2, y2)), color, cv::FONT_HERSHEY_SIMPLEX);
                }
                cv::imshow(streamName, oframe);
            }

            else{
                cv::imshow(streamName, zeroframe);
            }
         }
  
         int key = cv::waitKey(1);
         if(key == 'q' || key == 'Q') {
             return 0;
         }
     }
     return 0;
}
 

The code is running normally but it fails randomly by producing above error when the cameras are left in an ideal environment when their is no activity please let me know if anything else need to be added in the pipeline to debug this error efficiently.

loginabhay, 2023-05-25:
@Erol444 Hi Erik,

Can you please help me here i am kind of stuck here some of the cameras are crashing randomly can you let me know what i am doing wrong inside of it

themarpe, 2023-06-07:
@loginabhay sorry for the delay.

@JanLipovsek please try to reproduce and we'll debug locally

themarpe, 2023-06-07:
@loginabhay also, are you running latest release (v2.21.2) or latest `develop`? Might make sense to try latest develop, due to that being a release candidate for v2.22.0 release

loginabhay, 2023-06-08:
@themarpe Yes i have tried it with the latest build as well but still getting the issue i am restarting the pipeline internally for the quick fix but after some time it gives me fatal error and stops below is the error

_`[1844301011CF6F0E00] [1.4] [33.656] [system] [critical] Fatal error. Please report to developers. Log: 'ic_main' '272'`_

JanLipovsek, 2023-06-14:
@loginabhay I was able to reproduce your issue. I expect that your problem is related to the devices not being powered sufficiently, which it turn causes them the crash. Try connecting them to a reliable power source and report back to us if your issue still persists. Additionally, as a friendly tip, please make sure that the code you provide is formatted in a single code block and compiles (your code had errors that I had to fix before I was able to run it). Thanks.