luxonis/depthai-core #797: Very slow startup with new v2.21.0

saulthu, 2023-04-05:
I have a system with many cameras connected. Some have an IMU, some do not. With version `v2.20.2` and earlier, I would just create the pipeline as if all cameras had an IMU (because there was no way to ask) and the program would start up fine, just printing out a warning.

Now, with `v2.21.0`, the same program throws an exception if the pipeline is created with an IMU node but the camera has no IMU, which forces the use of the new IMU detection API. That sounds nice, except that querying for IMU existence requires connection to the device, which is always very slow. Having to do it twice (first with the non-IMU pipeline, then check for IMU, then second connection with the IMU pipeline) is 2x very slow, which is unacceptable.

I have a very strong preference for the old method that allows creation of the IMU node **_without_** throwing an exception, just a warning. Alternatively, it would help to have a fast way of determining presence of an IMU **_prior_** to making the pipeline e.g. have the connection always being very fast, maybe by having an option to connect without creating a pipeline, just enough to query the device properties, then later start the pipeline. As it stands, I cannot upgrade to the new version.

szabi-luxonis, 2023-04-10:
@saulthu 

> Alternatively, it would help to have a fast way of determining presence of an IMU prior to making the pipeline e.g. have the connection always being very fast, maybe by having an option to connect without creating a pipeline, just enough to query the device properties, then later start the pipeline

This is already possible and was the intended way to use.
See example [here](https://github.com/luxonis/depthai-python/blob/main/examples/IMU/imu_version.py)

saulthu, 2023-04-10:
@szabi-luxonis

The linked example does this:
```
import depthai as dai
device = dai.Device()
imuType = device.getConnectedIMU()
imuFirmwareVersion = device.getIMUFirmwareVersion()
```

And that may be fine for a single device, yet it is even incomplete for a single device. In that single device case the code must continue by closing that `device` and reconnecting with a pipeline that either contains an IMU node or does not contain an IMU node, hence more than doubling the connection time. In my case I have many cameras, and connection is already very slow without increasing the connection time even more, which has happened in the transition from `2.20.x` to `2.21.x`.

saulthu, 2023-04-10:
### _try_2_20_x_method() with depthai==2.20.x:
    start: 0.000 s
    pipeline: 0.000 s
    len(all_infos)=7
    getAllAvailableDevices: 0.232 s
    [1944301081E8221300] [3.1.4.1] [1.312] [IMU(0)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!
    [1944301031F2221300] [3.1.4.3] [1.801] [IMU(0)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!
    [19443010F125231300] [3.1.4.4] [1.825] [IMU(0)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!
    [1944301091E8221300] [3.1.1] [1.378] [IMU(0)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!
    [18443010E1EA3A1300] [3.1.2] [1.845] [IMU(0)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!
    len(all_devs)=7
    open: 12.355 s
    [19443010F128231300] [3.1.4.2] [1.422] [IMU(0)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!
    close: 32.657 s
    del: 0.013 s

### _try_2_20_x_method() with depthai==2.21.x:
    start: 0.000 s
    pipeline: 0.000 s
    len(all_infos)=7
    getAllAvailableDevices: 0.262 s
    Traceback (most recent call last):
      File "C:\Users\saul\Documents\test_depthai\try_2.20.x.py", line 52, in <module>
        _try_2_20_x_method()
      File "C:\Users\saul\Documents\test_depthai\try_2.20.x.py", line 35, in _try_2_20_x_method
        dev = dai.Device(pipeline, info)
    RuntimeError: IMU(0) - IMU invalid settings!: IMU not detected. Your board doesn't have IMU or requires firmware update (Set imu.enableFirmwareUpdate(True) explicitly.). 1, 321

### _try_2_21_x_method() with depthai==2.21.x:
    start: 0.000 s
    imu_pipeline: 0.000 s
    non_pipeline: 0.000 s
    len(all_infos)=7
    getAllAvailableDevices: 0.238 s
    getConnectedIMU: 48.674 s
    len(all_devs)=7
    open: 12.880 s
    close: 32.204 s
    del: 0.013 s

saulthu, 2023-04-10:
To be clear, in these tests I'm connecting to 6x OAK-1-W which do not have IMU, and 1x OAK-D-W which does have IMU.

szabi-luxonis, 2023-04-10:
@saulthu Connecting to the device is separate from starting a pipeline.
You can connect to the device:
`device = dai.Device()`
and query connected cameras, IMU, and their capability and create the pipeline based on that, then start the pipeline, without rebooting the device.
See example [here](https://github.com/luxonis/depthai-python/blob/bfe65785b69c99f5ad96762ee480a7f30fa46075/examples/IMU/imu_rotation_vector.py#L42)

In your case it would be something like:

```
device = dai.Device()
imuType = device.getConnectedIMU()
pipeline = dai.Pipeline()
if imuType != "NONE":
  imu = pipeline.create(dai.node.IMU)
  xlinkOut = pipeline.create(dai.node.XLinkOut)
  xlinkOut.setStreamName("imu")
  imu.enableIMUSensor(...)
....
with device:
    device.startPipeline(pipeline)
```

saulthu, 2023-04-10:
OK, this issue appears to be a misunderstanding! I will try this immediately. Thanks.

closed 2023-04-10 by saulthu