luxonis/depthai-core #802: [BUG] Unable to ping or ssh into RCV3 SOM after updating image to 1.8

asa, 2023-04-11:
**Describe the bug**
After updating the firmware on my early access version of the 6p, I no longer was able to ping the interface. I tried the two ip addresses mentioned in the doc without success, from both linux(ubu 20.04) and osx (m2 ventura 13.3.1).

**Minimal Reproducible Example**
Start with early access hardware with unknown early version of os on it. Flash with fip-oak-ffc-6p-rvc3-1.2.0.bin using movisoc-fwu -a [path/to/fip.bin]. after reboot device would not come up as a usbethernet device on either linux or osx.

On linux used the flashtool to put 1.8 image and latest firmware fip-oak-ffc-6p-rvc3-1.2.0.bin onto the device. The flash reported success. 
After this I was still unable to get either interface ip address in either ip range to work on either the mac or linux host. Previous to this flashing was able to ssh into this device on the 169.254.0.55 address.

**Expected behavior**
The firmware and os update tools should recover a device to a state it can be sshed into.

szabi-luxonis, 2023-04-11:
@asa IP address changed in newer OS to: `ssh root@192.168.197.55`

themarpe, 2023-04-12:
@asa 

https://docs.google.com/document/d/1r3en4EqgDG3hum8o5yS1DJ5vSCb6aNqyA38tIM8723U/edit#

You may follow the recovery steps outlined in the above document

asa, 2023-04-12:
Those are the recovery steps I followed.

Synergyst, 2023-05-28:
I followed these same steps and ended up also soft-bricking my device which was previously working by following the update instructions.

Here's the dmesg logs after the update:
```
[ 5386.088616] usb 2-1.4: new SuperSpeed USB device number 9 using tegra-xusb
[ 5386.111403] usb 2-1.4: New USB device found, idVendor=1d6b, idProduct=0103
[ 5386.111475] usb 2-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5386.111523] usb 2-1.4: Product: USB Ethernet
[ 5386.111565] usb 2-1.4: Manufacturer: Luxonis
[ 5386.111606] usb 2-1.4: SerialNumber: t00000002
[ 5386.138498] cdc_ncm 2-1.4:1.0: MAC-Address: 8a:d8:3f:ad:b2:ac
[ 5386.141172] cdc_ncm 2-1.4:1.0 usb0: register 'cdc_ncm' at usb-70090000.xusb-1.4, CDC NCM, 8a:d8:3f:ad:b2:ac
[ 5386.460480] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[ 5386.490781] cdc_ncm 2-1.4:1.0 usb0: 3750 mbit/s downlink 3750 mbit/s uplink
[ 5386.522772] cdc_ncm 2-1.4:1.0 usb0: 3750 mbit/s downlink 3750 mbit/s uplink
[ 5386.938777] cdc_ncm 2-1.4:1.0 usb0: 3750 mbit/s downlink 3750 mbit/s uplink
[ 5386.970839] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
```

Also see:
```
root@jetson:~# ifconfig usb0 ; ping -c 1 192.168.197.10 ; ping -c 1 192.168.197.55
usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.197.10  netmask 255.255.255.0  broadcast 192.168.197.255
        inet6 fe80::88d8:3fff:fead:b2ac  prefixlen 64  scopeid 0x20<link>
        ether 8a:d8:3f:ad:b2:ac  txqueuelen 1000  (Ethernet)
        RX packets 146  bytes 6093 (6.0 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 50  bytes 6259 (6.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

PING 192.168.197.10 (192.168.197.10) 56(84) bytes of data.
64 bytes from 192.168.197.10: icmp_seq=1 ttl=64 time=0.072 ms
--- 192.168.197.10 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.072/0.072/0.072/0.000 ms

PING 192.168.197.55 (192.168.197.55) 56(84) bytes of data.
From 192.168.197.10 icmp_seq=1 Destination Host Unreachable
--- 192.168.197.55 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms
root@jetson:~# 
root@jetson:~# 
root@jetson:~# cd ~/depthai-python
root@jetson:~/depthai-python# python3.8 examples/host_side/device_information.py
Searching for all available devices...

Couldn't find any available devices.
root@jetson:~/depthai-python#
root@jetson:~/depthai-python# 
root@jetson:~/depthai-python# lsusb --tree
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 4: Dev 9, If 0, Class=Communications, Driver=cdc_ncm, 5000M
        |__ Port 4: Dev 9, If 1, Class=CDC Data, Driver=cdc_ncm, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/5p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
root@jetson:~/depthai-python# 
root@jetson:~/depthai-python# lsusb
Bus 002 Device 009: ID 1d6b:0103 Linux Foundation NCM (Ethernet) Gadget
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@jetson:~/depthai-python#
```

Synergyst, 2023-06-01:
Solution:
Use DHCP to fetch an IP from the RVC3 device rather than using a static IP (`169.254.0.10` netmask `255.255.0.0` or `192.168.197.10` netmask `255.255.255.0`) after upgrading the OS as the subnet changes to `255.255.255.240`

CC @asa