luxonis/depthai-core #813: [BUG] OAK-SoM-Pro-S3 + OAK-FFC-4P IMU not detected

AtsushiSaito, 2023-04-17:
In OAK-SoM-Pro-S3 + OAK-FFC-4P, the following error occurred in the IMU example included in depthai-python.

```sh
atsushi@ubuntu:~/depthai-python/examples/IMU$ python3 imu_gyroscope_accelerometer.py
[########] [192.168.197.55] [########] [IMU(0)] [error] Node threw exception, stopping the node. Exception message: IMU BMI270 not detected
```

Switch to branch rvc3_develop and used commit 8d6ca6191995317858303a76c22658799e2164ea.

### environment
- Ubuntu 22.04.1 LTS
- Python 3.10.6
- depthai version: 2.19.1.0.dev0+8d6ca6191995317858303a76c22658799e2164ea

Issue #783 of a similar problem, although the device is different.

themarpe, 2023-04-23:
Thanks for the report - CC: @Nitram1195

Nitram1195, 2023-04-25:
The support of IMU (BNO086) was added. Use the following [link](https://luxonisos.fra1.digitaloceanspaces.com/release/recovery/fip-oak-ffc-4p-rvc3-1.4.0.bin) to download the latest FIP image.

AtsushiSaito, 2023-04-27:
@Nitram1195 
Thank you for updating the FIP image!
Does this update include support for BMI270?

themarpe, 2023-05-03:
@AtsushiSaito no - its either or. If your board has BMI270, it won't work

closed 2023-05-03 by themarpe