luxonis/depthai-core #815: [BUG] Stereo from host no longer works with v2.21.2

developer-mayuan, 2023-04-20:
Currently with depthai-core v2.21.2, I found it's no longer able to generate stereo depth by feeding a image pair directly to a StereoDepth through XLinkIn. The error I got is like the following:

```
[18443010E1DF1AF500] [3.3.2] [1.060] [StereoDepth(4)] [error] There is no available extrinsic calibration between camera ID: '0' and '0'
```

I guess this regression was introduced when implementing the feature of "Multi stereo support, ability to run stereo between any 2 calibrated cameras". Since the new StereoDepth node works with any calibrated camera pairs, the input streams may carry the metadata which **real** camera port this data is coming from and the stereo depth node fetches extrinsic calibration data based on it. I'm wondering if we the SDK can provide a flag to allow user to disable this feature to make the stereoDepth node works as previous versions?

**Minimal Reproducible Example**
```cpp
  dai::CalibrationHandler calibration_data{kTestCalibrationData};

  dai::Pipeline pipeline;
  pipeline.setCalibrationData(calibration_data);

  // Read images of left and right image pairs.
  auto right_in = pipeline.create<dai::node::XLinkIn>();
  right_in->setStreamName("right_in");
  auto left_in = pipeline.create<dai::node::XLinkIn>();
  left_in->setStreamName("left_in");

  auto stereo_node = pipeline.create<dai::node::StereoDepth>();

  left_in->out.link(stereo_camera_node->left);
  right_in->out.link(stereo_camera_node->right);

  auto xout_depth = pipeline.create<dai::node::XLinkOut>();
  xout_depth->setStreamName("depth");
  stereo_node->out.link(xout_depth->input);

  dai::DeviceInfo device_info;

  auto wait = std::chrono::steady_clock::now();
  bool found;
  do {
    std::tie(found, device_info) = dai::Device::getFirstAvailableDevice();
  } while (!found && std::chrono::steady_clock::now() - wait < 5s);

  ASSERT_TRUE(found);

  dai::Device device{pipeline, device_info};

  // Feed them into the pipeline.
  cv::Mat left_test_image = cv::imread(kLeftTestImage);
  cv::Mat right_test_image = cv::imread(kRightTestImage);

  dai::ImgFrame left_frame;
  left_frame.setSize(stereo_config.image_size.height, stereo_config.image_size.width);
  left_frame..setType(dai::ImgFrame::Type::BGR888p);
  toPlanar(left_test_image, left_frame.getData());

  dai::ImgFrame right_frame;
  right_frame.setSize(stereo_config.image_size.height,
                    stereo_config.image_size.width);
  right_frame.setType(dai::ImgFrame::Type::BGR888p);
  toPlanar(right_test_image, right_frame.getData());

  auto left_in_queue = device.getInputQueue("left_in");
  auto right_in_queue = device.getInputQueue("right_in");
  auto depth_out_queue = device.getOutputQueue("depth", 4, false);

  left_in_queue->send(left_frame);
  right_in_queue->send(right_frame);

  std::this_thread::sleep_for(1s);
  auto depth_frame = depth_out_queue->tryGet<dai::ImgFrame>();

  ASSERT_NE(depth_frame, nullptr);
```

**Expected behavior**
A correct stereo depth image is generated.

**Additional context**
This error was observed when using depthai-core SDK v2.21.2 with FFC-3P board with 2 camera modules connecting to socket CAM B and C.

themarpe, 2023-04-22:
Thanks for the report - cc @szabi-luxonis 

@developer-mayuan i guess setting instance number (imgframe - corresponds to cam socket id) to the one corresponding to the passed calibration data, will work as expected.

szabi-luxonis, 2023-04-22:
@developer-mayuan as @themarpe mentioned above you need to set the instance number correctly. Previously calibration in the stereo node was hard-coded to LEFT and RIGHT camera sockets, that's why it worked with incorrectly set input frame instance numbers.
This is not a regression, but rather an improvement, in prior versions stereo was limited to LEFT and RIGHT sockets.
We have a reference example for stereo from host, where the instance number is set correctly: https://github.com/luxonis/depthai-python/blob/main/examples/StereoDepth/stereo_depth_from_host.py#L678

developer-mayuan, 2023-04-23:
Thanks @themarpe  and @szabi-luxonis for your quick solution. Setting instance number does fix the issue I got here which is great! However, I do find other issues cause by this change.

On our robots, we have mounted our oak-d camerasin upside down way due to some limitations in our hardware design, which means we also have to calibrate these cameras in this way. To workaround this issue, in our previous calibration tool and runtime driver, we apply the following code to make the stereo node thinking the camera is still in normal position by swapping the left and right cameras streams and also flip images by 180 degree:

```c++
  if (upside_down) {
    // left and right camera position is swapped if upside down.
    monoLeft->setBoardSocket(dai::CameraBoardSocket::RIGHT);
    monoRight->setBoardSocket(dai::CameraBoardSocket::LEFT);

    // also need to rotate image output for each camera.
    monoLeft->setImageOrientation(dai::CameraImageOrientation::ROTATE_180_DEG);
    monoRight->setImageOrientation(dai::CameraImageOrientation::ROTATE_180_DEG);
    camRgb->setImageOrientation(dai::CameraImageOrientation::ROTATE_180_DEG);
  } else {
    monoLeft->setBoardSocket(dai::CameraBoardSocket::LEFT);
    monoRight->setBoardSocket(dai::CameraBoardSocket::RIGHT);
  }
```

This workaround works in previous sdk, because stereo depth node still can fetch correct calibration data which is independent of the actual camera sockets of incoming images, which is not true anymore with the latest SDK.

I'm wondering if there is a way to make the current SDK provides an option to perform the previous behavior? Or any ideas how I can address this issue with the latest SDK? Thanks!

szabi-luxonis, 2023-04-24:
@developer-mayuan you can swap calibration data for the 2 cameras: 
https://github.com/luxonis/depthai-python/commit/60d5688335748113c0d71218741b3c756ea23134#diff-bc263f47c3c296cff090513c408eb10d48523d751d6b537f6c433441bc20dd64R191
or follow the latest develop, for features like [this](https://github.com/luxonis/depthai-core/pull/716) one. 
>This is also a preparation for potential name aliasing, where a device will be able to be 180 degree rotated without having to modify which camera is now "left / right / ..."

closed 2023-05-07 by szabi-luxonis