luxonis/depthai-core #820: [BUG] Pipeline crashes inside StereoDepth Node with error `PRINT:LeonMss: Temporal filter shave 11 didn't exit gracefully. Force stopping. `

developer-mayuan, 2023-04-25:
**Describe the bug**

I have a simple spatial object tracking pipeline using OAK-FFC-3P board, which including:
* Stereo Depth Node
* Object Detection Node
* Object Tracking Node
* Spatial Calculator Node

and this pipeline worked fine with depthai-core version before v2.21. However, recently when I upgrade the sdk version to v2.21.2, I observed a stability issue that 70% of time, the pipeline will crashes within 5 seconds after the pipeline launched.

To debug this issue, I set the environment variable `DEPTHAI_DEBUG=1` to get the full debug logs as shown below:
```
[2023-04-25 21:12:41.160] [debug] Environment 'DEPTHAI_DEBUG' set to '1'
[2023-04-25 21:12:41.160] [debug] Library information - version: 2.21.2, commit: 125feb8c2e16ee4bf71b7873a7b990f1c5f17b18 from 2023-04-05 21:24:25 +0300, build: 2023-04-19 18:50:09 +0000
[2023-04-25 21:12:41.163] [debug] Initialize - finished
[ INFO] [1682457161.164393423]: Found device with mxid: 184430108105851200
[2023-04-25 21:12:41.276] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 3ms, archive read: 111ms
[2023-04-25 21:12:41.758] [debug] Resources - Archive 'depthai-device-fwp-3575b77f20e796b4e79953bf3d2ba22f0416ee8b.tar.xz' open: 3ms, archive read: 593ms
[ INFO] [1682457162.356161274]: Created pipeline to device: 184430108105851200
[2023-04-25 21:12:42.356] [debug] Device - OpenVINO version: 2021.4
[2023-04-25 21:12:42.357] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"imu":null,"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 11 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 00 bb 00 be
[2023-04-25 21:12:43.256] [debug] Searching for booted device: DeviceInfo(name=1.7.2.4, mxid=184430108105851200, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[184430108105851200] [1.7.2.4] [3.880] [system] [warning] PRINT:LeonCss: BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0xEB7B0756
BootloaderConfig.options2 checksum doesn't match. Is: 0x00003C0A should be: 0x533EDEEC
GPIO boot mode 0x16, interface USBD
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
DepthAI Firmware - version: 3575b77f20e796b4e79953bf3d2ba22f0416ee8b
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Didn't find any boards matching the eeprom data
Board options: 00000000
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: ff ff ff
Invalid Flash JEDEC ID... No NOR available
Could not register the spi device
inited hal ---
def: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
val: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
Failed to probe IR driver LM3644
Opening bus for IR driver: 3
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Is booted from flash by bootloader: 0
Networking not available...
=== Enumerating on socket: Cam_A / RGB / Center
initial keepalive, countdown: 9
=== Enumerating on socket: Cam_B / Left
  >> Registered camera TG161B (ov9282) as /dev/Camera_1
camera socket: 1, name: left
        config - w: 1280, h: 720, type: MONO
        config - w: 1280, h: 720, type: COLOR
        config - w: 1280, h: 800, type: MONO
        config - w: 1280, h: 800, type: COLOR
Adding socket 1: cam 5. Sen name: OV9282
=== Enumerating on socket: Cam_C / Right
[... 185 lines cut ...]
  // Can't enable temporalFilter with depthai-core v2.21.2.
  auto config = depth->initialConfig.get();
  config.postProcessing.temporalFilter.enable = true;
  config.postProcessing.temporalFilter.persistencyMode =
      dai::RawStereoDepthConfig::PostProcessing::TemporalFilter::
          PersistencyMode::VALID_2_IN_LAST_3;
  depth->initialConfig.set(config);

  if (!stereo_config.left_mesh_file.empty() &&
      !stereo_config.right_mesh_file.empty()) {
    depth->loadMeshFiles(stereo_config.left_mesh_file,
                         stereo_config.right_mesh_file);
    // TODO(yuan): The mesh_step_size used here is a magic number. Need to make
    // sure it matches the value used in stereo node and calibration tool.
    depth->setMeshStep(stereo_config.mesh_step_size,
                       stereo_config.mesh_step_size);
  }
  return depth;
}
```

I didn't observe pipeline crashes anymore if I disabled the temporal filter.

**Expected behavior**
Pipeline didn't crash with temporal filter enabled in StereoDepth Node.

szabi-luxonis, 2023-05-02:
@developer-mayuan thanks for the report, however, I can't reproduce it.
If possible, please further reduce the MRE based on [this](https://github.com/luxonis/depthai-python/blob/main/examples/StereoDepth/depth_post_processing.py) example. Please also attach calibration file, might be relevant.

szabi-luxonis, 2023-05-02:
> @developer-mayuan thanks for the report, however, I can't reproduce it. If possible, please further reduce the MRE based on [this](https://github.com/luxonis/depthai-python/blob/main/examples/StereoDepth/depth_post_processing.py) example. Please also attach calibration file, might be relevant.

I was able to reproduce and pushed a fix on latest `develop`. Let me know how it goes on your side

closed 2023-05-07 by szabi-luxonis