luxonis/depthai-core #833: GCC13 compatibility: XLinkStream.hpp needs to include stdexcept for std::runtime_error

MarioSavu, 2023-05-19:
The issue and solution are straight forward, an #include <stdexcept> should be added in XLinkStream.hpp, and then gcc13 will compile the file.
Otherwise you'll see errors as this: 
....../depthai-test/depthai-core/include/depthai/xlink/XLinkStream.hpp:81:47: error: expected class-name before ‘{’ token
   81 | struct XLinkError : public std::runtime_error {

MarioSavu, 2023-05-31:
I tried pushing a branch with a fix to open a PR when I opened this issue, but I don't have write permissions, so instead I'll just say that you should modify this https://github.com/luxonis/depthai-core/blob/main/include/depthai/xlink/XLinkStream.hpp#L16 to include this: `#include <stdexcept>`

themarpe, 2023-08-09:
Resolved in `develop` and `main` now as well

closed 2023-08-09 by themarpe