luxonis/depthai-core #837: [BUG] camera doesn't connect on boot usbfs: usb_submit_urb returned -19

kisdia, 2023-05-27:
**Describe the bug**
My OAK-D PRO W doesn't connect on boot. If I disconnect and re-connect the cable it works. But I need it to work on boot. 
I am using x86 computer. I have 2 other OAK-D PRO W cameras that work without this issue on the same system, in the same USB port with the same cable.

**Expected behavior**
Camera works on boot.

**Attach system log**
[http://autodiscovery.co.uk:8088/index.php/s/aRNSkKNNzTQGeYd]

**Additional context**
Here is the dmesg output:
[ 26.485963] audit: type=1107 audit(1685173493.427:50): pid=554 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call" bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.4" pid=1753 label="snap.snap-store.snap-store" peer_pid=577 peer_label="unconfined"  exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[ 26.486343] audit: type=1107 audit(1685173493.427:51): pid=554 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call" bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="CheckAuthorization" mask="send" name=":1.4" pid=1753 label="snap.snap-store.snap-store" peer_pid=577 peer_label="unconfined" exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[ 26.488982] audit: type=1107 audit(1685173493.431:52): pid=554 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call" bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.4" pid=1753 label="snap.snap-store.snap-store" peer_pid=577 peer_label="unconfined" exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[ 26.489270] audit: type=1107 audit(1685173493.431:53): pid=554 uid=103 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call" bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="CheckAuthorization" mask="send" name=":1.4" pid=1753 label="snap.snap-store.snap-store" peer_pid=577 peer_label="unconfined" exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[ 26.914258] audit: type=1400 audit(1685173493.855:54): apparmor="DENIED" operation="open" profile="snap.snap-store.snap-store" name="/etc/PackageKit/Vendor.conf" pid=1753 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[ 27.320213] audit: type=1400 audit(1685173494.259:55): apparmor="DENIED" operation="open" profile="snap.snap-store.snap-store" name="/etc/appstream.conf" pid=1753 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[ 484.277285] usb 1-2: USB disconnect, device number 3
[ 484.895258] usb 2-2: new SuperSpeed USB device number 3 using xhci_hcd
[ 484.916496] usb 2-2: New USB device found, idVendor=03e7, idProduct=f63b, bcdDevice= 1.00
[ 484916509] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 484.916516] usb 2-2: Product: Luxonis Device
[ 484.916520] usb 2-2: Manufacturer: Intel Corporation
[ 484.916525] usb 2-2: SerialNumber: 1844301001FD840E00
[ 485.265924] usb 1-1: USB disconnect, device number 2
[ 485.979268] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[ 486.003985] usb 2-1: New USB device found, idVendor=03e7, idProduct=f63b, bcdDevice= 1.00
[ 486.004022] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 486.004028] usb 2-1: Product: Luxonis Device
[ 486.004033] usb 2-1: Manufacturer: Intel Corporation
[ 486.004038] usb 2-1: SerialNumber: 18443010E1F3930F00
[ 486.699109] usb 1-2: new high-speed USB device number 9 using xhci_hcd
[ 486.847795] usb 1-2: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[ 486.847813] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 486.847819] usb 1-2: Product: Movidius MyriadX
[ 486.847824] usb 1-2: Manufacturer: Movidius Ltd.
[ 486.847828] usb 1-2: SerialNumber: 03e72485
[ 486.975217] usb 2-2: USB disconnect, device number 3
[ 489.248674] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.248854] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.249032] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.249210] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.249397] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.249581] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.249769] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.249949] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.250127] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.250310] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.250489] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.250672] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.250861] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.251074] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.251266] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.251462] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.251652] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.251812] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.252001] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.252181] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.252360] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.252540] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.252719] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.252906] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.253088] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.253265] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.253442] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.253620] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.253798] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.253983] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.254164] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.254342] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.254519] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.254697] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.254882] usb 2-1: usbfs: usb_submit_urb returned -19
[ 489.255061] usb 2-1: usbfs: usb_submit_urb returned -19

themarpe, 2023-05-29:
@kisdia 

What state is the device when you plug it in? (dmesg output)
```
[ 486.847819] usb 1-2: Product: Movidius MyriadX
```
This?

Can you delimit the dmesg timestamps based on the action you took? (Initial run of some script upon reboot, then connect/disconnect of cable, then run of some script again when its supposed to work)

Can you also showcase the same for the other device where this procedure works?

kisdia, 2023-06-05:
Clean boot, 3 cameras plugged in I am using the depthai-experiments gen2-multiple-devices example. 
2 of the cameras always works no problem. They work in any of the 3 ports with any of the 3 cables. The 3rd one not detected on boot. dmesg as per above. But if I disconnect the cable and re-connect it works.

I unpluged the camera that doesn't work (it is always the same, regardless of cable and socket used), and re-connected it. dmesg shows:

[  526.269926] usb 2-2: SerialNumber: 1844301001FD840E00
[  526.985230] usb 1-2: new high-speed USB device number 13 using xhci_hcd
[  527.137881] usb 1-2: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[  527.137889] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  527.137891] usb 1-2: Product: Movidius MyriadX
[  527.137892] usb 1-2: Manufacturer: Movidius Ltd.
[  527.137893] usb 1-2: SerialNumber: 03e72485
[  527.241240] usb 2-2: USB disconnect, device number 5

To trace back the same serial was detected during boot:

[  340.838705] usb 1-2: USB disconnect, device number 3
[  341.519951] usb 2-2: new SuperSpeed USB device number 3 using xhci_hcd
[  341.544734] usb 2-2: New USB device found, idVendor=03e7, idProduct=f63b, bcdDevice= 1.00
[  341.544741] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  341.544743] usb 2-2: Product: Luxonis Device
[  341.544744] usb 2-2: Manufacturer: Intel Corporation
[  341.544746] usb 2-2: SerialNumber: 1844301001FD840E00

But then this happens:

[  349.238732] usb 2-2: usbfs: usb_submit_urb returned -19
[  349.238892] usb 2-2: usbfs: usb_submit_urb returned -19
[  349.252373] usb 1-2: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[  349.252380] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  349.252382] usb 1-2: Product: Movidius MyriadX
[  349.252384] usb 1-2: Manufacturer: Movidius Ltd.
[  349.252385] usb 1-2: SerialNumber: 03e72485
[  349.375783] usb 2-2: USB disconnect, device number 3
[  370.340369] loop16: detected capacity change from 0 to 943224

Erol444, 2023-07-03:
@kisdia we just received your device - it works as expected. We tried on Windows, latest depthai version.
[image]
[image]

kisdia, 2023-07-05:
Thank you for the assistance.
I have received the replacement, unfortunately the issue persists but now with a different camera. 
I don't think it is the camera, it must be related to USB or boot process, or the computer. I will test a different computer.

kisdia, 2023-07-06:
Mitigated the issue by running a usb-reset before starting the script. Something must be not correct during boot...

`#!/usr/bin/env bash

# USB 3.1 Only
for port in $(lspci | grep xHCI | cut -d' ' -f1); do
    echo -n "0000:${port}"| sudo tee /sys/bus/pci/drivers/xhci_hcd/unbind;
    sleep 5;
    echo -n "0000:${port}" | sudo tee /sys/bus/pci/drivers/xhci_hcd/bind;
    sleep 5;
done

# All USB
for port in $(lspci | grep USB | cut -d' ' -f1); do
    echo -n "0000:${port}"| sudo tee /sys/bus/pci/drivers/xhci_hcd/unbind;
    sleep 5;
    echo -n "0000:${port}" | sudo tee /sys/bus/pci/drivers/xhci_hcd/bind;
    sleep 5;
done`