luxonis/depthai-core #839: [BUG] with onboard USB3 connection, gotten error Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

chengguizi, 2023-06-01:
**Describe the bug**

I have been trying to design a PCB board that has USB3 connection connecting the CPU and the VPU SoM. I could see the enumeration of the bootloader device. I can also flash the bootloader version correctly.

However, I can't launch the camera through depthAI properly. Below is the error message.

```
[2023-06-01 11:21:51.712] [debug] Library information - version: 2.21.2, commit: c2d6ba3a9e9a3145628537437bc37255aa1212a2 from 2023-04-08 16:28:03 +0800, build: 2023-05-29 07:59:56 +0000
[2023-06-01 11:21:51.718] [debug] Initialize - finished
[2023-06-01 11:21:51.935] [info] Found 1 devices:
[2023-06-01 11:21:51.936] [info] 0: id=184430107185BD0F00, name=5.1
[2023-06-01 11:21:51.936] [info] main: start device with id 184430107185BD0F00
[2023-06-01 11:21:51.938] [info] initialising device 184430107185BD0F00
[2023-06-01 11:21:51.939] [info] fps = 20
[2023-06-01 11:21:51.939] [info] Creating device with specific id 184430107185BD0F00
[2023-06-01 11:21:51.942] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 5ms, archive read: 221ms
[2023-06-01 11:21:52.151] [info] found device with specified id 184430107185BD0F00
[2023-06-01 11:21:52.151] [debug] Device - OpenVINO version: 2022.1
[2023-06-01 11:21:52.153] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[[41,{"direction":1,"drive":2,"level":0,"mode":7,"pull":0,"schmitt":false,"slewFast":false}],[46,{"direction":1,"drive":2,"level":0,"mode":7,"pull":0,"schmitt":false,"slewFast":false}]],"imu":null,"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null}
libnop:
0000: b9 11 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 02 29 b9 07 07 01
0020: 00 00 02 00 00 2e b9 07 07 01 00 00 02 00 00 bb 00 be be be be be be 01 00 bb 00 be
[2023-06-01 11:21:52.876] [debug] Resources - Archive 'depthai-device-fwp-3575b77f20e796b4e79953bf3d2ba22f0416ee8b.tar.xz' open: 5ms, archive read: 1157ms
[2023-06-01 11:21:53.651] [debug] Searching for booted device: DeviceInfo(name=5.1, mxid=184430107185BD0F00, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
Aborted
```

dmesg
```
[   43.929365] usb 5-1: USB disconnect, device number 2
[   53.274352] usb 5-1: new high-speed USB device number 3 using xhci-hcd
[   53.414782] usb 5-1: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[   53.414789] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   53.414793] usb 5-1: Product: Movidius MyriadX
[   53.414796] usb 5-1: Manufacturer: Movidius Ltd.
[   53.414800] usb 5-1: SerialNumber: 03e72485
[   62.666891] usb 5-1: USB disconnect, device number 3
[   72.014343] usb 5-1: new high-speed USB device number 4 using xhci-hcd
[   72.154726] usb 5-1: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[   72.154733] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   72.154737] usb 5-1: Product: Movidius MyriadX
[   72.154740] usb 5-1: Manufacturer: Movidius Ltd.
[   72.154743] usb 5-1: SerialNumber: 03e72485

```

May I know what could be wrong here? It seems that the bootloader did not reboot into the actual USB device? It just re-populated as bootloader device after about 10 second.

chengguizi, 2023-06-01:
The last responsive dmesg is along the line of `usb 5-1: USB disconnect, device number 3`. Then about 10 second the device come back, but depthAI does not recognise it (as it does not mount as movidius stick correctly?)

chengguizi, 2023-06-03:
Root cause found. It is due to I2C signal lines accidentally shorted to GND on the baseboard. This would prevent the SoM from booting.

closed 2023-06-03 by chengguizi

Shivam7Sharma, 2024-06-03:
Hi @chengguizi , 

I am getting this error and I don't understand what I have to change to make the camera work again with Oak FFC3P and a color camera. Can you please elaborate on how you solved this issue? Any help will be greatly appreciated.