luxonis/depthai-core #845: [BUG] Occasional X_LINK_ERROR on POE devices

justin-larking-pk, 2023-06-12:
We are repeatedly getting occasional disconnects from our oakd-w-poe devices that reads with the following error. The disconnects seems to happen randomly or just after some time.

The error:
"terminate called after throwing an instance of 'std::runtime_error'
  what():  Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'depth' (X_LINK_ERROR)'
[oakd-11] process has died [pid 1499128, exit code -6, cmd /home/grover/catkin_ws/devel/lib/grover_vision/oakd_node __name:=oakd __log:=/home/grover/.ros/log/659fb568-d1e9-11ed-b878-23da073fdadf/oakd-11.log].
log file: /home/grover/.ros/log/659fb568-d1e9-11ed-b878-23da073fdadf/oakd-11*.log"

We noticed the same error on our usb devices occasionally, we put that down to vibration/usb-cable issues. 

What could be causing this? Could vibration be causing Ethernet disconnect? It is using the Luxonis cable and we've confirmed the POE switch the camera is connected to isn't losing power. Could it be an issue with the pipeline?

themarpe, 2023-06-19:
@justin-larking-pk 

Run the `examples/CrashReport/crash_report.py` example after that happens and post it here.

WRT vibration - yes that might be the case if connection temporarily drops. Both issues can show themselves in similar light

diablodale, 2023-07-03:
The unhandled throw leading to terminate could be https://github.com/luxonis/XLink/issues/59, the bugs I fix in https://github.com/luxonis/depthai-core/pull/809, or the bug I'm waiting on Luxonis to answer https://github.com/luxonis/depthai-core/pull/809#issuecomment-1581150389