luxonis/depthai-core #847: Question about specTranslation params

borongyuan, 2023-06-13:
The release note of v2.21.2 mentioned:
> Added disparityToDepthUseSpecTranslation, rectificationUseSpecTranslation, and depthAlignmentUseSpecTranslation options for advanced usage

Looks like their default values are:
```
bool disparityToDepthUseSpecTranslation = true;
bool rectificationUseSpecTranslation = false;
bool depthAlignmentUseSpecTranslation = true;
```
I don't understand why calibration data is not used as default for `disparityToDepth` and `depthAlignment`.
Not sure if these are related to #840. I didn't see a noticeable effect in my tests.

Erol444, 2023-06-20:
Hi @borongyuan , I believe it really depends on the calibration quality. Usually, you will have more accurate translation if you look at specs (CAD model) than the one obtained from camera matrix. But for rectification we found that we obtained better results if we took extrinsics from calibration.

borongyuan, 2023-06-21:
Thanks, I get it.

closed 2023-06-21 by borongyuan