luxonis/depthai-core #850: [BUG] OAK-FFC-6P RVC3 No available devices

RaymondCM, 2023-06-16:
We received two OAK-FFC-6P devices today. Neither of the devices work in any of the following configurations:

- depthai-python (on the rvc3_develop branch)
- depthai-core (on the rvc3_develop branch)
- depthai-core (on the rvc3_develop branch) and updating the OS to 1.10 and the FIP to latest
- With no modules plugged in or with one or multiple
- With DC and USB input or with only USB input

We can successfully ssh into the device, but we cannot get either depthai-core examples or depthai-python utilities or examples to discover the 6P device.

Expected behaviour of plugging in the device and installing the development branch of rvc3 support was for the camera to be connected. Neither 6P device, whether updated, nor works.

```bash
# DepthAI-Python (rvc3_develop)
# DEPTHAI_DEBUG=1 python utilities/cam_test.py --cameras rgb,c 
Traceback (most recent call last):
  File "/home/raymond/projects/pycast/depthai-python/utilities/cam_test.py", line 284, in <module>
    with dai.Device(*dai_device_args) as device:
RuntimeError: No available devices
```

```bash
# DepthAI-Core (rvc3_develop)
# ./build/examples/script_get_device_info
terminate called after throwing an instance of 'std::runtime_error'
  what():  No available devices
```

themarpe, 2023-06-19:
@RaymondCM 

Please run with following set:
`DEPTHAI_LEVEL=debug utilities/cam_test.py --cameras rgb,c` 

Also, make sure that you have installed `examples/install_dependencies.py` under the selected `rvc3_develop` branch

RaymondCM, 2023-06-20:
@themarpe 

```bash
# DEPTHAI_LEVEL=debug utilities/cam_test.py --cameras rgb,c

[2023-06-20 11:33:14.582] [debug] Python bindings - version: 2.19.1.0.dev+0f2e70d9067fec82e709cd1543621cb76a01a44c from  build: 2023-06-14 08:42:11 +0000
[2023-06-20 11:33:14.582] [debug] Library information - version: 2.19.1, commit: b70cfc3df8604fa515e22a7c39fab6283b8abe66 from 2023-06-12 18:45:53 +0200, build: 2023-06-14 08:13:15 +0000
[2023-06-20 11:33:14.584] [debug] Initialize - finished
Enabled cameras:
    rgb : color
DepthAI version: 2.19.1.0.dev+0f2e70d9067fec82e709cd1543621cb76a01a44c
DepthAI path: ./depthai-python/venv/lib/python3.10/site-packages/depthai.cpython-310-x86_64-linux-gnu.so
[2023-06-20 11:33:14.677] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.22.tar.xz' open: 3ms, archive read: 91ms
[2023-06-20 11:33:14.952] [debug] Resources - Archive 'depthai-device-fwp-b511dc1a23aa9016007f7d6e9979a5a322bcefbe.tar.xz' open: 3ms, archive read: 366ms
Traceback (most recent call last):
  File "./depthai-python/utilities/cam_test.py", line 209, in <module>
    with dai.Device() as device:
RuntimeError: No available devices
```

RaymondCM, 2023-06-23:
@themarpe Any update? We can successfully get the devices to work on Windows, but not Linux based systems.

themarpe, 2023-06-23:
@RaymondCM wrt Linux, if you can ssh - I assume it has to do with firewall blocking UDP broadcast. I'd check that first (port 11491).

Otherwise, we are transitioning away from ETH adapter approach into native bulk USB EP's, but will take a bit longer to have ready. This will however drastically improve UX and interoperability

RaymondCM, 2023-06-27:
@themarpe This was the issue. I can now see the device. 

As a note, the port mappings (CAM_D is CAM_F) are wrong, i.e. an IMX477 in CAM_A doesn't work, but in CAM_D launched with CAM_F does work. The device at the moment is incredibly delicate, with intermittent connection issues and multiple cameras not working (defeating the purpose of the 6P board!), I.e. IMX477 in CAM_F only supports a single mono camera being plugged in, any more and everything breaks. I think the switch to USB will be helpful!

Also having an issue where the Xout Queue only provides a single frame and is then empty. I'll close this issue for now waiting for a more stable device.

closed 2023-06-27 by RaymondCM

RaymondCM, 2023-06-27:
For context, the single queue issue was due to colour preview being enabled (resulted in RTEMS_FATAL_SOURCE_ASSERT)

themarpe, 2023-08-09:
@apirrone 

Which OS are you currently on? (on device: `cat /etc/os-release`)

You may update by transferring latest 1.11 (https://luxonisos.fra1.digitaloceanspaces.com/release/1.11/luxonisos-1.11.dm-verity.mender) to device & flashing with `mender -install [os.dm-verity.mender] && reboot`

apirrone, 2023-08-09:
Hi @themarpe, I am on Ubuntu 22.
I already tried flashing this version of the os, did not work :/

Thanks !

apirrone, 2023-08-10:
I tried again to install the 1.11 version on the second board I ordered. It still does not work and now I can't ssh into it anymore

themarpe, 2023-08-10:
@apirrone I'd suggest trying the `flashtool` method - CC: @jakaskerl for instructions

RaymondCM, 2023-10-11:
@sferoze Ensure your firewall is disabled, update the camera firmware over SSH, unset the manual connection information in your system settings and leave auto DNS. If this still doesn't work, follow the instructions here to manually set it up. Verify your camera is connected by checking SSH connection.

https://docs.google.com/document/d/1r3en4EqgDG3hum8o5yS1DJ5vSCb6aNqyA38tIM8723U/edit

RaymondCM, 2023-10-11:
@apirrone follow the instructions in the link above. If you still cannot connect, verify your network settings and firewall. You shouldn't need to set the IP manually with the latest firmware, I find it's best to update the device when SSH'd in, once you get it working.