luxonis/depthai-core #853: [Feature-Request] Expose stats used for 3A algorithms

tk4218, 2023-06-20:
### Start with the `why`:
We are working to improve image quality in poor lighting situations and have found that the 3A evaluations (auto-focus/auto-exposure/auto-white balance) do not give us our desired results. Specifically we see ISO increase more quickly than exposure and we have limits that we want to keep things within. Additionally, the production we are building can supply additional light, which can be used to increase brightness of an image without increasing the ISO significantly. We've tried playing with auto exposure on a ROI and using custom-tuning bins, but would like more control over setting certain values such as exposure time and ISO sensitivity, lens position, etc. based on data gathered from the sensor. In discussion with Erik, it sounds like there are plans to expose the data used in calculation for the 3A algorithms so that they can be used in custom logic for focus/exposure/white balance.

### Move to the `what`:
Please make available the stats used in the 3A evaluation algorithms on the ISP available so that custom algorithms may be implemented either on host or on-device (flashed Script node). Our plan would be to use these values while manually setting the exposure and ISO.

chengguizi, 2023-07-04:
Related to https://github.com/luxonis/depthai-core/issues/761, almost the same request for the same purpose

chengguizi, 2023-11-01:
Hi @Erol444 , May I know if this feature could be added? It was a free stat from the ISP anyway? Or has to be explicitly computed onboard RVC?

themarpe, 2023-11-06:
@chengguizi this feature could be added but isn't high on priority list at the moment. Yes params would be used from previously computed ones.